rockchip-linux/kernel #1: missing .dts file for firefly rk3288 reload board

bschiett, 2016-10-10:
I compiled the kernel using branch release-20160920, hoping to use it with my 
firefly rk3288 reload board (see http://en.t-firefly.com/en/firenow/firefly_rk3288/). 

I configured the kernel with rockchip_linux_defconfig and I'm using rk3288-firefly.dts. I managed to boot with the kernel but it seems that the hardware on my firefly reload board doesn't work. I can see graphics via HDMI and see the linux desktop but when I plug in a mouse it doesn't work, probably because the .dts file for my reload board has to be different from the original .dts for the firefly board. 

Also, I get the below kernel messages on the serial console endlessly looping, I suppose this has to do with the .dts file not being correct for my board. See below for the kernel messages on the serial console which are looping non stop. 

I know that firefly has a kernel config and dts file for the reload board in their kernel repository. I thought I could copy their .dts and kernel config to use it with the above branch of rockchip's kernel but I'm not sure how to do it. Is it a matter of putting their kernel config in ./arch/arm/configs/ and .dts file in ./arch/arm/boot/dts/ and i can automatically use it or is there something else I should do? I would appreciate it if the kernel config/dts file for the reload board are added to rockchip's kernel repo. 

[  690.278519] dhd_txglom_enable: enable 0
[  690.282359] dhd_conf_set_txglom_params: swtxglom=0, txglom_ext=0
[  690.288406] dhd_conf_set_txglom_params: txglom_bucket_size=0
[  690.294061] dhd_conf_set_txglom_params: txglomsize=0,
deferred_tx_len=0, bus_txglom=0
[  690.301906] dhd_conf_set_txglom_params: tx_in_rx=1, tx_max_offset=0
[  690.308190] dhd_bus_devreset:  WLAN OFF DONE
[  690.313696] wifi_platform_set_power = 0
[  690.317576] ======== PULL WL_REG_ON LOW! ========
[  690.322278] [WLAN_RFKILL]: rockchip_wifi_power: 0
[  690.327626] [WLAN_RFKILL]: wifi shut off power.
[  690.332155] wl_android_wifi_off out
[  690.467023] dhd_stop: Exit
[  690.480667] dhd_set_mac_addr_handler: interface info not available/down
[  690.590926] dhd_open: Enter eebb7800
[  690.594506]
[  690.594506] Dongle Host Driver, version 1.363.59.144.1 (r)
[  690.601499] wl_android_wifi_on in 1
[  690.605414] wl_android_wifi_on in 2: g_wifi_on=0
[  690.610055] wifi_platform_set_power = 1
[  690.613885] ======== PULL WL_REG_ON HIGH! ========
[  690.618684] [WLAN_RFKILL]: rockchip_wifi_power: 1
[  690.623381] [WLAN_RFKILL]: wifi turn on power. -1
[  690.837146] sdio_reset_comm():
[  690.843390] dwmmc_rockchip ff0d0000.dwmmc: could not set regulator OCR (-22)
[  690.850729] dwmmc_rockchip ff0d0000.dwmmc: failed to enable vmmc regulator
[  690.887384] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req
400000Hz, actual 400000HZ div = 0)
[  690.910541] mmc_host mmc1: Bus speed (slot 0) = 200000Hz (slot req
200000Hz, actual 200000HZ div = 0)
[  690.958322] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[  690.966781] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[  690.975271] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[  690.986141] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[  691.098590] mmc_host mmc1: Bus speed (slot 0) = 50000000Hz (slot
req 50000000Hz, actual 50000000HZ div = 0)
[  692.167059] dwmmc_rockchip ff0d0000.dwmmc: Successfully tuned phase to 117
[  692.174157] sdioh_start: set sd_f2_blocksize 128
[  692.183550]
[  692.183550]
[  692.183550] dhd_bus_devreset: == WLAN ON ==
[  692.192021] F1 signature read @0x18000000=0x16224335
[  692.207003] F1 signature OK, socitype:0x1 chip:0x4339 rev:0x1 pkg:0x2
[  692.214942] DHD: dongle ram size is set to 786432(orig 786432) at 0x180000
[  692.224124] dhd_os_open_image: /system/etc/firmware/config.txt (204
bytes) open success
[  692.233653] dhd_conf_read_config: dhd_doflow = 1
[  692.238549] dhd_conf_read_config: srl = 15
[  692.242687] dhd_conf_read_config: lrl = 15
[  692.247548] dhd_conf_set_fw_name_by_chip:
firmware_path=/system/etc/firmware/fw_bcm4339a0_ag.bin
[  692.256352] dhdsdio_download_firmware: set use_rxchain 0
[  692.261742] dhdsdio_download_firmware: set txglomsize 40
[  692.267139] sdioh_set_mode: set txglom_mode to copy
[  692.272035] Final fw_path=/system/etc/firmware/fw_bcm4339a0_ag.bin
[  692.278286] Final nv_path=/system/etc/firmware/nvram_AP6335.txt
[  692.284222] Final conf_path=/system/etc/firmware/config.txt
[  692.297615] dhd_os_open_image:
/system/etc/firmware/fw_bcm4339a0_ag.bin (517576 bytes) open success
[  692.360876] dhd_os_open_image:
/system/etc/firmware/nvram_AP6335.txt (2375 bytes) open success
[... 93 lines cut ...]
[  693.075423] [<c07cefbc>] (dev_change_flags) from [<c07de7bc>]
(do_setlink+0x308/0x738)
[  693.083338] [<c07de7bc>] (do_setlink) from [<c07dfbf0>]
(rtnl_newlink+0x378/0x6d0)
[  693.090907] [<c07dfbf0>] (rtnl_newlink) from [<c07e00f0>]
(rtnetlink_rcv_msg+0x1a8/0x1bc)
[  693.099084] [<c07e00f0>] (rtnetlink_rcv_msg) from [<c07fd694>]
(netlink_rcv_skb+0x68/0xbc)
[  693.107347] [<c07fd694>] (netlink_rcv_skb) from [<c07dde34>]
(rtnetlink_rcv+0x28/0x34)
[  693.115263] [<c07dde34>] (rtnetlink_rcv) from [<c07fcfec>]
(netlink_unicast+0x158/0x250)
[  693.123353] [<c07fcfec>] (netlink_unicast) from [<c07fd4ec>]
(netlink_sendmsg+0x358/0x374)
[  693.131616] [<c07fd4ec>] (netlink_sendmsg) from [<c07b2160>]
(sock_sendmsg+0x44/0x54)
[  693.139445] [<c07b2160>] (sock_sendmsg) from [<c07b2b2c>]
(___sys_sendmsg+0x190/0x204)
[  693.147360] [<c07b2b2c>] (___sys_sendmsg) from [<c07b398c>]
(__sys_sendmsg+0x4c/0x70)
[  693.155188] [<c07b398c>] (__sys_sendmsg) from [<c07b39c8>]
(SyS_sendmsg+0x18/0x1c)
[  693.162758] [<c07b39c8>] (SyS_sendmsg) from [<c0107200>]
(ret_fast_syscall+0x0/0x3c)
[  693.170494] ---[ end trace 94868236ff675595 ]---

bschiett, 2016-10-10:
I can also add that in the mainline there is no kernel config for rockchip but there is a .dts file for the reload board, see 

https://github.com/torvalds/linux/tree/master/arch/arm/configs
https://github.com/torvalds/linux/blob/master/arch/arm/boot/dts/rk3288-firefly-reload.dts

hizukiayaka, 2016-10-11:
Yes, it is what I submit to mainline. For the firefly rk3288 reload, I would suggest you to use the mainline kernel.
You may not use the firefly release dts for firefly reload, the hardware connection is a little different.

bschiett, 2016-10-11:
@hizukiayaka thanks, yesterday I checked out mainline stable tag 4.8.1 (latest) and then noticed your rk3288-firefly-reload.dts is not in it yet, so I went to download it manually together with its core board .dtsi file, and I put them in the dts folder and modified the Makefile to compile your .dts file. For the linux defconfig I used the defconfig from the firefly kernel tree (original v3 kernel, not mainline) since was not sure what the correct defconfig was I should use. I can now boot the kernel but I don't have any graphics output via HDMI. I assume this is because you haven't added HDMI out yet? I haven't manually installed any modules in the rootfs yet after compiling the kernel and haven't manually added any drivers yet. Should it be a matter of adding HDMI suport to your .dts file for the reload board to get basic graphics output working with mainline, or is there more to be done?

hizukiayaka, 2016-10-12:
Those dts files for firefly reload have been merged into linux next. Anyway, that dts is designed for mainline not rockchip develop branch, you need a extend property "supports-sd;" in sdmmc node to enable the support for SDMMC.
You may meeet such problem with this branch. Besides, I think the HDMI would work in mainline, but I have not confirmed it yet, you may have a try.

bschiett, 2016-10-12:
>  Besides, I think the HDMI would work in mainline, but I have not confirmed it yet, you may have a try.
> 
Thanks, I tried enabling it by setting the status of hdmi and i2c5 node to okay in your reload .dts file using the 4.8.1 stable branch on mainline but that was not enough? Is there anything else I should check?

bschiett, 2016-10-12:
In the rockchip kernel repo, "supports-sd;" is already in the sdmmc node,
in the file rk3288-firefly.dtsi. It's just that as soon as I enable the
lvds node and try to boot from SD, the kernel can't find the SD card
anymore so it can't find the root FS. Of course, this is for the classic
firefly board and not my reload board, maybe on the reload board the pins
used for lvds are the ones for the sd card so the moment I turn on the lvds
I disable the sd automatically?

In the mainline, I had no problem booting using your reload .dts but there
I haven't switched on lvds yet because some driver code seems to be missing
and I'm not sure what I need to import.

wzyy2, 2016-10-12:
hi
try this patch
https://patchwork.kernel.org/patch/9348571/

wzyy2, 2016-10-12:
And the endlessly looping is caused by networkmanager bug.
I have fix it in this commit
https://github.com/rockchip-linux/rk-rootfs-build/commit/64162ec00ca2ec62507be88b14402550339d01c9

bschiett, 2016-10-13:
I tried this patch but nothing changed, for some reason now that i've
enabled hdmi and lvds in my .dts file the sd card is no longer found by the
kernel ...  I now have:

[    2.638645] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req
400000Hz, actual 400000HZ div = 0)
[    2.658531] dwmmc_rockchip ff0d0000.dwmmc: 1 slots initialized
[    2.665639] dwmmc_rockchip ff0f0000.dwmmc: IDMAC supports 32-bit address
mode.
[    2.674916] dwmmc_rockchip ff0f0000.dwmmc: Using internal DMA controller.
[    2.681716] dwmmc_rockchip ff0f0000.dwmmc: Version ID is 270a
[    2.687495] dwmmc_rockchip ff0f0000.dwmmc: DW MMC controller at irq
31,32 bit host data width,256 deep fifo
[    2.708513] mmc_host mmc2: Bus speed (slot 0) = 400000Hz (slot req
400000Hz, actual 400000HZ div = 0)
[    2.729527] dwmmc_rockchip ff0f0000.dwmmc: 1 slots initialized
[    2.737451] sdhci-pltfm: SDHCI platform and OF driver helper

previously it was this:

[    2.198200] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req
400000Hz, actual 400000HZ div = 0)
[    2.218083] dwmmc_rockchip ff0d0000.dwmmc: 1 slots initialized
[    2.218333] dwmmc_rockchip ff0f0000.dwmmc: IDMAC supports 32-bit address
mode.
[    2.218342] dwmmc_rockchip ff0f0000.dwmmc: Using internal DMA controller.
[    2.218348] dwmmc_rockchip ff0f0000.dwmmc: Version ID is 270a
[    2.218367] dwmmc_rockchip ff0f0000.dwmmc: DW MMC controller at irq
31,32 bit host data width,256 deep fifo
[    2.248080] mmc_host mmc2: Bus speed (slot 0) = 400000Hz (slot req
400000Hz, actual 400000HZ div = 0)
[    2.259091] mmc_host mmc0: Bus speed (slot 0) = 50000000Hz (slot req
50000000Hz, actual 50000000HZ div = 0)
[    2.259126] mmc0: new high speed SDHC card at address aaaa
[    2.259347] mmcblk0: mmc0:aaaa SL08G 7.40 GiB
[    2.264320]  mmcblk0: p1 p2 p3 p4 p5 p6 p7
[    2.268097] dwmmc_rockchip ff0f0000.dwmmc: 1 slots initialized
[    2.268198] sdhci-pltfm: SDHCI platform and OF driver helper

wzyy2, 2016-10-13:
remove networkmanager and install wicd instead

wzyy2, 2016-10-13:
https://github.com/rockchip-linux/rk-rootfs-build/commit/d59c9874ef66d89740f8d10a60aeac95120d5ad2

wzyy2, 2016-10-13:
LVDS shouldn't affect sd-card.
Could you show me your LVDS dts?

wzyy2, 2016-10-14:
The DTS you send seems no problem.
Try add rootwait to bootcmd.

bschiett, 2016-10-15:
I think the problem with the SD card not being found might have to do with
the regulators for vcc_sd and vccio_sd, I read that vccio_sd has to be able
to switch between 1.8v and 3.3v depending on the type of SD card (high
speed)? Maybe the .dts is missing some definitions to make this possible?

wzyy2, 2016-10-15:
Firefly don't support uhs, thet have a hardware bug in it.
You should remove sd-uhs-* in sdcard dts node and vccio_sd will not switch between 1.8v and 3.3v.

bschiett, 2016-10-15:
Thanks, I will try it.

bschiett, 2016-10-15:
I restored the sdmmc settings and I still have the same problem - kernel
doesn't find sd card and thus can't mount root fs ... for some reason when
i turn on lvds stops the sd from working, I don't understand ... I'll try
again with Randy's original reload .dts without my changes to see if it can
boot from sd (it did when I first tried it)

wzyy2, 2016-10-15:
LVDS and SD-Card work well in our other boards..
I think there might have hardware bugs....=_=

wzyy2, 2016-10-15:
I think you can ask firefly folks.

bschiett, 2016-10-16:
You are probably correct, it seems that pulling current from vcc_io on the firefly board is tricky since the sd card power comes from that supply... so for example enabling lcd_en immediately to connect vcc_lcd to the soc without any kind of startup delay is not a great idea... same with enabling the gpu, for some reason if not setting a ramp and it gets enabled then the sd card dissppears. So the ramp property should be set in the vdd_gpu for the reload board just like for the original ff board dts.
Thanks for helping, I appreciate it.

closed 2016-10-17 by wzyy2