rockchip-linux/kernel #2: instructions on rockchip wiki page to boot from sd card not correct?

bschiett, 2016-10-10:
I am using rockchip's kernel and u-boot repos with my firefly rk3288 reload board. I used the instructions at http://rockchip.wikidot.com/linux-user-guide to install an SPL in the emmc:

tools/mkimage -n rk3288 -T rksd -d spl/u-boot-spl-nodtb.bin u-boot-dtb.bin
sudo upgrade_tool db  ../rkbin/rk32/rk3288_boot.bin
sudo upgrade_tool wl 64 u-boot-dtb.bin

I then followed the instructions under "booting from sd card" to prepare and SD card. However, one problem I had when booting from it, is that the root=... in extlinux.conf referring to /dev/mmcblk0p7 does not work because as soon as the kernel starts it detects the emmc flash memory (16gb on my firefly rk3288 reload board), and it will name it mmcblk0 as device. If I changed it to mmcblk1p7 then it also doesn't work because then the SD card becomes mmcblk0 and the internal mmc becomes mmcblk1. Naming it mmcblk2p7 also doesn't work, so in the end I took the partition UUID I saw in the kernel output and used root=PARTUUID=... in extlinux.conf and that worked. When I inspected the UUID for the rootfs partition on my linux dev PC it was different than the UUID shown on the serial console when I booted with my board so I wrote down the UUID from the serial console and used that after PARTUUID=... 

It would be better to make it clear in the instructions at http://rockchip.wikidot.com/linux-user-guide that the u-boot SPL will not be able to find the emmc and load uboot, and that as such, without patching u-boot, the board will boot from SD card and that it's best to use PARTUUID=... instad of referring to the mmcblk0 device which can be renamed by the linux kernel when it starts.

hizukiayaka, 2016-10-11:
You need https://patchwork.ozlabs.org/patch/657573/
Also select the mmc device in u-boot you want to use before writing GPT table into it.

closed 2016-10-11 by hizukiayaka

wzyy2, 2016-10-15:
Hi， please update kernel to latest release-4.4。
We merge a new commit(https://github.com/rockchip-linux/kernel/commit/a0a1e633bc191a30c91a11ca14af3fafb8c0c46f) and now SD card is always mmc0 and emmc is always mmc2.
See http://rockchip.wikidot.com/linux-user-guide#toc1

bschiett, 2016-10-15:
> We merge a new commit(a0a1e63

Great, thanks so much!