rockchip-linux/kernel #4: Compilation errors for camsys on RK3288 device

mq002, 2017-01-11:
root@x:/build/koala/linux_sdk# ./build/mk-kernel.sh rk3288-koala
Building kernel for rk3288-koala board!
Using rk3288-koala_defconfig
#
# configuration written to .config
#
scripts/kconfig/conf  --silentoldconfig Kconfig
make[1]: 'arch/arm/boot/dts/rk3288-koala.dtb' is up to date.
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
make[1]: 'include/generated/mach-types.h' is up to date.
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  GZIP    kernel/config_data.gz
  CHK     kernel/config_data.h
  CC      drivers/media/video/rk_camsys/camsys_drv.o
  CC      drivers/media/video/rk_camsys/camsys_marvin.o
  CC      drivers/media/video/rk_camsys/camsys_mipicsi_phy.o
  CC      drivers/media/video/rk_camsys/camsys_soc_priv.o
  CC      drivers/media/video/rk_camsys/camsys_soc_rk3288.o
  CC      drivers/media/video/rk_camsys/camsys_soc_rk3368.o
  CC      drivers/media/video/rk_camsys/camsys_soc_rk3399.o
  CC      drivers/media/video/rk_camsys/ext_flashled_drv/rk_ext_fshled_ctl.o
  LD      drivers/media/video/rk_camsys/ext_flashled_drv/built-in.o
  LD      drivers/media/video/rk_camsys/built-in.o
  LD      drivers/media/built-in.o
  LD      drivers/built-in.o
  LINK    vmlinux
  LD      vmlinux.o
  MODPOST vmlinux.o
  GEN     .version
  CHK     include/generated/compile.h
  UPD     include/generated/compile.h
  CC      init/version.o
  LD      init/built-in.o
drivers/built-in.o：在函数‘camsys_soc_init’中：
/build/koala/linux_sdk/kernel/drivers/media/video/rk_camsys/camsys_soc_priv.c:61：对‘camsys_rk3368_cfg’未定义的引用
/build/koala/linux_sdk/kernel/drivers/media/video/rk_camsys/camsys_soc_priv.c:61：对‘rockchip_soc_id’未定义的引用
/build/koala/linux_sdk/kernel/drivers/media/video/rk_camsys/camsys_soc_priv.c:61：对‘camsys_rk3366_cfg’未定义的引用
/build/koala/linux_sdk/kernel/drivers/media/video/rk_camsys/camsys_soc_priv.c:61：对‘camsys_rk3399_cfg’未定义的引用
Makefile:947: recipe for target 'vmlinux' failed
make: *** [vmlinux] Error 1

wzyy2, 2017-01-11:
RK_CAMSYS driver is used in Android.

V4l2 camera driver(Although it's still ugly. ... It was migrated from Sofia-3GR) will be included in release 4.4  in a few days.

At current, you can got it from other branch.
https://github.com/rockchip-linux/kernel/tree/release-20160818-miniarm/drivers/media/platform/rk-isp10

mq002, 2017-01-11:
I will try the branch. thanks

closed 2017-01-11 by mq002