rockchip-linux/kernel #7: DRM: problem with drmModeSetPlane() scaling

mcerveny, 2017-04-03:
Hello.

I have problem with DRM overlay scaling. If I use "drmModeSetPlane(fd, plane_id, crtc_id, fb_id, 0, fb_x, fb_y, fb_width, fb_height, 0, 0, width << 16, height << 16);" and (height == fb_height*2) or (height == fb_height*4) the display image is corrupted. About first 16 lines are unexpectedly shifted and after image to end of screen is corrupted frame buffer (see attached image). Testing program is attached too.
HW/SW - TinkerBoard (rk3288), TinkerOS 1.4 (kernel 4.4.16#25) or TinkerOS 1.6 (kernel 4.4.17#1).

Any hit or patch ?
Thanks, Martin

[drm_test.zip](https://github.com/rockchip-linux/kernel/files/891436/drm_test.zip)

wzyy2, 2017-04-04:
Is height greater than hor_stride?

mcerveny, 2017-04-04:
You can use attached code. Compare height and hor_stride (or ver_stride?).
My results (all buggy) for less or grater as you requested ("printf("%d(%d)x%d(%d) %dx%d\n", width, hor_stride, height, ver_stride, fb_width, fb_height);"):

for **hor_stride>height**:
1024(**2048**)x**1280**(2048) 640x640
1024(2048)x1312(2048) 640x656
1024(2048)x1344(2048) 640x672
1024(2048)x1376(2048) 640x688
1024(2048)x1408(2048) 640x704
1024(2048)x1440(2048) 640x720
1024(2048)x1472(2048) 640x736
1024(2048)x1504(2048) 640x752
1024(2048)x1536(2048) 640x768
1024(2048)x1568(2048) 640x784
1024(2048)x1600(2048) 640x800
1024(2048)x1632(2048) 640x816
1024(2048)x1664(2048) 640x832
1024(2048)x1696(2048) 640x848
1024(2048)x1728(2048) 640x864
1024(2048)x1760(2048) 640x880
1024(2048)x1792(2048) 640x896
1024(2048)x1824(2048) 640x912
1024(2048)x1856(2048) 640x928
1024(2048)x1888(2048) 640x944
1024(2048)x1920(2048) 640x960
1024(2048)x1952(2048) 640x976
1024(2048)x1984(2048) 640x992
1024(2048)x2016(2048) 640x1008
1024(2048)x2048(2048) 640x1024

for **hor_stride<hight**:
1024(**1024**)x**1280**(2048) 640x640
1024(1024)x1312(2048) 640x656
1024(1024)x1344(2048) 640x672
1024(1024)x1376(2048) 640x688
1024(1024)x1408(2048) 640x704
1024(1024)x1440(2048) 640x720
1024(1024)x1472(2048) 640x736
1024(1024)x1504(2048) 640x752
1024(1024)x1536(2048) 640x768
1024(1024)x1568(2048) 640x784
1024(1024)x1600(2048) 640x800
1024(1024)x1632(2048) 640x816
1024(1024)x1664(2048) 640x832
1024(1024)x1696(2048) 640x848
1024(1024)x1728(2048) 640x864
1024(1024)x1760(2048) 640x880
1024(1024)x1792(2048) 640x896
1024(1024)x1824(2048) 640x912
1024(1024)x1856(2048) 640x928
1024(1024)x1888(2048) 640x944
1024(1024)x1920(2048) 640x960
1024(1024)x1952(2048) 640x976
1024(1024)x1984(2048) 640x992
1024(1024)x2016(2048) 640x1008
1024(1024)x2048(2048) 640x1024

wzyy2, 2017-04-05:
Is plane overlay plane， or primary plane?

mcerveny, 2017-04-05:
Overlay plane with NV12 support. you can see in sources as crtc_id = 48/plane_id=49.
Output of "modetest" (part of libdrm test):

```
 ./modetest 
trying to open device 'i915'...failed
trying to open device 'amdgpu'...failed
trying to open device 'radeon'...failed
trying to open device 'nouveau'...failed
trying to open device 'vmwgfx'...failed
trying to open device 'omapdrm'...failed
trying to open device 'exynos'...failed
trying to open device 'tilcdc'...failed
trying to open device 'msm'...failed
trying to open device 'sti'...failed
trying to open device 'tegra'...failed
trying to open device 'imx-drm'...failed
trying to open device 'rockchip'...done
Encoders:
id	crtc	type	possible crtcs	possible clones	
64	48	TMDS	0x00000002	0x00000000

Connectors:
id	encoder	status		name		size (mm)	modes	encoders
65	64	connected	HDMI-A-1       	380x300		14	64
  modes:
	name refresh (Hz) hdisp hss hse htot vdisp vss vse vtot)
  1280x1024 60 1280 1328 1440 1688 1024 1025 1028 1066 108000 flags: phsync, pvsync; type: preferred, driver
  1280x1024 75 1280 1296 1440 1688 1024 1025 1028 1066 135000 flags: phsync, pvsync; type: driver
  1152x864 75 1152 1216 1344 1600 864 865 868 900 108000 flags: phsync, pvsync; type: driver
  1024x768 75 1024 1040 1136 1312 768 769 772 800 78800 flags: phsync, pvsync; type: driver
  1024x768 70 1024 1048 1184 1328 768 771 777 806 75000 flags: nhsync, nvsync; type: driver
  1024x768 60 1024 1048 1184 1344 768 771 777 806 65000 flags: nhsync, nvsync; type: driver
  832x624 75 832 864 928 1152 624 625 628 667 57284 flags: nhsync, nvsync; type: driver
  800x600 75 800 816 896 1056 600 601 604 625 49500 flags: phsync, pvsync; type: driver
  800x600 72 800 856 976 1040 600 637 643 666 50000 flags: phsync, pvsync; type: driver
  800x600 60 800 840 968 1056 600 601 605 628 40000 flags: phsync, pvsync; type: driver
  640x480 75 640 656 720 840 480 481 484 500 31500 flags: nhsync, nvsync; type: driver
  640x480 73 640 664 704 832 480 489 491 520 31500 flags: nhsync, nvsync; type: driver
  640x480 60 640 656 752 800 480 490 492 525 25200 flags: nhsync, nvsync; type: driver
  720x400 70 720 738 846 900 400 412 414 449 28320 flags: nhsync, pvsync; type: driver
  props:
	1 EDID:
		flags: immutable blob
		blobs:

		value:
			00ffffffffffff0022f0932601010101
			3413010380261e78eeee95a3544c9926
			0f5054adef8081800101010101010101
			010101010101302a009851002a403070
			13007c2c1100001e000000fd00304c18
			530e000a202020202020000000fc0048
			50204c50313936350a202020000000ff
			00434e34393532303259430a20200031
	2 DPMS:
		flags: enum
		enums: On=0 Standby=1 Suspend=2 Off=3
		value: 0

CRTCs:
id	fb	pos	size
26	0	(0,0)	(0x0)
   0 0 0 0 0 0 0 0 0 0 flags: ; type: 
  props:
	22 FEATURE:
		flags: immutable bitmask
		values: scale=0x1
		value: 1
48	82	(0,0)	(1280x1024)
[... 125 lines cut ...]
		value: 3
50	0	0	0,0		0,0	0       	0x00000002
  formats: XR24 AR24 XB24 AB24 RG24 BG24 RG16 BG16
  props:
	5 type:
		flags: immutable enum
		enums: Overlay=0 Primary=1 Cursor=2
		value: 0
	6 SHARE_ID:
		flags: immutable range
		values: 0 4294967295
		value: 50
	51 rotation:
		flags: bitmask
		values: rotate-0=0x1 reflect-x=0x10 reflect-y=0x20
		value: 1
	43 FEATURE:
		flags: immutable bitmask
		values: scale=0x1 alpha=0x2
		value: 2

Frame buffers:
id	size	pitch

```

mcerveny, 2017-04-17:
Tested with release-20170417, not resolved.
Build with https://github.com/rockchip-linux/build crtc_id=61/plane_id=62.

wzyy2, 2017-04-20:
Interesting, I reproduced it.
This bug occur when src_h == crtc_h*2  or  src_h == crtc_h*4, so i think there are something in scale calculation.

mcerveny, 2017-05-01:
Probably the same problem: http://www.youtube.com/watch?v=yoZnlrHM1T0&t=1m57s

Kwiboo, 2017-05-02:
@mcerveny I think I had a similar issue and https://github.com/Kwiboo/linux-rockchip/commit/ca8b478e0269b18d2040c6cba4a39b57e7647044 seems to have helped with 1080p to 2160p scaling, unsure if that change will cause issues for other resolutions

wzyy2, 2017-05-02:
Hi,
Mark have give it a fix and it will be included in next release.
https://github.com/wzyy2/linux/commit/615a562aa3815c85892abbe005c681f6ff254822

Close it.

closed 2017-05-02 by wzyy2