rockchip-linux/kernel #10: DRM: totally broken VOP output mode selector rk3288

mcerveny, 2017-04-16:
Hello.

There is totally broken VOP for rk3288 platforms (TinkerBoard, ...). It usually boots only to 800x600 resolution that pass both errors (see also http://tinkerboarding.co.uk/forum/showthread.php?tid=61&pid=323#pid323).
I identified 2 issues in rockchip DRM/kernel but unable to correct:

**1) nearly all output modes are blocked for DMT/VESA modes**

Problem is in new validity checking (commit 7e3e0c5) and/or modification of "dclk_vop" clock driver (commit d072a98b maybe other). Due to strictly checking available clock in vop_crtc_mode_valid() (see https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/gpu/drm/rockchip/rockchip_drm_vop.c#L1443) and some modification in "**dclk_vop**" clock driver it is not possible to set even basic DMT/VESA modes. **Mode validation fails with MODE_CLOCK_RANGE**. Example (compare clock != request_clock from code):
```
Modeline 100:"1280x1024" 75 135000 1280 1296 1440 1688 1024 1025 1028 1066 0x40 0x5
clock error: 133333 != 135000
Modeline 80:"1280x1024" 60 108000 1280 1328 1440 1688 1024 1025 1028 1066 0x48 0x5
clock error: 100000 != 108000
Modeline 88:"1024x768" 75 78800 1024 1040 1136 1312 768 769 772 800 0x40 0x5
clock error: 74250 != 78800
Modeline 90:"1024x768" 60 65000 1024 1048 1184 1344 768 771 777 806 0x40 0xa
clock error: 62500 != 65000
```

**2) check modes to "fail-safe" framebuffer size (1024x768) and not to maximum available**

The problem is in late hdmi start (state "connected"). Framebuffer is already set do "1024x768" default but new modes are validated to this resolution and not to maximum available (8192x8192?). **Mode validation fails with MODE_VIRTUAL_X**. Sample from dmesg (XXXX is inside drm_mode_validate_size() "DRM_DEBUG_KMS("XXXX %dx%d %dx%d\n", mode->hdisplay, mode->vdisplay, maxX, maxY);" and "dump_stack()" shows call from drm_fb_helper_hotplug_event() (https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/gpu/drm/drm_fb_helper.c#L2161) and from rockchip_drm_output_poll_changed() (https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/gpu/drm/rockchip/rockchip_drm_fb.c#L220)):
```
[    1.440674] [drm:drm_helper_probe_single_connector_modes_merge_bits] [CONNECTOR:78:HDMI-A-1] disconnected
[    1.440693] rockchip-drm display-subsystem: No connectors reported connected with modes
[    1.448764] [drm:drm_setup_crtcs] 
[    1.448795] [drm:drm_sysfs_hotplug_event] generating hotplug event
[    1.448849] [drm:drm_enable_connectors] connector 78 enabled? no
[    1.448865] [drm:drm_setup_crtcs] picking CRTCs for 8192x8192 config
[    1.448881] [drm:drm_fb_helper_single_fb_probe] Cannot find any crtc or sizes - going 1024x768
[    1.462950] [drm:rockchip_gem_get_pages] rockchip_gem_get_pages, 201, end = 768, n_pages = 768
[    1.467034] [drm:rockchip_drm_fbdev_create] FB [1024x768]-24 kvaddr=f0921000 offset=0 size=3145728
....
[    4.246398] [drm:drm_helper_hpd_irq_event] [CONNECTOR:78:HDMI-A-1] status updated from disconnected to connected
[    4.246411] [drm:drm_sysfs_hotplug_event] generating hotplug event
[    4.246451] [drm:drm_fb_helper_hotplug_event] 
[    4.246467] [drm:drm_helper_probe_single_connector_modes_merge_bits] [CONNECTOR:78:HDMI-A-1]
....
[    4.572432] [drm:drm_mode_validate_size] XXXX 1280x1024 1024x768
```

Are there any new patches to resolve this issues ?

Thanks, Martin

wzyy2, 2017-04-17:
@yueshu

mcerveny, 2017-04-17:
Tested with release-20170417, not resolved.
```
[    3.570313] [drm:drm_mode_debug_printmodeline] Modeline 81:"1280x1024" 60 108000 1280 1328 1440 1688 1024 1025 1028 1066 0x48 0x5
[    3.570319] [drm:drm_mode_prune_invalid] Not using 1280x1024 mode: VIRTUAL_X
[    3.570333] [drm:drm_mode_debug_printmodeline] Modeline 83:"640x480" 0 31500 640 656 720 840 480 481 484 500 0x40 0xa
[    3.570339] [drm:drm_mode_prune_invalid] Not using 640x480 mode: CLOCK_RANGE
[    3.570351] [drm:drm_mode_debug_printmodeline] Modeline 84:"640x480" 0 31500 640 664 704 832 480 489 491 520 0x40 0xa
[    3.570357] [drm:drm_mode_prune_invalid] Not using 640x480 mode: CLOCK_RANGE
[    3.570368] [drm:drm_mode_debug_printmodeline] Modeline 85:"640x480" 0 25200 640 656 752 800 480 490 492 525 0x40 0xa
[    3.570374] [drm:drm_mode_prune_invalid] Not using 640x480 mode: CLOCK_RANGE
[    3.570385] [drm:drm_mode_debug_printmodeline] Modeline 86:"720x400" 0 28320 720 738 846 900 400 412 414 449 0x40 0x6
[    3.570390] [drm:drm_mode_prune_invalid] Not using 720x400 mode: CLOCK_RANGE
[    3.570402] [drm:drm_mode_debug_printmodeline] Modeline 87:"1280x1024" 0 135000 1280 1296 1440 1688 1024 1025 1028 1066 0x40 0x5
[    3.570407] [drm:drm_mode_prune_invalid] Not using 1280x1024 mode: VIRTUAL_X
[    3.570418] [drm:drm_mode_debug_printmodeline] Modeline 88:"1024x768" 0 78800 1024 1040 1136 1312 768 769 772 800 0x40 0x5
[    3.570424] [drm:drm_mode_prune_invalid] Not using 1024x768 mode: CLOCK_RANGE
[    3.570435] [drm:drm_mode_debug_printmodeline] Modeline 89:"1024x768" 0 75000 1024 1048 1184 1328 768 771 777 806 0x40 0xa
[    3.570441] [drm:drm_mode_prune_invalid] Not using 1024x768 mode: CLOCK_RANGE
[    3.570452] [drm:drm_mode_debug_printmodeline] Modeline 90:"1024x768" 0 65000 1024 1048 1184 1344 768 771 777 806 0x40 0xa
[    3.570458] [drm:drm_mode_prune_invalid] Not using 1024x768 mode: CLOCK_RANGE
[    3.570469] [drm:drm_mode_debug_printmodeline] Modeline 91:"832x624" 0 57284 832 864 928 1152 624 625 628 667 0x40 0xa
[    3.570474] [drm:drm_mode_prune_invalid] Not using 832x624 mode: CLOCK_RANGE
[    3.570486] [drm:drm_mode_debug_printmodeline] Modeline 94:"1152x864" 0 108000 1152 1216 1344 1600 864 865 868 900 0x40 0x5
[    3.570492] [drm:drm_mode_prune_invalid] Not using 1152x864 mode: VIRTUAL_X
[    3.570502] [drm:drm_helper_probe_single_connector_modes_merge_bits] [CONNECTOR:78:HDMI-A-1] probed modes :
[    3.570513] [drm:drm_mode_debug_printmodeline] Modeline 92:"800x600" 75 49500 800 816 896 1056 600 601 604 625 0x40 0x5
[    3.570524] [drm:drm_mode_debug_printmodeline] Modeline 93:"800x600" 72 50000 800 856 976 1040 600 637 643 666 0x40 0x5
[    3.570535] [drm:drm_mode_debug_printmodeline] Modeline 82:"800x600" 60 40000 800 840 968 1056 600 601 605 628 0x40 0x5
[    3.570552] [drm:drm_setup_crtcs] 
[    3.570560] [drm:drm_enable_connectors] connector 78 enabled? yes
[    3.570567] [drm:drm_target_preferred] looking for cmdline mode on connector 78
[    3.570573] [drm:drm_target_preferred] looking for preferred mode on connector 78 0
[    3.570578] [drm:drm_target_preferred] found mode 800x600
```

mcerveny, 2017-07-07:
Ping.
Do you have patch for that ?
Is anyone working on it ?

markyzq, 2017-07-10:
seems rk3288 need a similar modify to commit d072a98, get accurate clock for hdmi.
commit d072a98 is for rk3368

2) check modes to "fail-safe" framebuffer size (1024x768) and not to maximum available
the problem is drm framework limit, when there is no connector plug, drm framework hard code to use 1024x768 framebuffer size.
drivers/gpu/drm/drm_fb_helper.c
    sizes.fb_width = sizes.surface_width = 1024;
    sizes.fb_height = sizes.surface_height = 768;

You can custom modify it for you product, or if we can solved issue, this problem would disappear.

mcerveny, 2017-07-10:
I tried this:
```
diff --git a/drivers/clk/rockchip/clk-rk3288.c b/drivers/clk/rockchip/clk-rk3288.c
index 7ceee5c..47352ef 100644
--- a/drivers/clk/rockchip/clk-rk3288.c
+++ b/drivers/clk/rockchip/clk-rk3288.c
@@ -179,6 +179,7 @@ PNAME(mux_aclk_cpu_src_p)   = { "cpll_aclk_cpu", "gpll_aclk_cpu" };
 PNAME(mux_pll_src_cpll_gpll_p)         = { "cpll", "gpll" };
 PNAME(mux_pll_src_npll_cpll_gpll_p)    = { "npll", "cpll", "gpll" };
 PNAME(mux_pll_src_cpll_gpll_npll_p)    = { "cpll", "gpll", "npll" };
+PNAME(mux_pll_src_dmycpll_dmygpll_npll_p)      = { "dymmy_cpll", "dummy_gpll", "npll" };
 PNAME(mux_pll_src_cpll_gpll_usb480m_p) = { "cpll", "gpll", "usbphy480m_src" };
 PNAME(mux_pll_src_cpll_gll_usb_npll_p) = { "cpll", "gpll", "usbphy480m_src", "npll" };
 
@@ -428,7 +429,7 @@ static struct rockchip_clk_branch rk3288_clk_branches[] __initdata = {
                        RK3288_CLKSEL_CON(30), 14, 2, MFLAGS, 8, 5, DFLAGS,
                        RK3288_CLKGATE_CON(3), 4, GFLAGS),
 
-       COMPOSITE(DCLK_VOP0, "dclk_vop0", mux_pll_src_cpll_gpll_npll_p, 0,
+       COMPOSITE(DCLK_VOP0, "dclk_vop0", mux_pll_src_dmycpll_dmygpll_npll_p, CLK_SET_RATE_PARENT,
                        RK3288_CLKSEL_CON(27), 0, 2, MFLAGS, 8, 8, DFLAGS,
                        RK3288_CLKGATE_CON(3), 1, GFLAGS),
        COMPOSITE(DCLK_VOP1, "dclk_vop1", mux_pll_src_cpll_gpll_npll_p, 0,
```

But kernel crash:
```
[    3.377843] Unable to handle kernel NULL pointer dereference at virtual address 00000000
...
[    3.377912] PC is at clk_hw_round_rate+0x28/0x54
...
[    3.379627] [<c08bb8cc>] (clk_hw_round_rate) from [<c08c0c5c>] (divider_round_rate+0x280/0x3b0)
[    3.379640] [<c08c0c5c>] (divider_round_rate) from [<c08c0e20>] (clk_divider_round_rate+0x94/0x9c)
[    3.379653] [<c08c0e20>] (clk_divider_round_rate) from [<c08c1d4c>] (clk_composite_determine_rate+0x128/0x1ec)
[    3.379667] [<c08c1d4c>] (clk_composite_determine_rate) from [<c08bb818>] (clk_core_round_rate_nolock+0x58/0xb4)
[    3.379680] [<c08bb818>] (clk_core_round_rate_nolock) from [<c08bdc74>] (clk_round_rate+0x48/0x68)
[    3.379697] [<c08bdc74>] (clk_round_rate) from [<c04d34d0>] (vop_crtc_mode_valid+0x50/0x80)
[    3.379716] [<c04d34d0>] (vop_crtc_mode_valid) from [<c04c983c>] (dw_hdmi_rockchip_mode_valid+0x138/0x15c)
[    3.379732] [<c04c983c>] (dw_hdmi_rockchip_mode_valid) from [<c04e26c8>] (dw_hdmi_connector_mode_valid+0x28/0x34)
[    3.379750] [<c04e26c8>] (dw_hdmi_connector_mode_valid) from [<c04998d4>] (drm_helper_probe_single_connector_modes_merge_bits+0x36c/0x448)
[    3.379767] [<c04998d4>] (drm_helper_probe_single_connector_modes_merge_bits) from [<c04999cc>] (drm_helper_probe_single_connector_modes+0x1c/0x20)
[    3.379781] [<c04999cc>] (drm_helper_probe_single_connector_modes) from [<c04a2164>] (drm_fb_helper_probe_connector_modes+0x58/0x68)
[    3.379794] [<c04a2164>] (drm_fb_helper_probe_connector_modes) from [<c04a4540>] (drm_fb_helper_hotplug_event+0x9c/0xe8)
[    3.379807] [<c04a4540>] (drm_fb_helper_hotplug_event) from [<c04d0684>] (rockchip_drm_output_poll_changed+0x28/0x2c)
[    3.379821] [<c04d0684>] (rockchip_drm_output_poll_changed) from [<c0499288>] (drm_kms_helper_hotplug_event+0x34/0x38)
[    3.379835] [<c0499288>] (drm_kms_helper_hotplug_event) from [<c0499bc4>] (drm_helper_hpd_irq_event+0x128/0x15c)
[    3.379847] [<c0499bc4>] (drm_helper_hpd_irq_event) from [<c04e2d40>] (repo_hpd_event+0x24/0x28)
[    3.379859] [<c04e2d40>] (repo_hpd_event) from [<c01347b4>] (process_one_work+0x234/0x418)
[    3.379872] [<c01347b4>] (process_one_work) from [<c0135558>] (worker_thread+0x2cc/0x40c)
[    3.379884] [<c0135558>] (worker_thread) from [<c013a0e8>] (kthread+0x100/0x118)
[    3.379898] [<c013a0e8>] (kthread) from [<c0107378>] (ret_from_fork+0x14/0x3c)
```

mcerveny, 2017-07-10:
I temporary resolve this issue by workaround - https://tinkerboarding.co.uk/forum/thread-321-post-1350.html#pid1350

markyzq, 2017-07-11:
apply attach patch, and assigned dclk on dtsi, then may solved your problem

assigned-clocks = <&cru DCLK_VOP0>;
assigned-clock-parents = <&cru PLL_NPLL>;

markyzq, 2017-07-11:
[rk3288_npll_for_vop0.patch.txt](https://github.com/rockchip-linux/kernel/files/1137491/rk3288_npll_for_vop0.patch.txt)

mcerveny, 2017-07-11:
Thanks, but partially working. 
Some resolution (NPLL/DCLK_VOP0 clock) does not work at all (blank output) or blinking (few seconds show and then resync - probably PLL lost sync ?). 
**Working NPLL** clocks:
65000 kHz
78750 kHz
135000 kHz
**NOT working NPLL** clocks:
1080000 kHz (this works in my workaround perfectly eg. NPLL 216000000 and divide by 2 to get DCLK_VOP0)
75000 kHz 
I suppose that PLL should use some frequency that is multiply of requested DCLK_VOP0.  
I cannot view problem on TMDS clock pins with my oscilloscope to verify problem - try to test clock in your HW lab.

Test:
```
# egrep  'npll|dclk_vop0' /sys/kernel/debug/clk/clk_summary
    pll_npll                              1            1   108000000          0 0  
       npll                               1            1   108000000          0 0  
          dclk_vop0                       2            3   108000000          0 0  
```

mcerveny, 2018-01-24:
Kwiboo/linux-rockchip@580da0d8928801b3e5cf162ca3ba6fee7e3c198c fixes rk3328 platform for me (not using rk3288 any more).

mcerveny, 2018-01-27:
Closing the issue due to lost confidence in Rockchip solution.

closed 2018-01-27 by mcerveny