rockchip-linux/kernel #12: DRM: HDMI CEC support

mcerveny, 2017-04-29:
Hello.
I do not known how long this branch will be supported but some HDMI CEC support should be added to code. I backported and corrected CEC support from UPSTREAM (4.10+) and from FROMLIST patches (level RFCv2) of media tree (but development of this API is very active and planned to 4.12+). 

Please check situation and do decision.

See: http://tinkerboarding.co.uk/forum/thread-258-post-687.html#pid687
See: https://git.linuxtv.org/hverkuil/media_tree.git/log/?h=cec-notifiers
See: https://patchwork.kernel.org/project/linux-media/list/?submitter=520
See: https://github.com/mcerveny/kernel/tree/release-4.4-cec
Attached patches too: [tinker-board-cec-patch-20170428.zip](https://github.com/rockchip-linux/kernel/files/965947/tinker-board-cec-patch-20170428.zip)

Thanks, Martin

phhusson, 2017-10-07:
FWIW, CEC framework and cec on dw-hdmi support has been mainlined.
It now integrates the support for 4-bytes words, and it fixed the various EDID problems yo uhad.

I've enabled and tested CEC on rk3399-firefly with the following patch:
https://gist.github.com/phhusson/fffeb89ae7b179d685a9386b9dd44a1c

I'm afraid you'll still need your gpio7c0/gpio7c7 hack though.
(On the rk3399 there is only one CEC pin, so no problem)

phhusson, 2017-10-16:
gpio7c0/gpio7c7 hack should no longer be needed with https://patchwork.kernel.org/patch/10006451/

mcerveny, 2018-01-24:
Series of commits from [UPSTREAM](https://github.com/rockchip-linux/kernel/search?q=cec&type=Commits).

mcerveny, 2018-01-26:
I successfully tested CEC (Rock64/RK3328). There is only need to enable CEC in kernel config (mcerveny/rockchip-linux@ce35f3782390d8196166359aa172e71a6e8a5b16) and test with utility cec-compliance/cec-ctl (apt-get install v4l-utils (1.12.3-1)). 
**Compliance test:**
```
# ## There is TV set and BD-player on CEC bus
# cec-compliance -v -T
		CEC_ADAP_G_CAPS returned 0 (Success)
cec-compliance SHA                 : not available

Driver Info:
	Driver Name                : dwhdmi-rockchip
	Adapter Name               : dw_hdmi
	Capabilities               : 0x0000000e
		Logical Addresses
		Transmit
		Passthrough
	Driver version             : 4.4.103
	Available Logical Addresses: 4
		CEC_ADAP_G_PHYS_ADDR returned 0 (Success)
	Physical Address           : 2.0.0.0
		CEC_ADAP_G_LOG_ADDRS returned 0 (Success)
	Logical Address Mask       : 0x0100
	CEC Version                : 2.0
	Vendor ID                  : 0xaabbcc
	Logical Addresses          : 1 (Allow RC Passthrough)

	  Logical Address          : 8
	    Primary Device Type    : Playback
	    Logical Address Type   : Playback
	    All Device Types       : Playback
	    RC TV Profile          : None
	    Device Features        :
		None

Compliance test for device /dev/cec0:

    The test results mean the following:
        OK                  Supported correctly by the device.
        OK (Not Supported)  Not supported and not mandatory for the device.
        OK (Presumed)       Presumably supported.  Manually check to confirm.
        OK (Unexpected)     Supported correctly but is not expected to be supported for this device.
        OK (Refused)        Supported by the device, but was refused.
        FAIL                Failed and was expected to be supported by this device.

Find remote devices:
		CEC_TRANSMIT returned 0 (Success)
		CEC_TRANSMIT returned 0 (Success)
		CEC_TRANSMIT returned 0 (Success)
		CEC_TRANSMIT returned 0 (Success)
		CEC_TRANSMIT returned 0 (Success)
		CEC_TRANSMIT returned 0 (Success)
		CEC_TRANSMIT returned 0 (Success)
		CEC_TRANSMIT returned 0 (Success)
		CEC_TRANSMIT returned 0 (Success)
		CEC_TRANSMIT returned 0 (Success)
		CEC_TRANSMIT returned 0 (Success)
		CEC_TRANSMIT returned 0 (Success)
		CEC_TRANSMIT returned 0 (Success)
		CEC_TRANSMIT returned 0 (Success)
		CEC_TRANSMIT returned 0 (Success)
	Polling: OK

Network topology:
	System Information for device 0 (TV) from device 8 (Playback Device 2):
		CEC Version                : 		CEC_TRANSMIT returned 0 (Success)
1.4
		Physical Address           : 		CEC_TRANSMIT returned 0 (Success)
0.0.0.0
		Primary Device Type        : TV
		Vendor ID                  : 		CEC_TRANSMIT returned 0 (Success)
0x00e091
[... 170 lines cut ...]
Received from TV to Playback Device 2 (0 to 8): CEC_MSG_USER_CONTROL_RELEASED (0x45)
		CEC_RECEIVE returned 0 (Success)
	ui-cmd: F2 (Red) (0x72)
		CEC_RECEIVE returned 0 (Success)
Received from TV to Playback Device 2 (0 to 8): CEC_MSG_USER_CONTROL_RELEASED (0x45)
		CEC_RECEIVE returned 0 (Success)
Received from TV to Playback Device 2 (0 to 8): CEC_MSG_USER_CONTROL_PRESSED (0x44):
	ui-cmd: F3 (Green) (0x73)
		CEC_RECEIVE returned 0 (Success)
Received from TV to Playback Device 2 (0 to 8): CEC_MSG_USER_CONTROL_RELEASED (0x45)
		CEC_RECEIVE returned 0 (Success)
Received from TV to Playback Device 2 (0 to 8): CEC_MSG_USER_CONTROL_PRESSED (0x44):
	ui-cmd: F4 (Yellow) (0x74)
		CEC_RECEIVE returned 0 (Success)
Received from TV to Playback Device 2 (0 to 8): CEC_MSG_USER_CONTROL_RELEASED (0x45)
		CEC_RECEIVE returned 0 (Success)
Received from TV to Playback Device 2 (0 to 8): CEC_MSG_USER_CONTROL_PRESSED (0x44):
	ui-cmd: F1 (Blue) (0x71)
		CEC_RECEIVE returned 0 (Success)
Received from TV to Playback Device 2 (0 to 8): CEC_MSG_USER_CONTROL_RELEASED (0x45)
		CEC_RECEIVE returned 0 (Success)
Received from TV to Playback Device 2 (0 to 8): CEC_MSG_USER_CONTROL_PRESSED (0x44):
	ui-cmd: Back (0x0d)
		CEC_RECEIVE returned 0 (Success)
```

closed 2018-01-26 by mcerveny

RosiersRobin, 2023-02-08:
Maybe a stupid question, but I currently have a mini pc, that is running a rockchip kernel and debian. However, I want CEC (hardware should have it). Reading this article, it seems like it's it the current kernel (I run 5.10.110) Debian GNU/Linux 11 (bullseye) but I don't have a `/etc/cec0` bus, how can I have this? How can I enable this? I don't have the ability to re-compile from source and then flash.