rockchip-linux/kernel #13: The 4.4.55 kernel does not start on RK3288, in contrast to the 4.4.52 kernel

pro777s, 2017-04-29:
Hello, 
The 4.4.55 kernel does not start on RK3288, in contrast to the 4.4.52 kernel :(

wzyy2, 2017-04-29:
Which board do you use? Any log?
There is no many rk3288 changes between 4.4.55 and 4.4.52.

Maybe you could check rootfs mmc index in kernel cmdline. it might be changed in 4.4.55.

pro777s, 2017-04-30:
I'm use XT-Q8L-V10 RK3288 board. I'm load Ubuntu 16.04 from SD-card. My kernel command line: 
CMDLINE:console=ttyS2,115200 console=tty0 earlyprintk root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 init=/sbin/init initrd=0x62000000,0x00800000 mtdparts=rk29xxnand:0x00010000@0x00002000(boot),-@0x00012000(linuxroot)
With Rockchip kernel 4.4.52 all fine load.

wzyy2, 2017-05-02:
It's better you can attach serial ports, or try revert this commit https://github.com/rockchip-linux/kernel/commit/44f988df9d08873f0e3fd19adb7c062449db9bf3.

pro777s, 2017-05-03:
I did the recommended revert of the commit, but this, unfortunately, did not lead to success. In the box there are no conclusions of COM port, I'll try to think of something.

pro777s, 2017-05-31:
May 28 I compiled the kernel 4.4.66. The result is the same: the kernel does not start.
I attach console output from the UART port.
[rk3288.txt](https://github.com/rockchip-linux/kernel/files/1042785/rk3288.txt)

eldonT, 2017-06-19:
@pro777s 
can you tell me which dts config you used to build your working q8 kernel (Myy and rk builds) ?
My understanding is that the onboard dtb is not compatible with 4.x kernel ?
thx

pro777s, 2017-06-19:
All right. Q8 successfully starts with DTS from firefly or miqi.

closed 2017-07-24 by wzyy2