rockchip-linux/kernel #14: Firefly-RK3288 Linux Kernel

sghazagh, 2017-05-09:
Sorry I am writing this question here as an issue.
I have a Firefly-RK3288 board. It's dual boot with android so recovery partition is used for linux kernel and at the moment official kernel 3.10.0 is compiled and loaded.

Can I use "kernel" in this repository to update only my Linux kernel on recovery partition of my Firefly-RK3288 board?
Please advise if boot partition (u-boot) needs to be updated as well.

Can you please give me the link to the best guide that I can use?
It seems that it is not normal kernel and dts file copy and I apparently have to make a kernel image with different format compatible with Firefly-rk3288 partitioning, correct?

Thanks for your assistance,

wzyy2, 2017-05-09:
You can find guides in below link.
http://opensource.rock-chips.com/wiki_Main_Page

sghazagh, 2017-05-09:
Thank you,
By the way, the part confuses me is "Kernel Build"!

Is Kernel Build, builds the kernel for Linux or for Android?
As I explained my board has dual boot which main OS in Android.
So it uses "Recovery" partition as boot partition for linux. In fact the Linux Kernel needs to go to "Recovery" partition.
So the "Building kernel" section of the guide, creates the kernel for Linux or for Android?

wzyy2, 2017-05-10:
There is no guide for booting kernel from old u-boot. I can show you a script which we use to ceate kernel image for old u-boot, but i'm not sure if it's suitable.
https://github.com/rockchip-linux/meta-rockchip/blob/morty/classes/rockchip-update-img.bbclass

I don't how your dual boot is implemented and I'm afraid you have to study it by yourself.

sghazagh, 2017-05-10:
Thanks wzyy2 for your help.
I actually started to create a SD image from scratch by creating kernel, u-boot and linaro-rootfs.

So fa so good.
I have created the system image and trying to write it to SD.
It takes long time and still has not finished writing to SD. Not sure if it's right or wrong.

If that works, I stick to this repository for all Linux rootfs needs for my Firefly.

I do suggest to make a wiki a little bit more clear at the level explaining about SD images and making the SD image.

I will let you the result if it works fine.

I'll forget about the dual boot as I believe it will be more hassle while I can make a working board just with Linux....

will let you  know the result ans will ask if I need any question.

I do appreciate your help...

sghazagh, 2017-05-10:
No mate,
ubfortunatly the SD image does  not boot the Firefly.

Do I need to load u-boot or something else in device itself first?

Mine starts reading SD but cannot find the Partition tags.
Apparently it still looking for my old partition from partition table from eMMC and not SD itself?

Any advise?

wzyy2, 2017-05-10:
erase emmc

sghazagh, 2017-05-10:
All eMMC or only boot, kernel, u-boot? What about bootloader? how the board boots then?

What command do you use?
I only know "rkflashkit erase" which only erase partitions!

wzyy2, 2017-05-11:
only loader1
You can use `rkdeveloptool wl 64(loader1 offset) dirty_content` to erase it

sghazagh, 2017-05-11:
Jacob,
In official Firefly-rk3288 firmware, there is no loader1 partition.
The partition system named differently.
For example it has boot, misc, kernel, recovery...
Let me listthe partition file for your better understanding:

```
#Created By Firefly RK3288 'Parameter' File Generator 3.00
#ELAR Systems (http://www.elar-systems.com)
#Written By: Saeid Ghazagh (June 2016)
###########################################################
FIRMWARE_VER:4.4.2
MACHINE_MODEL: rk30sdk
MACHINE_ID:007
MANUFACTURER: RK30SDK
MAGIC: 0x5041524B
ATAG: 0x60000800
MACHINE: 3066
CHECK_MASK: 0x80
PWR_HLD: 0,0,A,0,1
#KERNEL_IMG: 0x62008000
#FDT_NAME: rk-kernel.dtb
#RECOVER_KEY: 1,1,0,20,0
CMDLINE:console=tty0 console=ttyS2 earlyprink androidboot.hardware=rk30board androidboot.console=ttyS2 board.ap_has_alsa=0 root=/dev/block/mtd/by-name/linuxroot rw rootfstype=ext4 init=/sbin/init initrd=0x62000000,0x00800000 mtdparts=rk29xxnand:0x00002000@0x00002000(uboot),0x00002000@0x00004000(misc),0x00008000@0x00006000(resource),0x00008000@0x0000E000(kernel),0x00010000@0x00016000(boot),0x00010000@0x00026000(recovery),0x0001A000@0x00036000(backup),0x00040000@0x00050000(cache),0x00002000@0x00090000(kpanic),0x00180000@0x00092000(system),0x00002000@0x00212000(metadata),0x00200000@0x00214000(userdata),0x00620000@0x00414000(linuxroot),-@0x00A34000(user)
```

So my question is that, how the boeard canuse SD with new partitioning as explaind in this repository?
Should I wipe whole eMMC and reload the system.img to eMMC first?
If not, what is the best approach to leave eMMC as what it is and just use SD to boot the Firefly with SD partitioning text file?

wzyy2, 2017-05-11:
Ok.... Forget those boot things, it will confuse you..

**You should look at firefly wiki for how to boot into sdcard . I think they must have provided guides or tools to help.**

sghazagh, 2017-05-11:
I can boot from SD to the Linux image I have created myself.
It actually uses same concept but I still use uboot, resource, kernel, misc and linuxrootfs partition.
So I know how it boots to SD. My Ubuntu 16.04 boots ok when I insert in SD.
The problem is that , I have no source for Kernel to update my version thats why I want to follow this repository and stick to this partitioning and setup.

Let me try by erasing the eMMC and load the loade1 to eMMC. I belive that is required to find the rootfs as it acts as first bootloader. That should then find the rootfs from SD and boot it.

Will let you know the result...

wzyy2, 2017-05-11:
How do you boot from SD-Card without press maskrom button or earse emmc?
http://rockchip.wikidot.com/boot-sequence

Firefly board only support sd-card boot in mainline u-boot, so you don't need to try eMMC boot.

sghazagh, 2017-05-11:
I give up mate...  even cannot erase the eMMc.
command "rkdeveloptool EF" does not work and says device do not support!

thanks for your support. I think it's better I stick to my current setup.

Hopefullt this repository works on my Asus Tinkerboard which is on the way to arrive....

wzyy2, 2017-05-11:
Close it ....

I suggest you to google some keywords like  "firefly mainline u-boot" to boot mainline u-boot on your board at first....

closed 2017-05-11 by wzyy2

sghazagh, 2017-05-11:
I am using the mainline u-boot base on this and all is ok to load my both android and linux.
Also can load my own linux image from SD.

I don't know how I have to fit this repository system.img into eMMC or SD and how Firefly can boot them?

Hope anyone can help... I think it's better to keep it open as I as a user of Firefly-RK3288 couldn't be able to use this GIT Repository to boot my device!

wzyy2, 2017-05-11:
Are you sure you are using mainline u-boot?

http://git.denx.de/?p=u-boot.git;a=summary
https://github.com/u-boot/u-boot/blob/master/doc/README.rockchip

geekerlw, 2017-05-11:
you can change the kernel like firefly in this branch : https://github.com/T-Firefly/kernel_4.4/tree/firefly-rk3288  , just change some file like this commit : https://github.com/T-Firefly/kernel_4.4/commit/c2d8092ec7444cf11aa8ad6883bd4c7d3c5f74fd?diff=unified#diff-c085a0b86214742198c6dc024bb476efL379   , maybe you can get the kernel.img firefly use

wzyy2, 2017-05-11:
> you can change the kernel like firefly in this branch : https://github.com/T-Firefly/kernel_4.4/tree/firefly-rk3288 , just change some file like this commit : T-Firefly/kernel_4.4@c2d8092?diff=unified#diff-c085a0b86214742198c6dc024bb476efL379 , maybe you can get the kernel.img firefly use

I think it's not enough to boot.
Kernel config also need be changed to use kernel.img.  e.g: CONFIG_ARM_APPENDED_DTB,  CONFIG_CMDLINE

wzyy2, 2017-05-11:
> I give up mate... even cannot erase the eMMc.
command "rkdeveloptool EF" does not work and says device do not support!
thanks for your support. I think it's better I stick to my current setup.
Hopefullt this repository works on my Asus Tinkerboard which is on the way to arrive....

You miss a key word : **MASKROM mode**
That's why you are stick ...
**rkdeveloptool are used in maskrom mode.**

Please read blow link to know what is loader mode, what is maskrom mode, why it need enter maskrom mode, why it need erase emmc.
http://opensource.rock-chips.com/wiki_Rockusb
http://rockchip.wikidot.com/boot-sequence
http://opensource.rock-chips.com/wiki_Boot_option
http://wiki.t-firefly.com/index.php/Firefly-RK3288/Boot_mode/en
https://github.com/u-boot/u-boot/blob/master/doc/README.rockchip#L65

sghazagh, 2017-05-11:
Jacob,
Many thanks for documents links. I read all of them and now better understand this repository Linux and all it's components like u-boot,kernel...

Firefly official uses different partitioning and tools but here is straight forward too.
I needed to dig into all the scripts to see what is happening to understand them.

Very good view...
Thanks a bunch ;)

sghazagh, 2017-05-12:
Jacob/All,
Yesterday I didn't know much about this repository and it's RK3288 images and partitions.

Today, I was able to load my own created Ubuntu 16.4.2 LTS from eMMC with kernel 4.4.55 from this repository Kernel source, plus able to boot from both eMMC and SD.(boots from SD if it presents otherwise loads from eMMC)

I would like to thank you all for providing the Documentation.
It helped a lot.

Thank you