rockchip-linux/kernel #15: pinctrl: RK3288 different GPIO pin numbers

mcerveny, 2017-05-20:
Hello.

This kernel using different pin numbers for GPIO (due to commit 1d9964a) than other kernels (TinkerOS, upstream...).
Problem is [here](https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/pinctrl/pinctrl-rockchip.c#L3006) "PIN_BANK_IOMUX_FLAGS(0, **32**, "gpio0",..." but in other kernels [here](https://github.com/TinkerBoard/debian_kernel/blob/linux4.4-rk3288/drivers/pinctrl/pinctrl-rockchip.c#L2885) or [here](https://github.com/torvalds/linux/blob/master/drivers/pinctrl/pinctrl-rockchip.c#L2843) "PIN_BANK_IOMUX_FLAGS(0, **24**, "gpio0"...).
This kernel: GPIO5_B2 = 5\*32 + ("B") 1\*8 + 2 = **gpio170**
Other kernels:  GPIO5_B2 = 5\*32 + ("B") 1\*8 + 2 **- 8** = **gpio162** (subtract 8 for >gpio24)
This leads to incompatibility of "/sys/class/gpio/*" and libraries for [TinkerBoard](https://www.asus.com/uk/Single-Board-Computer/Tinker-Board/HelpDesk_Download/) GPIO_API_for_C and GPIO_API_for_Python.

How to resolve this, will be pushed to upstream ?

Thanks, Martin

wzyy2, 2017-05-22:
It seems GPIO0_D0 ~ GPIO0_D7  are in TRM, not sure why they don't add it.

@mmind 
https://github.com/rockchip-linux/kernel/commit/304f077d4c07d315b9325cb101fc47ba2ffc5466#diff-3635f39427067fc9fdb5c98648e2e1a5R1813

wzyy2, 2017-06-15:
It's a mistake and it will be deleted.

closed 2017-06-15 by wzyy2