rockchip-linux/kernel #19: SPI: random data length/tx/rx for DMA SPI transfers (eg. size over 32 bytes) "CONFIG_SPI_ROCKCHIP"

mcerveny, 2017-07-03:
SPI driver CONFIG_SPI_ROCKCHIP (drivers/spi/spi-rockchip.c) does not work for transfers over 32 bytes (switched to DMA transfers, [see rockchip_spi_can_dma()](https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/spi/spi-rockchip.c#L631)). Totally random length/rx/tx are send and received with rk3288 (TinkerBoard, TinkerOS 1.8). Checked with external protocol analyzer and testing code:
```
#include <string.h>
#include <unistd.h>
#include <stdlib.h>
#include <stdio.h>
#include <fcntl.h>
#include <inttypes.h>
#include <linux/spi/spidev.h>
#include <linux/types.h>
#include <sys/ioctl.h>
#include <linux/ioctl.h>

// SZ <= 32 OK (PIO transfers), >32 radom length send, random data send (checked from logic analyzer), random data received
#define SZ 33

int main()
{
	int fd = open("/dev/spidev2.0", O_RDWR);

        struct spi_ioc_transfer xfer;
	memset(&xfer, 0, sizeof(xfer));

	uint8_t tx_buf[SZ] = "\03\0\0\0";  // read from SPI flash
        uint8_t rx_buf[SZ];

	xfer.tx_buf = (unsigned long)&tx_buf;
	xfer.rx_buf = (unsigned long)&rx_buf;
	xfer.len = sizeof(rx_buf);
	xfer.speed_hz = 10000000;
	xfer.bits_per_word = 8;

	int status = ioctl(fd, SPI_IOC_MESSAGE(1), &xfer);

	printf("status %d\n", status);
	int i;
	for(i=0; i<sizeof(rx_buf); i++) printf("%02x ", rx_buf[i]);
	printf("\n");

	close(fd);

	return 0;
}
```

wzyy2, 2017-07-03:
Coud you revert this commit and test again?
https://github.com/rockchip-linux/kernel/commit/e0a1e9b7cded261847ac972311082724afb75cbd

mcerveny, 2017-07-03:
Not resolved by revert (tested on https://github.com/TinkerBoard/debian_kernel). Even worse than previous state. The kernel became corrupted after few runs (~3, with patch need ~20-30 - tested again) and all application are failing and kernel finally hangs (usually "**Internal error: Oops: 37 [#<nolink>1] SMP ARM**").

Can anyone describe differences to other "DW SPI" code in https://github.com/rockchip-linux/kernel/tree/release-4.4/drivers/spi (eg. spi-rockchip-* and spi-dw* by intel) ?

Tonymac32, 2017-07-17:
I believe this may also be an error on Rock64 at the present time:

https://forum.armbian.com/index.php?/topic/4583-rock64/&do=findComment&comment=34978

wzyy2, 2017-07-24:
> I believe this may also be an error on Rock64 at the present time:
https://forum.armbian.com/index.php?/topic/4583-rock64/&do=findComment&comment=34978

It's fixed.

closed 2017-07-24 by wzyy2

mgilge, 2017-08-14:
Did you do the disable DMA workaround in the rockchip-linux or the tinkerboard-linux? What is the difference between the two anyway?

mcerveny, 2017-08-14:
Workaround was tested rockchip-linux and should work with all forks. Tinkerboard linux was forked from rockchip-linux (probably branch "release-4.4-miniarm" sometimes last year). ASUS TinkerBoard team is ignoring any development in rockchip-linux (do not rebase or merge patches - compare with https://github.com/TinkerBoard/debian_kernel). TinkerBoard is dead for me (check [tinkerboard forum](https://tinkerboarding.co.uk/forum/thread-440-post-1386.html#pid1386)).
I did not find or test announced fix.

mgilge, 2017-08-14:
Thanks for getting back to me. We just use the Tinkerboard as a server platform.
But we need the SPI in order to connect a status display (paper = PaPiRus).

Is there a possibility to exchange just the patched SPI-driver on our SD-card without recompiling the kernel?

geneshuman, 2019-04-25:
I'm a bit confused.  Was this issue ever resolved?  I seem to be running into it as well, although maybe I'm just running an old kernel.