rockchip-linux/kernel #20: Tinker Board KErnel Modules

sghazagh, 2017-07-10:
Hi,
I have tried to test the kernel on this repository for ASUS Tinker Board.
I have compiled the kernel and modules and boot the device with created kernel.

The kernel was load the device successfully however, modules does not load at all as this repository do not create all the required modules.

Device Tree seems different compare to official ASUS DTB file for miniarm.
Cn you please advise if modules like WIFI and Bluetooth are included in this repository or not?

Thanks

wzyy2, 2017-07-10:
WIFI and Bluetooth driver for tinker board are not included in main 4.4 kernel branch.

https://github.com/rockchip-linux/kernel/tree/release-20160818-miniarm
You can try this branch.

Tonymac32, 2017-07-19:
If that branch does not work for your needs, the Tinkerboard 4.4.76 kernel at Armbian.com has Bluetooth and rtl8723bs driver 4.4.0, as well as the Rockchip reboot fix.  The build environment allows for creating user patches for any other adjustments you may wish to make.  I follow the patches here and on the Tinker Board repo for any fixes.

wzyy2, 2017-07-22:
> If that branch does not work for your needs, the Tinkerboard 4.4.76 kernel at Armbian.com has Bluetooth and rtl8723bs driver 4.4.0, as well as the Rockchip reboot fix. The build environment allows [...]

I will rebase release-20160818-miniarm branch to the latest release-4.4 and re-organiza the code to make it easy to merge release-4.4 branch.

Maybe armbian could switch to that branch.

Tonymac32, 2017-07-22:
Thanks for the update, Armbian tries to use a single kernel source with patches to support all devices using a particular SoC.  Switching to the Rockchip miniarm branch would most likely break the MiQi, or we'd have to have an entirely separate config and patchset for it.

Tonymac32, 2017-07-23:
Also, I've had very good success (as has @MiouYouYou) running Mainline kernel on RK3288 devices, the patches for Mainline are primarily dts and Mali.

Miouyouyou, 2017-07-23:
Yeah, we're only missing bluetooth and VPU (well, I didn't the test the VPU packages ATM) with the Mainline kernel.

wzyy2, 2017-07-24:
> Yeah, we're only missing bluetooth and VPU (well, I didn't the test the VPU packages ATM) with the Mainline kernel.

You can look at this repo for VPU driver.
https://github.com/rockchip-linux/rockchip_forwardports

I remember 4K HDMI is also missing, maybe you folks can have a look. ; ).

Tonymac32, 2017-07-24:
Haha yes, 4k is missing as well, I was focusing on the reboot thing first.  Thanks for the link

Miouyouyou, 2017-07-24:
I'll take a look at the VPU, yeah !

I also take generous 4K screens donations ! 
＼(・ヮ・)／

Meanwhile, I don't have 4K screens so testing this will be fairly difficult (´・ω・`).

The 4.13.0-rc2 update should be available very soon. This update includes various DTS/DTSI fixes and the reboot patch. Turns out I missed regulator-always-on and regulator-boot-on properties on an SD vcc node.

Tonymac32, 2017-07-24:
I have a 4k screen, I was looking at some of the code, let me get the Armbian Dev  up to 4.13 first.  ;-)

Caesar-github, 2019-04-19:
Obsolete issue

closed 2019-04-19 by Caesar-github