rockchip-linux/kernel #21: Starting kernel ... hang

Pillar1989, 2017-07-10:
你好，我使用RK3229 echo板。
[image]
启动的时候到Starting kernel ...就挂了。下面是具体的log:
```
SPL Inittimer init done
rk322x_dmc_probe
ddr clk dpll
ret=600000000
sdram_all_config bw 2 col 10
sdram_all_config reg 110005d0 6281
Returning to boot ROM...

U-Boot 2017.05-g6c615f4-dirty (Jul 10 2017 - 10:03:48 +0800)

Model: Rockchip RK3229 Evaluation board
DRAM:  rk322x_dmc_probe
rockchip_sdram_size 110005d0 6281
rank 1 col 10 bk 3 cs0_row 15 bw 2 row_3_4 0
size 40000000                                                                   
SDRAM base=60000000, size=40000000                                              
992 MiB                                                                         
MMC:   dwmmc@30020000: 0                                                        
*** Warning - bad CRC, using default environment                                
                                                                                
In:    serial@11030000                                                          
Out:   serial@11030000                                                          
Err:   serial@11030000                                                          
Net:   Net Initialization Skipped                                               
No ethernet found.                                                              
Hit any key to stop autoboot:  0                                                
                                                                                
MMC read: dev # 0, block # 32768, count 20480 ... 20480 blocks read: OK         
Wrong Image Format for bootm command                                            
ERROR: can't get kernel image!                                                  
=> run bootcmd_mmc0                                                             
switch to partitions #0, OK                                                     
mmc0(part 0) is current device                                                  
Scanning mmc 0:6...                                                             
Found /extlinux/extlinux.conf                                                   
Retrieving file: /extlinux/extlinux.conf                                        
reading /extlinux/extlinux.conf                                                 
170 bytes read in 8 ms (20.5 KiB/s)                                             
1:      kernel-4.4                                                              
Retrieving file: /zImage                                                        
reading /zImage                                                                 
7340608 bytes read in 5229 ms (1.3 MiB/s)                                       
append: earlyprintk console=ttyFIQ0,115200n8 rw root=/dev/mmcblk0p7 rootfstype=t
Retrieving file: /rk3229-echo-v10.dtb                                           
reading /rk3229-echo-v10.dtb                                                    
26513 bytes read in 28 ms (923.8 KiB/s)                                         
## Flattened Device Tree blob at 61f00000                                       
   Booting using the fdt blob at 0x61f00000                                     
   Loading Device Tree to 683f6000, end 683ff790 ... OK                         
                                                                                
Starting kernel ...  
```
这是我的extlinux.conf 
```
baozhu@server:~/studio/rockchip/rk-linux$ cat boot/extlinux/extlinux.conf 
label kernel-4.4
    kernel /zImage
    fdt /rk3229-echo-v10.dtb
    append  earlyprintk console=ttyFIQ0,115200n8 rw root=/dev/mmcblk0p7 rootfstype=ext4 init=/sbin/init

```
请帮忙看看这个问题，谢谢

eddiecailinux, 2017-07-10:
Hi, 请使用Rockchip提供的echo SDK for RK3229

Pillar1989, 2017-07-10:
我们之前在debian上面做了大量应用，目前我们急需在rk3229运行debian，快速验证软件和测试性能。而且我们团队对echo SDK的buildroot开发环境不熟悉，学习成本太高。
还是请你们帮忙看下这个问题，谢谢！

eddiecailinux, 2017-07-10:
请通过业务正式提出需求

keveryang, 2017-07-12:
kernel使用的是什么配置？
config试一下下面的内容，主要是把波特率更新一下．
append earlyprintk console=ttyS2,1500000n8 rw root=/dev/sda rootwait rootfstype=ext4 init=/sbin/init

Pillar1989, 2017-07-13:
@keveryang 
更新了config,目前的extlinux.conf为:
```
label kernel-4.4
    kernel /zImage
    fdt /rk3229-echo-v10.dtb
    append  earlyprintk console=ttyS2,1500000n8 rw root=/dev/mmcblk0p7 rootfstype=ext4 init=/sbin/init

```
启动的时候还是卡在这里：
```
U-Boot 2017.05-g6c615f4-dirty (Jul 13 2017 - 09:38:33 +0800)

Model: Rockchip RK3229 Evaluation board
DRAM:  rk322x_dmc_probe
rockchip_sdram_size 110005d0 6281
rank 1 col 10 bk 3 cs0_row 15 bw 2 row_3_4 0
size 40000000
SDRAM base=60000000, size=40000000
992 MiB
MMC:   dwmmc@30020000: 0
*** Warning - bad CRC, using default environment

In:    serial@11030000
Out:   serial@11030000
Err:   serial@11030000
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0 

MMC read: dev # 0, block # 32768, count 20480 ... 20480 blocks read: OK
Wrong Image Format for bootm command
ERROR: can't get kernel image!
=> run bootcmd_mmc0 
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:6...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
reading /extlinux/extlinux.conf
169 bytes read in 6 ms (27.3 KiB/s)
1:      kernel-4.4
Retrieving file: /zImage
reading /zImage
7340608 bytes read in 5228 ms (1.3 MiB/s)
append: earlyprintk console=ttyS2,1500000n8 rw root=/dev/mmcblk0p7 rootfstype=ext4 init=/sbin/init
Retrieving file: /rk3229-echo-v10.dtb
reading /rk3229-echo-v10.dtb
26513 bytes read in 26 ms (995.1 KiB/s)
## Flattened Device Tree blob at 61f00000
   Booting using the fdt blob at 0x61f00000
   Loading Device Tree to 683f6000, end 683ff790 ... OK

Starting kernel ...

```
下面是我用的https://github.com/rockchip-linux/build 脚本的配置信息，kernel用的是rockchip_linux_defconfig.
```
	"echo")
		DEFCONFIG=rockchip_linux_defconfig
		UBOOT_DEFCONFIG=evb-rk3229_defconfig
		DTB=rk3229-echo-v10.dts
		CHIP="rk3229"
		;;

```

keveryang, 2017-07-13:
kernel下面这几个配置尝试更新一下，原来是STRONG
 232 # CONFIG_CC_STACKPROTECTOR is not set
 233 CONFIG_CC_STACKPROTECTOR_NONE=y
 234 # CONFIG_CC_STACKPROTECTOR_REGULAR is not set
 235 # CONFIG_CC_STACKPROTECTOR_STRONG is not set

Pillar1989, 2017-07-13:
@keveryang 
我添加你说的配置：
```
baozhu@server:~/studio/rockchip/rk-linux/kernel$ git diff arch/arm/configs/rockchip_linux_defconfig
diff --git a/arch/arm/configs/rockchip_linux_defconfig b/arch/arm/configs/rockchip_linux_defconfig
index 0c7ec9e..e59ecb0 100644
--- a/arch/arm/configs/rockchip_linux_defconfig
+++ b/arch/arm/configs/rockchip_linux_defconfig
@@ -479,3 +479,8 @@ CONFIG_CRC7=y
 # CONFIG_XZ_DEC_POWERPC is not set
 # CONFIG_XZ_DEC_IA64 is not set
 # CONFIG_XZ_DEC_SPARC is not set
+
+# CONFIG_CC_STACKPROTECTOR is not set
+CONFIG_CC_STACKPROTECTOR_NONE=y
+# CONFIG_CC_STACKPROTECTOR_REGULAR is not set
+# CONFIG_CC_STACKPROTECTOR_STRONG is not set

```

但是还是卡在Starting kernel ...
能否提供一个测试过的config ?

wzyy2, 2017-07-13:
ttyfiq

Pillar1989, 2017-07-13:
是调试串口不对吗？？具体的是啥，这个串口是uart2

wzyy2, 2017-07-13:
console=ttyFIQ0,1500000n8
ttyS2被wrap成了ttyFIQ0 所以你会没串口输出

Pillar1989, 2017-07-13:
我现在换回了ttyFIQ0:
```
baozhu@server:~/studio/rockchip/rk-linux$ cat tmp/extlinux/extlinux.conf 
label kernel-4.4
    kernel /zImage
    fdt /rk3229-echo-v10.dtb
    append  earlyprintk console=ttyFIQ0,1500000n8 rw root=/dev/mmcblk0p7 rootfstype=ext4 init=/sbin/init

```
还是一样，卡在Starting kernel ..
```

U-Boot 2017.05-g6c615f4-dirty (Jul 13 2017 - 09:38:33 +0800)

Model: Rockchip RK3229 Evaluation board
DRAM:  rk322x_dmc_probe
rockchip_sdram_size 110005d0 6281
rank 1 col 10 bk 3 cs0_row 15 bw 2 row_3_4 0
size 40000000
SDRAM base=60000000, size=40000000
992 MiB
MMC:   dwmmc@30020000: 0
*** Warning - bad CRC, using default environment

In:    serial@11030000
Out:   serial@11030000
Err:   serial@11030000
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0 

MMC read: dev # 0, block # 32768, count 20480 ... 20480 blocks read: OK
Wrong Image Format for bootm command
ERROR: can't get kernel image!
=> 
=> run  bootcmd_mmc0
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:6...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
reading /extlinux/extlinux.conf
171 bytes read in 6 ms (27.3 KiB/s)
1:      kernel-4.4
Retrieving file: /zImage
reading /zImage
7340192 bytes read in 5227 ms (1.3 MiB/s)
append: earlyprintk console=ttyFIQ0,1500000n8 rw root=/dev/mmcblk0p7 rootfstype=ext4 init=/sbin/init
Retrieving file: /rk3229-echo-v10.dtb
reading /rk3229-echo-v10.dtb
26513 bytes read in 27 ms (958 KiB/s)
## Flattened Device Tree blob at 61f00000
   Booting using the fdt blob at 0x61f00000
   Loading Device Tree to 683f6000, end 683ff790 ... OK

Starting kernel ...

```

wzyy2, 2017-07-13:
试了下 可以起到一半 uart没输出主要是哪个rk322x-android.dtsi里同时enable了FIQ和uart2 你花下功夫配dts
应该可以起来
https://github.com/rockchip-linux/build/commit/94b399ed1ea7b5fb576e939cf57b2f59f900978c

wzyy2, 2017-07-13:
但这个是没正式支持的,   
pm, gpu, vpu这些都不能工作

如果你只是验证cpu性能, 你还得确保下cpufreq和thermal在正常工作

Pillar1989, 2017-07-13:
@wzyy2 
谢谢,我这边也可以启动到一半：
```
U-Boot 2017.05-g6c615f4-dirty (Jul 13 2017 - 09:38:33 +0800)

Model: Rockchip RK3229 Evaluation board
DRAM:  rk322x_dmc_probe
rockchip_sdram_size 110005d0 6281
rank 1 col 10 bk 3 cs0_row 15 bw 2 row_3_4 0
size 40000000
SDRAM base=60000000, size=40000000
992 MiB
MMC:   dwmmc@30020000: 0
*** Warning - bad CRC, using default environment

In:    serial@11030000
Out:   serial@11030000
Err:   serial@11030000
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0 

MMC read: dev # 0, block # 32768, count 20480 ... 20480 blocks read: OK
Wrong Image Format for bootm command
ERROR: can't get kernel image!
=> run bootcmd_mmc0
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:6...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
reading /extlinux/extlinux.conf
200 bytes read in 6 ms (32.2 KiB/s)
1:      kernel-4.4
Retrieving file: /zImage
reading /zImage
7340192 bytes read in 5228 ms (1.3 MiB/s)
append: earlycon=uart8250,mmio32,0x11030000 console=ttyS2,1500000n8 rw root=/dev/mmcblk2p7 rootfstype=ext4 init=/sbin/init rootwait
Retrieving file: /rk3229-echo-v10.dtb
reading /rk3229-echo-v10.dtb
26537 bytes read in 25 ms (1 MiB/s)
## Flattened Device Tree blob at 61f00000
   Booting using the fdt blob at 0x61f00000
   Loading Device Tree to 683f6000, end 683ff7a8 ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.70 (baozhu@server) (gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.4) ) #3 SMP Thu Jul 13 11:15:7
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: RK3229 Echo Board V1.0
[    0.000000] earlycon: Early serial console at MMIO32 0x11030000 (options '')
[    0.000000] bootconsole [uart0] enabled
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] PERCPU: Embedded 13 pages/cpu @ef7a8000 s24408 r8192 d20648 u53248
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 260608
[    0.000000] Kernel command line: earlycon=uart8250,mmio32,0x11030000 console=ttyS2,1500000n8 rw root=/dev/mmcblk2p7 rootfstype=ext4 init=/sbint
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 1021364K/1048576K available (11264K kernel code, 852K rwdata, 3040K rodata, 1024K init, 598K bss, 27212K reserved, 0K cma-)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
[... 141 lines cut ...]
[    0.000000] usbserial: USB Serial support registered for Keyspan 1 port adapter
[    0.000000] usbserial: USB Serial support registered for Keyspan 2 port adapter
[    0.000000] usbserial: USB Serial support registered for Keyspan 4 port adapter
[    0.000000] usbcore: registered new interface driver option
[    0.000000] usbserial: USB Serial support registered for GSM modem (1-port)
[    0.000000] usbcore: registered new interface driver oti6858
[    0.000000] usbserial: USB Serial support registered for oti6858
[    0.000000] usbcore: registered new interface driver pl2303
[    0.000000] usbserial: USB Serial support registered for pl2303
[    0.000000] usbcore: registered new interface driver qcserial
[    0.000000] usbserial: USB Serial support registered for Qualcomm USB modem
[    0.000000] usbcore: registered new interface driver sierra
[    0.000000] usbserial: USB Serial support registered for Sierra USB modem
[    0.000000] usbcore: registered new interface driver iforce
[    0.000000] usbcore: registered new interface driver xpad
[    0.000000] usbcore: registered new interface driver usbtouchscreen
[    0.000000] i2c /dev entries driver
[    0.000000] rk808 0-0018: Pmic Chip id: 0x8050
[    0.000000] rk808 0-0018: source: on=0x10, off=0x00
[    0.000000] rk808-rtc rk808-rtc: rtc core: registered rk808-rtc as rtc0
[    0.000000] rk3x-i2c 11050000.i2c: Initialized RK3xxx I2C bus at f0882000
[    0.000000] usbcore: registered new interface driver uvcvideo
[    0.000000] USB Video Class driver (1.1.1)

```

eddiecailinux, 2017-07-13:
3229没有正式的linux支持。需要正式的linux支持的话请联系业务部提出正式需求

closed 2017-07-13 by eddiecailinux

Wall-ee, 2019-10-21:
> 你好，我使用RK3229 echo板。
> [image]
> 启动的时候到Starting kernel ...就挂了。下面是具体的log:
>
> ```
> SPL Inittimer init done
> rk322x_dmc_probe
> ddr clk dpll
> ret=600000000
> sdram_all_config bw 2 col 10
> sdram_all_config reg [...]

兄弟，你的烧录步骤根据的是什么手册？另外 你的 loader 用的是哪个？我们也想快速验证一下想法