rockchip-linux/kernel #23: RK3288 kernel hang on boot

francescolavra, 2017-07-19:
I'm having issues with booting Linux on my RK3288 Firefly Reload board. With some configurations of kernel command line options and peripherals in the dts, the kernel hangs during boot: it hangs around the point where late initcalls are being executed. No stack trace or anything else is printed on the console, the system just hangs. This happens mostly when booting the kernel with the "quiet" command line option, but I have seen it happening also without this option. I tried removing various peripherals from the device tree (SD slot, eMMC, USB ports, I2C, UART) to single out possible sources of the problem, but the kernel still hangs without these peripherals.
I'm using an LVDS panel as a display interface, with the LVDS driver taken from this repository and added to mainline kernel (tried with 4.11 and 4.12 mainline kernel versions), but the problem is not in the LVDS stuff, because I have seen the system hanging with the mainline kernel (which doesn't have support for LVDS) too.

Are there any known issues that may cause the kernel to hang with this SoC?

wzyy2, 2017-07-19:
You can test with cpu/gpu devfreq disabled.

francescolavra, 2017-07-19:
Thanks for your reply, I tried disabling devfreq in the kernel configuration but it didn't solve the problem.
Then I tried disabling also cpufreq, and in this case I don't have the problem anymore. However, as far as I can see without cpufreq the CPU always runs at a reduced speed (I can see a constant VDD_CPU voltage of 1.0V, which indicates that the CPU is running at around 800 MHz), and since we need to use the full CPU capacity this is not an option.
Also I wonder whether the fact that it doesn't hang with cpufreq disabled is simply due to a different CPU speed and so different timings involved at initialization, because in this case the problem could come back when adding/removing kernel features of device tree nodes that impact boot time.

Caesar-github, 2019-04-19:
obsolete issue.

closed 2019-04-19 by Caesar-github

Victoriaentissar, 2026-01-07:
Hi all,

I know this issue is quite old, but I’m currently working on a board with the RK3288 SoC and building a Linux OS using Yocto and kernel 6.1. I’m seeing the same behavior described here — the kernel hangs during boot when CPU frequency scaling is enabled.

I need CPU frequency scaling to be enabled for my use case. Has anyone figured out how this was resolved, or are there any recommended workarounds to safely enable cpufreq without hanging?

Thanks in advance!