rockchip-linux/kernel #24: Bug: RK3399 No rule to make target 'zImage'

nightseas, 2017-07-21:
Hi,

I was trying to compile kernel branch release-4.4 for RK3399, following Rockchip wikidot. But got error: No rule to make target 'zImage'.  Stop.

Also tried bzImage, not work either.

It's possible that there's something wrong with the Makefile.

```sh
$~/oshw/rk3399/build/kernel$ export ARCH=arm64
$~/oshw/rk3399/build/kernel$ export CROSS_COMPILE=aarch64-linux-gnu-
$~/oshw/rk3399/build/kernel$ make rockchip_linux_defconfig
#
# configuration written to .config
#
$~/oshw/rk3399/build/kernel$ make zImage
scripts/kconfig/conf  --silentoldconfig Kconfig
make: *** No rule to make target 'zImage'.  Stop.
$~/oshw/rk3399/build/kernel$ git checkout 
Your branch is up-to-date with 'origin/release-4.4'.
```

Only this option worked. 
```sh
make rk3399-sapphire-excavator-linux.img
```
But how could I specify the output resource.img or kernel.img with extlinux.conf?

Thanks and BR,
nightseas

wzyy2, 2017-07-21:
There is no zImage in a arm64 build.

You could use below scripts.
https://github.com/rockchip-linux/build

nightseas, 2017-07-29:
原来是陈豪同学在维护啊。多谢解答。其他的问题挪到rockchip-linux/build吧。

Close this case and move other RK3399 build questions to rockchip-linux/build.
Thanks & BR.

closed 2017-07-29 by nightseas

geerlingguy, 2020-10-23:
And just a note here to any passers-by who are trying to rebuild the kernel for ARM64 / v8, the change you should make in the command is `make Image` instead of `make zImage` (in case that's not obvious) :P