rockchip-linux/kernel #28: RK3328: RGA (2D graphics acceleration) device tree entry is missing

OtherCrashOverride, 2017-07-26:
According to the RK3328TRM
http://opensource.rock-chips.com/images/9/97/Rockchip_RK3328TRM_V1.1-Part1-20170321.pdf

Page 13, Figure 1-1 shows RGA to be present at FF39_0000 to FF3A_0000.  However, the RK3328 device tree does not contain an entry for RGA.
https://github.com/rockchip-linux/kernel/blob/release-4.4/arch/arm64/boot/dts/rockchip/rk3328.dtsi

A device tree entry for RGA on RK3328 needs to be added.

wzyy2, 2017-07-30:
RGA driver is still under development, we are not very satisfied with the current driver.

nikitos1550, 2017-11-26:
What is status for rga now? can you provide some sample of rga dts for testing? For rk3328.

wzyy2, 2017-11-27:
RGA driver is finished now. You can add rga device tree node for rk3328 by yourself. If it work, you can send the patch to mainline kernel and i can backport it.

refer to:
https://github.com/rockchip-linux/kernel/blob/release-3.10/arch/arm64/boot/dts/rk322xh.dtsi#L1033
https://kernel.googlesource.com/pub/scm/linux/kernel/git/next/linux-next/+/master/Documentation/devicetree/bindings/media/rockchip-rga.txt

nikitos1550, 2017-11-27:
Thanks, I will try.