rockchip-linux/kernel #30: RK3288-FireFly no HDMI Sound

sghazagh, 2017-08-11:
Hi,
I just have used this repository to update my kernel and device tree for rk3288-Firefly.
The only available Sound system is SPDIF and we have no sound on HDMI to select.

When I use KODI I only have to connect SPDIF and the sound is not available on HDMI.
Can you please provide a patch and fix for having HDMI sound as an alternative option to SPDIF for sound system.

Thanks

sghazagh, 2017-08-16:
Seriously? no one even bother to reply?!!!!!

geekerlw, 2017-08-16:
you need enable sound support for hdmi, just do like this: https://github.com/geekerlw/kernel/commit/5d501039425c73d23277ff8227336748524bc652 , not complete

sghazagh, 2017-08-16:
hi,
thanks for reply.
I tried to adjust the dtsi you have provided.
- if I only add sounds section to existing rk3288-firefly.dtsi, the board does not boot at all.
- if I use completely the dtsi you gave the link, I have to remove the "io-domain" section as it said it is duplicated.
in that way, I could load the board but had some issue
= first of all, the HDMI sound wasn't available again
= the wifi went off. Even when I enable the "rkwifi section (line 266-269) which is commented by default, it does not detect the wifi anymore.
= the HDMI resolution drops to 1280x820 from 1920x1080

I think the patch is for Analog Audio and not HDMI!

Have you got audio on HDMI?
What if I compile the whole kernel you addressed above?
Does that work?

geekerlw, 2017-08-16:
The whole kernel under my github is working both earphone and hdmi, the commit above also change wifi or bluetooth, I forget, see another commit https://github.com/geekerlw/kernel/commit/17c71f754f0d3101a4caaadc701cc8371027bb7b , the two commit maybe complete. Anyway, just make sure dts have current sound node, do not use my dts, I disabled rkwifi system:)

sghazagh, 2017-08-17:
Unfortunately I don't have enough knowledge to manipulate Device Trees. I only can adjust simple changes.
Is it possible you help me write a proper part which needs to have HDMI sound?
Here is the DTS file I need to get fixed: 
[https://github.com/rockchip-linux/kernel/blob/release-4.4/arch/arm/boot/dts/rk3288-firefly.dtsi](https://github.com/rockchip-linux/kernel/blob/release-4.4/arch/arm/boot/dts/rk3288-firefly.dtsi)

If you fix that,I think everyone using official rockchip-linux kernel repository will get the benefit....
Hope you can have some spare time to help... !?!?!?

sghazagh, 2017-08-17:
I tried to only add  the changes on these two commits into the dtsi file which I mentioned the link in my last post.
The system cannot boot at all!!!

By the way, other than these changes, do I need to enable anything in Kernel itself?

sghazagh, 2017-08-17:
I just found that there is reference to i2c5 which was not exist in my dtsi.
Have added that and I could boot the machine.
However, the resolution was to big! 1356x2048 !!!
I just made it window by chance to see what is going on and Audio still didn't have the HDMI option!!!

I think I have to give up if no one helps :(

geekerlw, 2017-08-17:
I don't know why, but you didn't give any log and the board environment you run, no log no truth

发自我的 iPhone

elar-systems, 2017-08-17:
What do you need, I can post whatever can help.
As I mentioned, I use:
- linux-rockchip kernel, compiled as default defconfig from this repository
- OS is default rk-build-rootfs provided by rockchip wiki guide uses linaro base Debian Stretch
- KODI 17.3 provided by user mac-l1 in Firefly Forum.

What I have done:
- Used the default rk3288-firefly.dtsi file, in that case all good except no HDMI audio.
- I used the same file, manually patched the file with above two commits you mentioned above.
Then the board boots but screen scales too big by default. No HDMI listed in Audio sources
- To test if all setting all fine, I have used kernel and DTB was in image provided with user mac-l1 for kodi 17.3. That kernel and DTB works just fine and HDMI also is available.

The issue is that, I have no source for the working one and I don't know from where that guys compiled and imncluded the DTB file inside his image. So if I want any change in Device tree (LED for instance), or kernel feature, then I have no option.
Thats why I need a working kernel and dtb with access to source.

Another explanation, I am using KODi starts from command. So I installed "xserver-xorg-legacy" to be able to run it from command. I disabled display-manager service to directly get into KODi.

As I mentioned all good with that guys kernel and DTB so there is nothing wrong with settings.
I only do not have proper compiled kernel and device tree of mine!

geekerlw, 2017-08-18:
I want to see your boot log, you can send to me a persional email, maybe you can try my kernel, I'll put it in the reply email:)

sghazagh, 2017-08-18:
You mean "dmesg" for boot log, right? Ok, I will send you ...
(I also sent an email right now to have my email address)

I would like to test your Kernel, but you said don't use the dtsi file as you have customized it for your use!

Many thanks for your help...

sghazagh, 2017-08-22:
@geekerlw,
I double checked the difference of your kernel and the one in here.
I adjusted the DTS file and this time the sound got worked and I hade audio on HDMI.
However, if I use the new compiled one which I adjusted, the only resolution will be 1536x2048 therefore that does not fit to my screen and oversize.

Have you any idea why the resolution changes unexpectedly?
I search through "rk3288-firefly.dtsi" file and there is nothing hard-coded for display resolution.
Do you know why the HDMI resolution changes when we apply your sound changes to the 4.4.77 kernel DTSI file?

Anyone else have any idea how we can fix the display resolution issue (on KODI 17.3 I'm talking)!!!

geekerlw, 2017-08-22:
I don't know why. My release branch is 2066 commits behind rockchip-linux:release-4.4, I don't have enouth time to upgrade to the newest kernel.

sghazagh, 2017-08-23:
So, then please someone try to patch this kernel to have sound on HDMI...