rockchip-linux/kernel #31: RK3328 kernel hang on h264 encoding

sueshieh, 2017-08-21:
Using the updated source code:
- kernel release-4.4 branch (03d6a38) + rockchip_linux_defconfig + rk3328-evb.dtb
- mpp release branch (60cbbff / tag release_20170811)
- run mpi_enc_test

> rock64@rock64:~/mpp/out$ sudo ./test/mpi_enc_test -t 7 -d 255 -i ./oo.raw -o ./oo.h264 -w 960 -h 544
mpi_enc_test: cmd parse result:
mpi_enc_test: input  file name: ./oo.raw
mpi_enc_test: output file name: ./oo.h264
mpi_enc_test: width      : 960
mpi_enc_test: height     : 544
mpi_enc_test: type       : 7
mpi_enc_test: debug flag : ff
mpi_enc_test: mpi_enc_test start
mpi_enc_test: mpi_enc_test encoder test start w 960 h 544 type 7
mpi: mpp version: 60cbbff author: Herman Chen [h264e]: Fix QP stuck error
mpi: mpp_init leave ret 0
mpi: mpi_control enter ctx 0x55828d4500 cmd 320003 parm 0x55828d2138
mpi: mpi_control leave ret 0
mpi_enc_test: mpi_enc_test bps 1958400 fps 30 gop 60
mpi: mpi_control enter ctx 0x55828d4500 cmd 320005 parm 0x55828d2184
h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 1958400 [1836000 : 2080800]
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x55828d4500 cmd 320007 parm 0x55828d21bc
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x55828d4500 cmd 32000f parm 0x55828d2790
mpi: mpi_control leave ret 0
mpi: mpi_control enter ctx 0x55828d4500 cmd 32000e parm 0x7fc4dc6038
mpi: mpi_control leave ret 0
mpi: mpi_poll enter ctx 0x55828d4500 type 0 timeout -1
mpi: mpi_poll leave ret 0
mpi: mpi_dequeue enter ctx 0x55828d4500 type 0 task 0x7fc4dc6030
mpi: mpi_dequeue leave ret 0
mpi: mpi_enqueue enter ctx 0x55828d4500 type 0 task 0x55828d7d80
mpi: mpi_enqueue leave ret 0
mpi: mpi_poll enter ctx 0x55828d4500 type 1 timeout -1

- Kernel log

>[   41.675708] ff340000.vepu: power on
[   41.676326] Unhandled fault: synchronous external abort (0x96000210) at 0xffffff8008066804
[   41.677599] Internal error: : 96000210 [#1] SMP
[   41.678269] Modules linked in:
[   41.678760] CPU: 1 PID: 525 Comm: mpp_enc_ctrl Not tainted 4.4.71 #3
[   41.679682] Hardware name: Rockchip RK3328 EVB (DT)
[   41.680403] task: ffffffc038530000 ti: ffffffc038538000 task.ti: ffffffc038538000
[   41.681498] PC is at rk_iommu_read+0x20/0x34
[   41.682132] LR is at rk_iommu_is_stall_active+0x44/0x64
[   41.682893] pc : [<ffffff8008452118>] lr : [<ffffff80084521c4>] pstate: 80000145
[   41.696025] sp : ffffffc03853bb60
[   41.708725] x29: ffffffc03853bb60 x28: ffffffc038538000
[   41.721762] x27: ffffff8008bb2000 x26: 000000000000001d
[   41.734521] x25: 000000000000011d x24: ffffffc0397cd030
[   41.747008] x23: ffffffc03977ded0 x22: ffffffc03e5cac10
[   41.759381] x21: ffffffc03e67b728 x20: ffffff8008066800
[   41.771668] x19: 0000000000000004 x18: 00000000000a4f7c
[   41.783931] x17: 0000007f7ab57ed0 x16: ffffff80081b6254
[   41.795952] x15: 000000000000000a x14: ffffff8009683c20
[   41.807699] x13: ffffff8009683c20 x12: 0000000000000000
[   41.819401] x11: ffffff8008c0d600 x10: 0000000000000001
[   41.830920] x9 : 0000000000000001 x8 : 00000000ffffffff
[   41.842345] x7 : ffffffc039053480 x6 : ffffff8009683c20
[   41.853679] x5 : 0000000000000000 x4 : 0000000000000000
[   41.864900] x3 : 0000000000000000 x2 : ffffff80091504e8
[   41.876019] x1 : 0000000000000004 x0 : ffffff8008066804
[   41.886960]
[   41.886960] PC: 0xffffff8008452098:
[   41.907716] 2098  910103a1 aa1703e0 d63f0040 350001a0 110006d6 910103a4 2a1603e3 aa1503e2
[   41.919268] 20b8  aa1403e1 aa1803e0 9411d621 350000e0 f94023a0 97ffffbd aa0003f3 b5fffde0
[   41.930935] 20d8  d2800000 14000002 aa1303e0 a94153f3 a9425bf5 a94363f7 a8c97bfd d65f03c0
[... 300 lines cut ...]
[   62.161263] ed40: ffffff80097242c0 000000000000000a 000000000000000e 0000000000000008
[   62.171601] [<ffffff8008129d80>] watchdog_timer_fn+0xfc/0x2e0
[   62.181632] [<ffffff8008100758>] __hrtimer_run_queues+0x15c/0x280
[   62.191791] [<ffffff8008100d8c>] hrtimer_interrupt+0xac/0x1bc
[   62.201729] [<ffffff80088a3548>] arch_timer_handler_phys+0x3c/0x4c
[   62.211908] [<ffffff80080f100c>] handle_percpu_devid_irq+0xc4/0x170
[   62.222234] [<ffffff80080ec978>] generic_handle_irq+0x2c/0x44
[   62.232649] [<ffffff80080eccd8>] __handle_domain_irq+0x90/0xb8
[   62.243218] [<ffffff8008080cc8>] gic_handle_irq+0x78/0xc8
[   62.253857] Exception stack(0xffffff80090c3e10 to 0xffffff80090c3f40)
[   62.264892] 3e00:                                   0000000000000140 0000000000000000
[   62.276417] 3e20: 0000000000000000 ffffff80090e6140 0100000000000000 0000000057d60bc3
[   62.287825] 3e40: 0000000000008cae 0000000000000000 ffffff80090c3f1c ffffff80090c0000
[   62.299370] 3e60: 0000000000000900 0000000000000400 0000000000000144 0000000000000004
[   62.310742] 3e80: 0000000000000001 0000000000000007 000000000000000e 0000000000000008
[   62.322002] 3ea0: 000000001ed3e33c 0000000000000000 0000000000000000 ffffff80090c0000
[   62.333127] 3ec0: ffffff80090cb000 ffffff80090cbe18 ffffff8009004b38 00000000032d5000
[   62.344441] 3ee0: 00000000032d8000 0000000002ba01f0 0000000002fb001c ffffff80090c3f40
[   62.355815] 3f00: ffffff80080df29c ffffff80090c3f40 ffffff80080df30c 0000000060000145
[   62.367338] 3f20: ffffff80090c0000 ffffff80090cb000 ffffffffffffffff ffffff80080df29c
[   62.378957] [<ffffff8008081fb4>] el1_irq+0xb4/0x140
[   62.390389] [<ffffff80080df30c>] cpu_startup_entry+0x23c/0x2b8
[   62.401705] [<ffffff8008b9aa30>] rest_init+0x78/0x80
[   62.412975] [<ffffff8008fb0bc0>] start_kernel+0x3d0/0x3e4
[   62.424114] [<ffffff8008fb01c4>] __primary_switched+0x30/0x6c

yanghanxing, 2017-08-31:
have fixed:
https://github.com/rockchip-linux/kernel/commit/db41eea25d20bfcb776f000294bccea16132a6d0

dalmatele, 2017-09-25:
Now I see this log from kernel:
mpp_dev_ioctl:545: unknown mpp ioctl cmd 40086c03

closed 2017-10-24 by hizukiayaka