rockchip-linux/kernel #32: RK3288 ov13850_v4l2-i2c-subdev

teseo-sw, 2017-08-24:
Hi,
I am trying to capture stream of a ov13850 mipi camera. I am able to communicate with sensor via i2c but stream always hang on VIDEOC_DQBUF. Is kernel MIPI infrastructure complete and working or something is under development yet?

Searching online I have seen this [thread](https://groups.google.com/a/chromium.org/forum/#!topic/chromium-os-reviews/3cKU_vCKkm8%5B1-25%5D) and I suppose kernel CIF_ISP camera driver is not complete yet. Is it so?

Could you give me some advice?
thank you for the attention, best regards
Andrea

wzyy2, 2017-08-25:
There is a new standard ISP driver are developing, it will takes some months to finish it.

You can get the early access one from my github for fun or study
https://github.com/wzyy2/linux/tree/tmp/cam

teseo-sw, 2017-08-25:
Hi Jacob,
so, with the current CIF_ISP driver is not possible to transfer any frame data successfully? Following wiki at [http://rockchip.wikidot.com/mipicsi](http://rockchip.wikidot.com/mipicsi) I suppose someone already have used this ISP driver to get sensor data using ION. Is this not so?

thank you, regards
Andrea

wzyy2, 2017-08-25:
There might be some bug in driver,  i don't know. It work in rk3288 years ago...

wzyy2, 2017-08-25:
Anyway,  we won't have any develop plan or bug fix in this drvier in rk3288.

wzyy2, 2017-08-31:
I have push the new MIPI CSI driver to `isp-early-access` branch.
https://github.com/rockchip-linux/kernel/tree/isp-early-access

I will maintain this branch until it be merged into release-4.4.
Report bugs are welcome.

Missing feature in this new driver:
- No MIPI-TXRX/ DVP support
- No iommu support
- No 3A support

closed 2017-08-31 by wzyy2

zhujiongfu, 2017-10-30:
@wzyy2 Hi, has the new driver supported MIPI-TXRX port yet?

zhujiongfu, 2017-10-30:
According to mipi_dphy_sy.c file, it hasn't configed any register about txrx port yet, which means new driver doesn't support txrx port? Is that right?