rockchip-linux/kernel #34: USB 3.0 disconnect hangs USB subsystem

rmbusy, 2017-09-10:
First saw this with xenial-minimal-rock64-0.4.17-85-arm64.img. Next tried jessie-minimal-rock64-0.5.7-101-arm64.img, and seeing same basic issue. After disconnecting USB HD, any attempt to interface with USB causes timeout and kernel panic.

Using rock64 4GB, booting from 4G SD Card. Connected a USB 3.0 external disk drive (already formatted w/ 2 linux partitions). Drive is a Seagate 2GB 2.5" hybrid.

Mounted both partitions under /media/sda1 & /media/sda3. Did 'ls -ahl' on both mounts. Then unmounted both partitions, and verified with "mount".

The jessie image console output at disconnect was :
[ 873.393783] usb usb5-port1: Cannot enable. Maybe the USB cable is bad?
[ 873.394805] xhci-hcd xhci-hcd.8.auto: Cannot set link state.
[ 873.395754] usb usb5-port1: cannot disable (err = -32)
[ 875.910788] usb usb5-port1: Cannot enable. Maybe the USB cable is bad?
[ 875.911801] xhci-hcd xhci-hcd.8.auto: Cannot set link state.
[ 875.912735] usb usb5-port1: cannot disable (err = -32)
[ 878.426774] usb usb5-port1: Cannot enable. Maybe the USB cable is bad?
[ 878.427787] xhci-hcd xhci-hcd.8.auto: Cannot set link state.
[ 878.428729] usb usb5-port1: cannot disable (err = -32)
[ 880.943791] usb usb5-port1: Cannot enable. Maybe the USB cable is bad?
[ 880.944809] xhci-hcd xhci-hcd.8.auto: Cannot set link state.
[ 880.945741] usb usb5-port1: cannot disable (err = -32)
[ 880.946543] xhci-hcd xhci-hcd.8.auto: Cannot set link state.
[ 880.947492] usb usb5-port1: cannot disable (err = -32)
[ 880.948328] scsi host0: uas_post_reset: alloc streams error -19 after reset
[ 880.952607] sd 0:0:0:0: [sda] Synchronizing SCSI cache

Awhile later, tried to do a lsusb, and it hung. Then kernel panic was next :
(attached as text file, since form is messing with content)
[kernelpanic.txt](https://github.com/rockchip-linux/kernel/files/1290965/kernelpanic.txt)

ayufan, 2017-09-14:
I'm facing the same problem, also on mainline kernel.

rmbusy, 2017-09-17:
Seeing additional related issues of unhandled level 3 translation faults :

 [432893.072550] cat[11190]: unhandled level 3 translation fault (11) at 0x005017
[432893.076615] pgd = ffffffc0c2d25000
[432893.080031] [005017e0] *pgd=00000000f1227003, *pud=00000000f1227003, *pmd=00000000c2e610030
[432893.084400] 
[432893.087993] CPU: 2 PID: 11190 Comm: cat Not tainted 4.4.77-rockchip-ayufan-101 #1
[432893.091970] Hardware name: Rockchip RK3328 Rock64 (DT)
[432893.095965] task: ffffffc0f4c73600 ti: ffffffc0c2d38000 task.ti: ffffffc0c2d38000
[432893.100084] PC is at 0x5017e0
[432893.103887] LR is at 0x401cb8
[432893.107979] pc : [<00000000005017e0>] lr : [<0000000000401cb8>] pstate: 60000000
[432893.112271] sp : 0000007fc1beaec0
[432893.116452] x29: 0000007fc1beaec0 x28: 0000000000000000 
[432893.120658] x27: 0000000000000000 x26: 0000000000000000 
[432893.124747] x25: 0000000000000000 x24: 0000000000000000 
[432893.128803] x23: 0000000000000000 x22: 0000000000000000 
[432893.132866] x21: 0000000000402850 x20: 0000007fc1beb1b8 
[432893.136859] x19: 0000000000000000 x18: 0000007fc1beaf10 
[432893.140844] x17: 00000000005017e0 x16: 000000000041a0f8 
[432893.144822] x15: 0000007fae102000 x14: 0000000000000040 
[432893.148857] x13: 0000007fae102028 x12: 0000000000008730 
[432893.152861] x11: 0000000000000090 x10: 00000000ffffffff 
[432893.156698] x9 : 0000007fae0ef0b8 x8 : 0000007fae0edad8 
[432893.160541] x7 : 00000000001a233a x6 : 0000007fae0d1aa0 
[432893.164229] x5 : fe42e3dc82313c67 x4 : 0000000000000000 
[432893.167763] x3 : 0000000000401c70 x2 : 0000007fc1beb1d0 
[432893.171428] x1 : 0000000000000000 x0 : 000000000041a000 
[432893.174936] 

This is after 5 days of uptime, running Motion 4.0.1 application.  The USB drive is still responsive, but I can produce the failure above by running lsusb.  Also, the lsusb does not return anything but "Segmentation fault"

Forgot to mention in the original report, that the drive is in a self powered Seagate Backup Plus 3.5" housing / carrier.

ayufan, 2017-09-26:
@wzyy2 does this solve problem with USB3 unable to change link state? It seems that this solves some other problem

wzyy2, 2017-09-26:
No.... It's just auto referenced by github...

ayufan, 2017-09-26:
OK. Thanks :) Do you have some idea why the usb fails?

f5Qz, 2018-04-02:
https://github.com/ayufan-rock64/linux-build/issues/140

ayufan, 2018-05-20:
@wzyy2 we still have this problem, it seems that for whatever reason the USB3 PHY is not being properly reset after soft reboot.

ayufan, 2018-07-18:
@wzyy2 @keveryang We are still seeing these problems, for example here: https://github.com/ayufan-rock64/linux-build/issues/112