rockchip-linux/kernel #35: RK3288 usb_host1 (dwc2) incomplete MJPEG frames

teseo-sw, 2017-09-12:
Hi,
has anyone ever get into an incomplete MJPG frame error streaming a UVC camera connected on **RK3288 host#1 (dw)**? connecting the same camera on host#0 (ehci) it work without any error. This happens only with format equal or grater than 1920x1080 so I suppose could be a bug in **dwc2** kernel driver. Is it possible?

I check it using a command like

    v4l2-ctl --verbose -d /dev/video2 --stream-mmap=2 --stream-to=./stream.jpg --stream-count=10 --stream-poll

I do not see any error on dmseg log but I obtain partial MJPG frames.
thank you
Andrea

wzyy2, 2017-09-13:
Could you try this patch?
```
 drivers/phy/phy-rockchip-usb.c | 22 +++++++++++++++++++++-
 1 file changed, 21 insertions(+), 1 deletion(-)

diff --git a/drivers/phy/phy-rockchip-usb.c b/drivers/phy/phy-rockchip-usb.c
index 39f419d..c8b6ab3 100644
--- a/drivers/phy/phy-rockchip-usb.c
+++ b/drivers/phy/phy-rockchip-usb.c
@@ -1,5 +1,5 @@
 /*
- * Rockchip usb PHY driver
+ * Rockchip usb PHY driver
  *
  * Copyright (C) 2014 Yunzhi Li <lyz@rock-chips.com>
  * Copyright (C) 2014 ROCKCHIP, Inc.
@@ -52,11 +52,14 @@ struct rockchip_usb_phys {
 	const char *pll_name;
 };
 
+struct rockchip_usb_phy_base;
+
 struct rockchip_usb_phy_pdata {
 	struct rockchip_usb_phys *phys;
 	unsigned int phy_pw_on;
 	unsigned int phy_pw_off;
 	bool siddq_ctl;
+	int (*phy_tuning)(struct rockchip_usb_phy_base *);
 };
 
 struct rockchip_usb_phy_base {
@@ -281,6 +284,16 @@ err_clk:
 	return err;
 }
 
+static int rk3288_usb_phy_tuning(struct rockchip_usb_phy_base *phy_base)
+{
+	unsigned int uoc1_word_if_control = 0x08000000;
+	int ret = 0;
+
+	ret |= regmap_write(phy_base->reg_base, 0x0334, uoc1_word_if_control);
+
+	return ret;
+}
+
 static const struct rockchip_usb_phy_pdata rk3066a_pdata = {
 	.phys = (struct rockchip_usb_phys[]){
 		{ .reg = 0x17c, .pll_name = "sclk_otgphy0_480m" },
@@ -313,6 +326,7 @@ static const struct rockchip_usb_phy_pdata rk3288_pdata = {
 	.phy_pw_on  = SIDDQ_WRITE_ENA | SIDDQ_OFF,
 	.phy_pw_off = SIDDQ_WRITE_ENA | SIDDQ_ON,
 	.siddq_ctl  = true,
+	.phy_tuning = rk3288_usb_phy_tuning,
 };
 
 static const struct rockchip_usb_phy_pdata rk336x_pdata = {
@@ -366,6 +380,12 @@ static int rockchip_usb_phy_probe(struct platform_device *pdev)
 		return PTR_ERR(phy_base->reg_base);
 	}
 
+	if (phy_base->pdata->phy_tuning) {
+		err = phy_base->pdata->phy_tuning(phy_base);
+		if (err)
+			dev_err(dev, "phy tuning fail\n");
+	}
+
 	/* Request the vbus_drv GPIO asserted */
 	phy_base->vbus_drv_gpio =
 		devm_gpiod_get_optional(dev, "vbus_drv", GPIOD_OUT_HIGH);
-- 
2.7.4

```

teseo-sw, 2017-09-13:
I have tried it, but nothing seems to change about my issue

teseo-sw, 2017-09-15:
Could someone tell me what could cause that USB problem? Is it something wrong with usb_host1 or with dwc2 usb driver? I need to delve into it cause I need to use both RK3288 usb hosts.

Could it be something related to USB bandwidth? I suppose that cause if I reduce image size this problem disappears or, at least, it becomes less frequent. Could someone give me some hints?

best regards

closed 2018-04-06 by hizukiayaka