rockchip-linux/kernel #36: Building Errors. build/mk-kernel.sh rk3288-firefly

Daniel-Vindio, 2017-10-07:
Hello!
I'm trying to compile the kernel for a rk3288-firefly, but I'm having some making errors.

Please, I would appreciate any help with this matter.

Here you can find some information about the OS and the machine, and a summary of the build/mk-kernel.sh rk3288-firefly output (only the error messages)

Asus Core i5. x86_64 GNU/Linux

$ sudo build/mk-kernel.sh rk3288-firefly
 ...
Building kernel for rk3288-evb board!
Using rockchip_linux_defconfig

net/rfkill/rfkill-bt.c: In function ‘rfkill_rk_remove’:
net/rfkill/rfkill-bt.c:703:5: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
error, forbidden warning: rfkill-bt.c:703
scripts/Makefile.build:258: fallo en las instrucciones para el objetivo 'net/rfkill/rfkill-bt.o'
make[2]: *** [net/rfkill/rfkill-bt.o] Error 1
scripts/Makefile.build:403: fallo en las instrucciones para el objetivo 'net/rfkill'
make[1]: *** [net/rfkill] Error 2
make[1]: *** Se espera a que terminen otras tareas....

Makefile:986: fallo en las instrucciones para el objetivo 'net'
make: *** [net] Error 2
make: *** Se espera a que terminen otras tareas....

drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/siutils.c: In function ‘si_doattach’:
drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/siutils.c:544:14: warning: self-comparison always evaluates to false [-Wtautological-compare]
error, forbidden warning: siutils.c:544
scripts/Makefile.build:258: fallo en las instrucciones para el objetivo 'drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/siutils.o'
make[6]: *** [drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/siutils.o] Error 1
make[6]: *** Se espera a que terminen otras tareas....

scripts/Makefile.build:403: fallo en las instrucciones para el objetivo 'drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd'
make[5]: *** [drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd] Error 2
scripts/Makefile.build:403: fallo en las instrucciones para el objetivo 'drivers/net/wireless/rockchip_wlan/rkwifi'
make[4]: *** [drivers/net/wireless/rockchip_wlan/rkwifi] Error 2
make[4]: *** Se espera a que terminen otras tareas....

drivers/net/wireless/rockchip_wlan/rtl8188eu/core/rtw_security.c: In function ‘aes_cipher’:
drivers/net/wireless/rockchip_wlan/rtl8188eu/core/rtw_security.c:1598:5: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
error, forbidden warning: rtw_security.c:1598
scripts/Makefile.build:258: fallo en las instrucciones para el objetivo 'drivers/net/wireless/rockchip_wlan/rtl8188eu/core/rtw_security.o'
make[5]: *** [drivers/net/wireless/rockchip_wlan/rtl8188eu/core/rtw_security.o] Error 1
make[5]: *** Se espera a que terminen otras tareas....

scripts/Makefile.build:403: fallo en las instrucciones para el objetivo 'drivers/net/wireless/rockchip_wlan/rtl8188eu'
make[4]: *** [drivers/net/wireless/rockchip_wlan/rtl8188eu] Error 2
scripts/Makefile.build:403: fallo en las instrucciones para el objetivo 'drivers/net/wireless/rockchip_wlan'
make[3]: *** [drivers/net/wireless/rockchip_wlan] Error 2
make[3]: *** Se espera a que terminen otras tareas....

scripts/Makefile.build:403: fallo en las instrucciones para el objetivo 'drivers/net/wireless'
make[2]: *** [drivers/net/wireless] Error 2
scripts/Makefile.build:403: fallo en las instrucciones para el objetivo 'drivers/net'
make[1]: *** [drivers/net] Error 2
Makefile:986: fallo en las instrucciones para el objetivo 'drivers'
make: *** [drivers] Error 2
MAKE KERNEL IMAGE FAILED.

And exits with error 255.

Thank you in advance for your help.

Daniel Gutierrez

wzyy2, 2017-10-07:
It requires gcc-5.

closed 2017-10-07 by wzyy2

Daniel-Vindio, 2017-10-08:
> It requires gcc-5.

Hi, Jacob,

I've moved to gcc-5

~# gcc --version ----> gcc (GCC) 5.4.0
~# g++ --version ----> g++ (GCC) 5.4.0

I have also re installed git-core gitk git-gui gcc-arm-linux-gnueabihf
u-boot-tools etc with gcc-5.

But I'm still having the same problems with make. Here I show a summary
of the errors

Again, thank you in advance for your help.

Building kernel for rk3288-firefly board!
Using rockchip_linux_defconfig 
-----
after 2400 lines
CC      net/mac80211/util.o
CC      net/ipv4/ipcomp.o
LD      net/netlink/built-in.o
CC      net/sched/sch_generic.o
CC      net/rfkill/rfkill-bt.o
LD      net/packet/built-in.o
CC      net/sunrpc/clnt.o
CC      net/ipv4/xfrm4_tunnel.o
net/rfkill/rfkill-bt.c: In function ‘rfkill_rk_remove’:
net/rfkill/rfkill-bt.c:703:5: warning: this ‘if’ clause does not
guard... [-Wmisleading-indentation]
error, forbidden warning: rfkill-bt.c:703
scripts/Makefile.build:258: fallo en las instrucciones para el objetivo
'net/rfkill/rfkill-bt.o'
make[2]: *** [net/rfkill/rfkill-bt.o] Error 1
scripts/Makefile.build:403: fallo en las instrucciones para el objetivo
'net/rfkill'
make[1]: *** [net/rfkill] Error 2
make[1]: *** Se espera a que terminen otras tareas....

----------------------------
LD      drivers/net/ethernet/rocker/built-in.o
Makefile:986: fallo en las instrucciones para el objetivo 'net'
make: *** [net] Error 2
make: *** Se espera a que terminen otras tareas....

-----------------------------------
CC      drivers/rtc/rtc-lib.o
drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/siutils.c: In function
‘si_doattach’:
drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/siutils.c:544:14:
warning: self-comparison always evaluates to false
[-Wtautological-compare]
error, forbidden warning: siutils.c:544
scripts/Makefile.build:258: fallo en las instrucciones para el objetivo
'drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/siutils.o'
make[6]: ***
[drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/siutils.o] Error 1
make[6]: *** Se espera a que terminen otras tareas....

-----------------------------------------------

CC      drivers/scsi/scsi.o
scripts/Makefile.build:403: fallo en las instrucciones para el objetivo
'drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd'
make[5]: *** [drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd] Error 2
scripts/Makefile.build:403: fallo en las instrucciones para el objetivo
'drivers/net/wireless/rockchip_wlan/rkwifi'
make[4]: *** [drivers/net/wireless/rockchip_wlan/rkwifi] Error 2
scripts/Makefile.build:403: fallo en las instrucciones para el objetivo
'drivers/net/wireless/rockchip_wlan'
make[3]: *** [drivers/net/wireless/rockchip_wlan] Error 2
make[3]: *** Se espera a que terminen otras tareas....
CC      drivers/scsi/hosts.o

---------------------------------------------------------
 
 LD      drivers/net/wireless/mwifiex/built-in.o
 scripts/Makefile.build:403: fallo en las instrucciones para el objetivo
 'drivers/net/wireless'
 make[2]: *** [drivers/net/wireless] Error 2
 scripts/Makefile.build:403: fallo en las instrucciones para el objetivo
 'drivers/net'
 make[1]: *** [drivers/net] Error 2
 make[1]: *** Se espera a que terminen otras tareas....
 CC      drivers/video/backlight/pwm_bl.o
 -----------------------------------------------------
   
   LD      drivers/usb/serial/built-in.o
   LD      drivers/usb/built-in.o
   Makefile:986: fallo en las instrucciones para el objetivo 'drivers'
   make: *** [drivers] Error 2
   MAKE KERNEL IMAGE FAILED.
> 
> https://github.com/rockchip-linux/kernel/issues/36#issuecomment-334942975

wzyy2, 2017-10-09:
https://tinkerboarding.co.uk/forum/archive/index.php/thread-599.html.
It might help you.

Daniel-Vindio, 2017-10-12:
> https://tinkerboarding.co.uk/forum/archive/index.php/thread-599.html.
> It might help you.

Hello again,
I have prepared a patch to fix the errors. Basically it consists in
deactivating the misleading indentation and tautological compare warnings
by adding extra cflags:
-Wno-misleading-indentation -Wno-tautological-compare

To these Makefiles:

rk-linux/kernel/net/rfkill/Makefile
rk-linux/kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/Makefile
rk-linux/kernel/drivers/net/wireless/rockchip_wlan/rtl8188eu/Makefile

Now the kernel is successfully compiled.

By the way, I would like to know if there will be a revision of the
kernel source avoid these errors without the need of patching.

Although my skills in C are a little bit limited, with some instructions
I could investigate or try a fix for the source.

Thank you and greetings / saludos y gracias,

Daniel G.

> 
> https://github.com/rockchip-linux/kernel/issues/36#issuecomment-335049861

-- 
--------------------------
Daniel Gutiérrez San José

diff -Naur kernel_0/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/Makefile kernel_1/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/Makefile
--- kernel_0/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/Makefile	2017-10-07 23:34:30.746811000 +0200
+++ kernel_1/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/Makefile	2017-10-12 10:09:45.492817091 +0200
@@ -9,7 +9,8 @@
 CONFIG_BCMDHD_PROPTXSTATUS := y
 CONFIG_BCMDHD_AG := y
 
-DHDCFLAGS = -Wall -Wstrict-prototypes -Dlinux -DBCMDRIVER -DSDTEST       \
+DHDCFLAGS = -Wall -Wno-misleading-indentation -Wno-tautological-compare \
+	-Wstrict-prototypes -Dlinux -DBCMDRIVER -DSDTEST       \
 	-DBCMDONGLEHOST -DUNRELEASEDCHIP -DBCMDMA32 -DBCMFILEIMAGE            \
 	-DDHDTHREAD -DDHD_DEBUG -DSHOW_EVENTS -DBCMDBG -DGET_OTP_MAC_ENABLE   \
 	-DWIFI_ACT_FRAME -DARP_OFFLOAD_SUPPORT -DSUPPORT_PM2_ONLY             \
diff -Naur kernel_0/drivers/net/wireless/rockchip_wlan/rtl8188eu/Makefile kernel_1/drivers/net/wireless/rockchip_wlan/rtl8188eu/Makefile
--- kernel_0/drivers/net/wireless/rockchip_wlan/rtl8188eu/Makefile	2017-10-07 23:34:30.786811000 +0200
+++ kernel_1/drivers/net/wireless/rockchip_wlan/rtl8188eu/Makefile	2017-10-12 10:09:45.496817091 +0200
@@ -15,6 +15,7 @@
 EXTRA_CFLAGS += -Wno-unused
 #EXTRA_CFLAGS += -Wno-uninitialized
 #EXTRA_CFLAGS += -Wno-error=date-time	# Fix compile error on gcc 4.9 and later
+EXTRA_CFLAGS += -Wno-misleading-indentation -Wno-tautological-compare
 
 EXTRA_CFLAGS += -I$(src)/include
 EXTRA_CFLAGS += -I$(src)/hal/phydm
diff -Naur kernel_0/net/rfkill/Makefile kernel_1/net/rfkill/Makefile
--- kernel_0/net/rfkill/Makefile	2017-10-07 23:34:33.730811000 +0200
+++ kernel_1/net/rfkill/Makefile	2017-10-12 10:09:45.488817091 +0200
@@ -8,3 +8,4 @@
 obj-$(CONFIG_RFKILL_REGULATOR)	+= rfkill-regulator.o
 obj-$(CONFIG_RFKILL_GPIO)	+= rfkill-gpio.o
 obj-y				+= rfkill-wlan.o rfkill-bt.o
+ccflags-y += -Wno-misleading-indentation

FallingSnow, 2017-10-13:
@Daniel-Vindio Can you paste a link to your `.patch` file?

Daniel-Vindio, 2017-10-13:
> @Daniel-Vindio Can you paste a link to your `.patch` file?

Yes, of course.
https://github.com/Daniel-Vindio/rk3288-Firefly
> 
> https://github.com/rockchip-linux/kernel/issues/36#issuecomment-336504675

FallingSnow, 2017-10-13:
[Just finished mine](https://github.com/FallingSnow/rk3399-firefly) lol, but thank you anyway.