rockchip-linux/kernel #37: rk3036-kylin compilation fails on arm-linux-gnueabihf-gcc 6.3.0 (Ubuntu 17.04)

harsszegi, 2017-10-09:
Hello,
have latest gcc installed via aptitude install, kernel fails to compile.

drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/siutils.c:544:14: warning: self-comparison always evaluates to false [-Wtautological-compare]
error, forbidden warning: siutils.c:544

net/rfkill/rfkill-bt.c: In function ‘rfkill_rk_remove’:                                                                      
net/rfkill/rfkill-bt.c:703:5: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]                         
error, forbidden warning: rfkill-bt.c:703

wzyy2, 2017-10-09:
Please search it first.
https://github.com/rockchip-linux/kernel/issues/36

Do you want to build image for rk3036? Unfortunately, the last time i have test it is one years ago...... I'm afraid that it's totally broken now since the software have been changed a lot....

harsszegi, 2017-10-09:
Yups, I have a Kylin 1.1 board which I'd like to build uboot+kernel+rootfs to build

wzyy2, 2017-10-09:
To build a workable image,  kernel/u-boot/build code-base might need be checkout to `2017.02`.

wzyy2, 2017-11-10:
We have fix some bugs in rk3036 and i think now source code in github can build a workable image for rk3036 platform.

harsszegi, 2017-12-03:
Hello,

net/rfkill/rfkill-bt.c: In function ‘rfkill_rk_remove’:
net/rfkill/rfkill-bt.c:703:5: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/siutils.c: In function ‘si_doattach’:
drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/siutils.c:607:14: warning: self-comparison always evaluates to false [-Wtautological-compare]
error, forbidden warning: siutils.c:607

what exact version of gcc is required/supported?
I am using 7.2

arm-linux-gnueabihf-gcc (Ubuntu/Linaro 7.2.0-6ubuntu1) 7.2.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Caesar-github, 2017-12-14:
we have fixed it for now, update it on next version

harsszegi, 2017-12-14:
Awesome, kernel compiles now after repo sync -d Thanks Ceasar!
t.

closed 2017-12-15 by harsszegi