rockchip-linux/kernel #38: RK3328 - ROCK64 Support

SinghBirender, 2017-10-15:
ROCK64 mentioned as device on http://opensource.rock-chips.com/wiki_RK3328
But not dedicated dts file in 
https://github.com/rockchip-linux/kernel/tree/release-4.4/arch/arm64/boot/dts/rockchip .
Please help by expert hands.

dalmatele, 2017-10-15:
one vote for this support.
I use rock64 to encode/decode video but not successful.

wzyy2, 2017-10-16:
It's better if someone can send us a pull request.
; )

ayufan, 2017-10-17:
You can always see all my changes to "be compatible with Rock64" here: https://github.com/rockchip-linux/kernel/compare/release-4.4...ayufan-rock64:release-4.4 :)

dalmatele, 2017-10-17:
@ayufan: did it support encode/decode? Because I still have seen that it can not encoded/decoded in RK3328.

ayufan, 2017-10-17:
What do you mean?

Decode / encode of vpu? Yes, it works for me

dalmatele, 2017-10-17:
@ayufan When I run mpp encode, in the kern.log file I've seen this:
mpp_dev_ioctl:545: unknown mpp ioctl cmd 40086c04
Then output is blank frame ( zero value when using hexdump to see).
What board you've used? I use rock64 with RK3328: https://www.pine64.org/?product=rock64-media-board-computer

SinghBirender, 2017-10-20:
I have requested the merge.
Thanks Ayufan.

Regards,
Birender Singh

hizukiayaka, 2017-10-24:
We have had an internal version, please wait it release

SinghBirender, 2017-10-27:
Hello Dear hizukiayaka,

Should wait more ?

Regards.

SinghBirender, 2017-11-02:
Dear All,

In https://github.com/torvalds/linux/ source tree we have

https://github.com/torvalds/linux/blob/master/arch/arm64/boot/dts/rockchip/rk3328-rock64.dts file.

Can we refer to this after kernel version release v4.14/ ?

Release.

wzyy2, 2017-11-10:
@yanghanxing It seems many people want a rock64 dts in relase-4.4 kernel, could you help sending it?

SinghBirender, 2017-11-13:
Congratulation to all,

Mr. Linus Torvalds, turn out to be faster.

4.14 having default support for ROCK64.

Regards.

SinghBirender, 2017-11-27:
Dear Mr. Jacob Chen,

Is somebody working on Android Kernel 4.4 for RK3328 ?

Regards.

SinghBirender, 2017-12-03:
Dear Mr. @wzyy2 

Please help to propagate my query to Mr. Xinhuang Li.

https://github.com/rockchip-linux/kernel/commit/1934ca1c8e1e82eb9a12a417de358e338d75ddd4

Regards.

SinghBirender, 2017-12-04:
Dear Mr. @ayufan ,

Please help to provide kernel version 4.4 for Android.

Regards.

SinghBirender, 2017-12-08:
Dear Me @wzyy2 ,

But we started this issue to include ROCK64 board default support in offcial git directory provided by Rock-Chip.

Regards.

SinghBirender, 2017-12-29:
Dear Mr. @ayufan ,

Can you please check reply of Mr. Jacob Chen in 
https://github.com/rockchip-linux/kernel/commit/1934ca1c8e1e82eb9a12a417de358e338d75ddd4

Regards,
Birender Singh

SinghBirender, 2018-01-15:
Dear Ms. @hizukiayaka ,

Thank you for the support.

But we are getting compilation failure with following command.
make ARCH=arm64 rockchip_defconfig
make -j8 ARCH=arm64 rk3328-rock64.img

scripts/kconfig/conf  --silentoldconfig Kconfig
Makefile:708: Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler
  WRAP    arch/arm64/include/generated/asm/bug.h
  WRAP    arch/arm64/include/generated/asm/bugs.h
  WRAP    arch/arm64/include/generated/asm/checksum.h
  WRAP    arch/arm64/include/generated/asm/clkdev.h
  WRAP    arch/arm64/include/generated/asm/cputime.h
  WRAP    arch/arm64/include/generated/asm/current.h
  CHK     include/config/kernel.release
  WRAP    arch/arm64/include/generated/asm/div64.h
  WRAP    arch/arm64/include/generated/asm/delay.h
  WRAP    arch/arm64/include/generated/asm/dma.h
  WRAP    arch/arm64/include/generated/asm/dma-contiguous.h
  WRAP    arch/arm64/include/generated/asm/early_ioremap.h
  WRAP    arch/arm64/include/generated/asm/emergency-restart.h
  WRAP    arch/arm64/include/generated/asm/errno.h
  WRAP    arch/arm64/include/generated/asm/ftrace.h
  WRAP    arch/arm64/include/generated/asm/hw_irq.h
  WRAP    arch/arm64/include/generated/asm/ioctl.h
  WRAP    arch/arm64/include/generated/asm/ioctls.h
  WRAP    arch/arm64/include/generated/asm/ipcbuf.h
  WRAP    arch/arm64/include/generated/asm/irq_regs.h
  CHK     include/generated/uapi/linux/version.h
  WRAP    arch/arm64/include/generated/asm/kdebug.h
  WRAP    arch/arm64/include/generated/asm/kmap_types.h
  WRAP    arch/arm64/include/generated/asm/kvm_para.h
  WRAP    arch/arm64/include/generated/asm/local.h
  UPD     include/generated/uapi/linux/version.h
  WRAP    arch/arm64/include/generated/asm/local64.h
  WRAP    arch/arm64/include/generated/asm/mcs_spinlock.h
  WRAP    arch/arm64/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/arm64/include/generated/asm/mman.h
  WRAP    arch/arm64/include/generated/asm/msgbuf.h
  WRAP    arch/arm64/include/generated/asm/mutex.h
  WRAP    arch/arm64/include/generated/asm/msi.h
  WRAP    arch/arm64/include/generated/asm/pci-bridge.h
  WRAP    arch/arm64/include/generated/asm/pci.h
  WRAP    arch/arm64/include/generated/asm/preempt.h
  WRAP    arch/arm64/include/generated/asm/resource.h
  WRAP    arch/arm64/include/generated/asm/rwsem.h
  UPD     include/config/kernel.release
  WRAP    arch/arm64/include/generated/asm/poll.h
  WRAP    arch/arm64/include/generated/asm/serial.h
  WRAP    arch/arm64/include/generated/asm/sembuf.h
  WRAP    arch/arm64/include/generated/asm/segment.h
  WRAP    arch/arm64/include/generated/asm/sizes.h
  WRAP    arch/arm64/include/generated/asm/socket.h
  WRAP    arch/arm64/include/generated/asm/simd.h
  WRAP    arch/arm64/include/generated/asm/sockios.h
  WRAP    arch/arm64/include/generated/asm/shmbuf.h
  WRAP    arch/arm64/include/generated/asm/swab.h
  WRAP    arch/arm64/include/generated/asm/switch_to.h
  WRAP    arch/arm64/include/generated/asm/termbits.h
  WRAP    arch/arm64/include/generated/asm/termios.h
  WRAP    arch/arm64/include/generated/asm/topology.h
  WRAP    arch/arm64/include/generated/asm/trace_clock.h
  WRAP    arch/arm64/include/generated/asm/unaligned.h
  WRAP    arch/arm64/include/generated/asm/types.h
  WRAP    arch/arm64/include/generated/asm/user.h
  WRAP    arch/arm64/include/generated/asm/xor.h
  WRAP    arch/arm64/include/generated/asm/vga.h
  CHK     include/generated/utsrelease.h
  WRAP    arch/arm64/include/generated/uapi/asm/kvm_para.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/genksyms/genksyms.o
  CC      scripts/mod/empty.o
  CC      kernel/bounds.s
gcc: error: unrecognized command line option ‘-mlittle-endian’
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
scripts/Makefile.build:266: recipe for target 'scripts/mod/empty.o' failed
make[2]: *** [scripts/mod/empty.o] Error 1
scripts/Makefile.build:411: recipe for target 'scripts/mod' failed
make[1]: *** [scripts/mod] Error 2
make[1]: *** Waiting for unfinished jobs....
  CHK     include/generated/timeconst.h
gcc: error: unrecognized command line option ‘-mlittle-endian’
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
Kbuild:19: recipe for target 'kernel/bounds.s' failed
make[1]: *** [kernel/bounds.s] Error 1
make[1]: *** Waiting for unfinished jobs....
  UPD     include/generated/timeconst.h
  HOSTCC  scripts/dtc/flattree.o
Makefile:1042: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2
make: *** Waiting for unfinished jobs....
  SHIPPED scripts/genksyms/parse.tab.c
  SHIPPED scripts/genksyms/lex.lex.c
  SHIPPED scripts/genksyms/keywords.hash.c
  SHIPPED scripts/genksyms/parse.tab.h
  HOSTCC  scripts/genksyms/parse.tab.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/genksyms/lex.lex.o
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/dtc/checks.o
  HOSTCC  scripts/dtc/util.o
  SHIPPED scripts/dtc/dtc-lexer.lex.c
  SHIPPED scripts/dtc/dtc-parser.tab.h
  SHIPPED scripts/dtc/dtc-parser.tab.c
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  HOSTLD  scripts/genksyms/genksyms
  HOSTLD  scripts/dtc/dtc
Makefile:573: recipe for target 'scripts' failed
make: *** [scripts] Error 2

SinghBirender, 2018-01-15:
Dear Ms. @hizukiayaka ,

Thank you for your support.

I have compiled the kernel.

Will test it and update here.

Regards.

hizukiayaka, 2018-01-15:
Hi, man, I have a penis as well.

SinghBirender, 2018-01-17:
Dear All,

Raised new issue 
https://github.com/ayufan-rock64/android-7.1/issues/16

Tested kernel build using
make ARCH=arm64 rockchip_defconfig
make -j8 ARCH=arm64 rk3328-rock64.img
It goes to infinite boot loop.

Regards.

SinghBirender, 2018-01-21:
I have replaced
https://github.com/rockchip-linux/kernel
source with
https://github.com/ayufan-rock64/android-kernel
As used in steps
https://forum.pine64.org/showthread.php?tid=5036

Now Meting boot loop.
Please check both logs

1. Error with 4.4 kernel.
2. Normal working with 3.10.104
[Error.txt](https://github.com/rockchip-linux/kernel/files/1649455/Error.txt)
[normaldbg.txt](https://github.com/rockchip-linux/kernel/files/1649456/normaldbg.txt)

mcerveny, 2018-01-24:
At least this patch ayufan-rock64/linux-kernel@b158cf169fe451f745bab066a5d38e03644b65ba is essential to get ethernet working. Maybe other patches for RTL8211F should be useful https://github.com/ayufan-rock64/linux-kernel/commits/release-4.4?after=fbcc627e5c200c4df59903e5992f1de3ba5d59ce+78

ayufan, 2018-01-24:
@mcerveny they are not needed

mcerveny, 2018-01-24:
@ayufan my ethernet is not stable (after few packets hungs) without ayufan-rock64/linux-kernel@b158cf169fe451f745bab066a5d38e03644b65ba or do you mean RTL8211F series ?

ayufan, 2018-01-24:
We need only https://github.com/ayufan-rock64/linux-kernel/commit/b158cf169fe451f745bab066a5d38e03644b65ba. The RTL8211F series is not needed.

SinghBirender, 2018-01-25:
Same state
[Error2.txt](https://github.com/rockchip-linux/kernel/files/1665402/Error2.txt)

SinghBirender, 2018-01-26:
Dear Martin Cerveny(@mcerveny ),

I have simply replaced kernel code as setup by Mr. Kamil Trzciński(@ayufan )
using https://github.com/ayufan-rock64/android-7.1/releases

I´m also looking for standard code provided by https://github.com/rockchip-linux/kernel
for rock64.
But for android system, not Linux system.

As per my understanding code setup done by Mr. Kamil Trzciński, is the one internally provided by RockChip Corporation for ROCK64 board.

But we need official code setup provided under
http://opensource.rock-chips.com/wiki_Main_Page project.  

Regards.

SinghBirender, 2018-01-27:
Dear All,

After applying logs in 
<android>/system/core/init/init.cpp
} else {
    for (int i = 0; i < EMMC_RETRY_COUNT_SD; i++) {
		proc_read( "/proc/cmdline", cmdline, sizeof(cmdline) );
		NOTICE("cmdline=%s\n",cmdline);
		s1 = strstr(cmdline, STORAGE_MEDIA_SD);
		s2 = strstr(cmdline, "androidboot.mode=sd");
		s3 = strstr(cmdline, " androidboot.hardware=rkd");
		ERROR("s1=%s\n",s1);
		ERROR("s2=%s\n",s2);
		ERROR("s3=%s\n",s3);
		if ((s1 > 0) && (s2 > 0)) {
			ERROR("OK,SDMMC DRIVERS INIT OK\n");
			property_set("ro.boot.mode", "sd");
			break;
		} else {
			ERROR("ERROR,SDMMC DRIVERS NOT READY, RERRY=%d\n", i);
		}
	}
}

Working:

SinghBirender, 2018-02-28:
Dear Zhangbin Tong.

I will test :
https://github.com/rockchip-linux/kernel/commit/c170469b8ba86e29e586d8e814904984edf82d8b
And will update here.

Regards.

SinghBirender, 2018-03-07:
Dear All,

It is having SD-MMC issue as mentioned above.

Regards.

closed 2018-04-06 by hizukiayaka