rockchip-linux/kernel #40: RK3288-miqi无法加载任何内核模块

Pillar1989, 2017-10-17:
HI 
   我手动编译了最新的ｋｅｒｎｅｌ, 然后在ＭＩＱＩ的板子的上面运行．无法加载任何内核模块，下面是我的编译步骤：
```sh
make rockchip_linux_defconfig ARCH=arm
fakeroot make  ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- LOCALVERSION=-miqi-r1  KDEB_PKGVERSION=1stable bindeb-pkg -j 8
```
编译生成的linux-image-4.4.83-miqi-r1_1stable_armhf.deb在armbian上面安装，可以正常启动，但是启动后无法加载内核模块，例如：
```sh
v2@miqi:~$ sudo modprobe echainiv
[sudo] password for v2: 
[   63.512700] ------------[ cut here ]------------
[   63.517321] kernel BUG at mm/memory.c:1928!
[   63.521501] Internal error: Oops - BUG: 0 [#1] SMP ARM
[   63.526633] Modules linked in:
[   63.529702] CPU: 0 PID: 2203 Comm: modprobe Not tainted 4.4.83-miqi-r1 #2
[   63.536480] Hardware name: Rockchip (Device Tree)
[   63.541179] task: ed4f5400 task.stack: e44a4000
[   63.545712] PC is at apply_to_page_range+0x30/0x1e0
[   63.550589] LR is at change_memory_common+0xa0/0xd4
[   63.555463] pc : [<c020f6fc>]    lr : [<c0116d08>]    psr: 600e0013
[   63.555463] sp : e44a5dd0  ip : e44a5e10  fp : e44a5e0c
[   63.566924] r10: 00000041  r9 : bf006000  r8 : c11233b8
[   63.572143] r7 : bf002180  r6 : bf006000  r5 : 00000000  r4 : bf006000
[   63.578663] r3 : c0116c24  r2 : 00000000  r1 : bf006000  r0 : c11233b8
[   63.585183] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[   63.592309] Control: 10c5387d  Table: 2cf0006a  DAC: 00000051
[   63.598047] 
[   63.598047] PC: 0xc020f67c:
[   63.602312] f67c  e081c00c e1a0c62c e173000c 2a00000c e590404c e15c0004 3a000009 e0832004
[   63.610598] f69c  e890000a e04cc004 e0433001 e15c0623 3a000003 e590c024 e58dc000 ebffff66
[   63.618874] f6bc  ea000000 e3e00015 e24bd010 e89da810 e1a0c00d e92ddff0 e24cb004 e24dd014
[   63.627152] f6dc  e52de004 e8bd4000 e0819002 e1a08000 e1510009 e1a06001 e50b303c 3a000000
[   63.635429] f6fc  e7f001f2 e5905020 e1a03aa1 e59fa190 e0855183 e2855008 e2867602 e2493001
[   63.643712] f71c  e1a07aa7 e1a07a87 e2472001 e1520003 21a07009 e2552008 0a000044 e158000a
[   63.651990] f73c  1a00000f e5153008 e3530000 0a000006 e5154008 e7e83656 e3c44eff e3c4400f
[   63.660268] f75c  e2444440 e0844103 ea00001f e1a01006 e1a00002 ebfff813 e3500000 0afffff3
[   63.668551] 
[   63.668551] LR: 0xc0116c88:
[   63.672816] 6c88  e3530000 e1a05601 e1a04000 e1a09002 00806005 0a00000b e59f7084 e3c04eff
[   63.681100] 6ca8  e3c4400f e0856004 e5d73000 e3530000 1a000004 e3a01031 e59f0068 eb001b57
[   63.689377] 6cc8  e3a03001 e5c73000 e2843441 e353060e 2a000010 e35604bf 3a00000e e1a02005
[   63.697661] 6ce8  e24b302c e1a01004 e58d3000 e59f0038 e59f3038 e50b902c e50b8028 eb03e270
[   63.705938] 6d08  e1a01006 e1a05000 e1a00004 ebffdfcf e1a00005 ea000000 e3e00015 e24bd024
[   63.714217] 6d28  e89dabf0 c11d32cf c0dc685a c11233b8 c0116c24 e1a0c00d e92dd800 e24cb004
[   63.722498] 6d48  e52de004 e8bd4000 e3a03000 e3a02080 ebffffc2 e89da800 e1a0c00d e92dd800
[   63.730774] 6d68  e24cb004 e52de004 e8bd4000 e3a03080 e3a02000 ebffffb9 e89da800 e1a0c00d
[   63.739051] 
[   63.739051] SP: 0xe44a5d50:
[   63.743316] 5d50  c11cf248 00000000 024280c0 c020f700 00000000 c010ca98 00000000 e7100000
[   63.751602] 5d70  e44a5e0c e44a5d80 c010c660 c010100c c11233b8 bf006000 00000000 c0116c24
[   63.759881] 5d90  bf006000 00000000 bf006000 bf002180 c11233b8 bf006000 00000041 e44a5e0c
[   63.768158] 5db0  e44a5e10 e44a5dd0 c0116d08 c020f6fc 600e0013 ffffffff 00000051 c0116d08
[   63.776440] 5dd0  c0116c24 c0116c30 c0116c24 e453d018 ef4847a0 bf006000 00000000 bf006000
[   63.784719] 5df0  bf002180 00000000 00000080 00000041 e44a5e44 e44a5e10 c0116d08 c020f6d8
[   63.792997] 5e10  e44a5e18 c010ec64 00000080 00000000 c0116d3c c0116d3c bf002180 bf002180
[   63.801279] 5e30  e44a5ee4 00000000 e44a5e54 e44a5e48 c0116d5c c0116c74 e44a5e6c e44a5e58
[   63.809561] 
[   63.809561] IP: 0xe44a5d90:
[   63.813825] 5d90  bf006000 00000000 bf006000 bf002180 c11233b8 bf006000 00000041 e44a5e0c
[   63.822105] 5db0  e44a5e10 e44a5dd0 c0116d08 c020f6fc 600e0013 ffffffff 00000051 c0116d08
[   63.830386] 5dd0  c0116c24 c0116c30 c0116c24 e453d018 ef4847a0 bf006000 00000000 bf006000
[   63.838670] 5df0  bf002180 00000000 00000080 00000041 e44a5e44 e44a5e10 c0116d08 c020f6d8
[   63.846951] 5e10  e44a5e18 c010ec64 00000080 00000000 c0116d3c c0116d3c bf002180 bf002180
[   63.855232] 5e30  e44a5ee4 00000000 e44a5e54 e44a5e48 c0116d5c c0116c74 e44a5e6c e44a5e58
[   63.863511] 5e50  c019f630 c0116d48 bf0021ac c0116d3c e44a5e8c e44a5e70 c01a0d20 c019f5d8
[   63.871788] 5e70  bf002080 e44a5f44 00000000 bf002180 e44a5f3c e44a5e90 c01a2a4c c01a0cd8
[   63.880069] 
[   63.880069] FP: 0xe44a5d8c:
[   63.884334] 5d8c  c0116c24 bf006000 00000000 bf006000 bf002180 c11233b8 bf006000 00000041
[   63.892615] 5dac  e44a5e0c e44a5e10 e44a5dd0 c0116d08 c020f6fc 600e0013 ffffffff 00000051
[... 47 lines cut ...]
[   64.246213] 5ee0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   64.254383] 5f00: 00000000 00000000 00000000 00000000 e44a5f2c 00000000 004fc84c 00000003
[   64.262553] 5f20: 0000017b c0107484 e44a4000 00000000 e44a5fa4 e44a5f40 c01a32e0 c01a13e0
[   64.270723] 5f40: 00000002 f1126000 000015d8 f11270d8 f1126f8c f1126c08 00003000 000030b0
[   64.278893] 5f60: 00000000 00000000 00000000 00001634 0000001e 0000001f 00000016 00000000
[   64.287063] 5f80: 00000011 00000000 00000000 00a8deb8 00a8d1e0 a7f17d00 00000000 e44a5fa8
[   64.295233] 5fa0: c01072c0 c01a327c 00a8deb8 00a8d1e0 00000003 004fc84c 00000000 004fd890
[   64.303404] 5fc0: 00a8deb8 00a8d1e0 a7f17d00 0000017b 00040000 00000000 00a8deb8 00000000
[   64.311574] 5fe0: bebe93b8 bebe93a8 004f44bb b6f326b2 800e0030 00000003 00000000 00000000
[   64.319756] [<c020f6fc>] (apply_to_page_range) from [<c0116d08>] (change_memory_common+0xa0/0xd4)
[   64.328625] [<c0116d08>] (change_memory_common) from [<c0116d5c>] (set_memory_ro+0x20/0x24)
[   64.336973] [<c0116d5c>] (set_memory_ro) from [<c019f630>] (frob_rodata+0x64/0x68)
[   64.344541] [<c019f630>] (frob_rodata) from [<c01a0d20>] (module_enable_ro+0x54/0x5c)
[   64.352368] [<c01a0d20>] (module_enable_ro) from [<c01a2a4c>] (load_module+0x1678/0x1d48)
[   64.360542] [<c01a2a4c>] (load_module) from [<c01a32e0>] (SyS_finit_module+0x70/0x78)
[   64.368371] [<c01a32e0>] (SyS_finit_module) from [<c01072c0>] (ret_fast_syscall+0x0/0x3c)
[   64.376541] Code: e1510009 e1a06001 e50b303c 3a000000 (e7f001f2) 
[   64.382631] ---[ end trace 8ea5028c99065b36 ]---
Segmentation fault
v2@miqi:~$ 
```

下面是dmesg的ｌｏｇ
https://gist.github.com/Pillar1989/2d9a4af65b83e7347fc8f6d8c882f0e3 
我没有改任何代码，这个问题很容易复现的．

Kwiboo, 2017-10-17:
You will need this commit to fix this issue: https://github.com/torvalds/linux/commit/f474c8c857d996f34c39f66bbed23faaa739fad6

Pillar1989, 2017-10-18:
But I didnt see any one backport this commit on kernel 4.4.

wzyy2, 2017-10-18:
We didn't test kernel modules in 4.4 before, usualy we just make them built-in.
If it could slove your problem, i will backport it.

Pillar1989, 2017-10-18:
Yes,  This commit solved my problem. But I hope more people test it . Make sure it does not bring new problems

Kwiboo, 2017-10-18:
I have included this upstream commit in LibreELEC for the last two weeks to fix module loading on rk3288, see https://github.com/Kwiboo/linux-rockchip/commits/rockchip-4.4
The issue was introduced with LSK commit https://git.linaro.org/kernel/linux-linaro-stable.git/commit/?h=lsk-v4.4-17.08-android&id=01fbd3fa49ba702d1f470eb310e7a84ae14d380b
Aarch64 already has the same fix in https://github.com/rockchip-linux/kernel/commit/a68f555363f5e43d0da4a6d025e23db74e2b4f1f
Ideally this should also be fixed in LSK or longterm

wzyy2, 2017-10-19:
I have push it and it will be included in the next update.

closed 2017-10-19 by wzyy2