rockchip-linux/kernel #43: rk-isp10 haven't any interrupts with ov13850

zhujiongfu, 2017-10-30:
Hi,
I was on rk3288 board, and have added ov13850_v4l2-i2c-subdev.c file to rk-isp10 driver referring to OV13850_table.c of android 7.0, but I cann't receive any interrupts. There are any reasons for this problem? Here is my configuration about OV13850:
static struct ov_camera_module_config ov13850_configs[] = {
        {    
                .name = "2112x1568_30fps",
                .frm_fmt = {
                        .width = 2112,
                        .height = 1568,
                        .code = MEDIA_BUS_FMT_SBGGR10_1X10
                },   
                .frm_intrvl = {
                        .interval = {
                                .numerator = 1, 
                                .denominator = 30 
                        }    
                },   
                .auto_exp_enabled = false,
                .auto_gain_enabled = false,
                .auto_wb_enabled = false,
                .reg_table = (void *)ov13850_init_twolane_2112_1568_30fps,
                .reg_table_num_entries =
                        sizeof(ov13850_init_twolane_2112_1568_30fps) /
                        sizeof(ov13850_init_twolane_2112_1568_30fps[0]),
                .v_blanking_time_us = 3078,
                .ignore_measurement_check = 1, 
                PLTFRM_CAM_ITF_MIPI_CFG(0, 2, 640, 24000000)
        }    
};

wzyy2, 2017-10-30:
We are developing a new isp driver, maybe you could switch to that.
https://github.com/rockchip-linux/kernel/tree/isp-early-access

If you have the permission, you can create a technical support ticket and get help from our camera team.
Or I don't intend to support this driver. ; )

closed 2017-10-30 by wzyy2

zhujiongfu, 2017-10-30:
Thanks your attention, now, which sensor does the new driver support?

wzyy2, 2017-10-30:
You can check my commits, refer to rk3288-miniarm/rk3288-firefly dts.

zhujiongfu, 2017-10-30:
okay, thanks a lot.