rockchip-linux/kernel #48: RK3288 UART4 wrong defined at rk3288.dtsi

armoreno, 2017-11-17:
I want to enable the UART4 in my RK3288-Firefly board but it is not working at real pins level, although it does appear at kernel level as **/dev/ttyS4**. 

I've disabled the SPI0 node in the **rk3288-firefly.dtsi** file as follows:
```
&spi0 {
   pinctrl-names = "default";
   pinctrl-0 = <&spi0_clk>, <&spi0_cs0>, <&spi0_tx>, <&spi0_rx>, <&spi0_cs1>;
   status = "disabled";
};
```

And I've enabled the UART4 as follows:
```
&uart4 {
    status = "okay";
};
```

Then I've also checked with dmesg the UART4 is probed:
`
[    1.177999] ff1c0000.serial: ttyS4 at MMIO 0xff1c0000 (irq = 40, base_baud = 1500000) is a 16550A`

But when I write some characters to the UART4 using:

```
stty -F /dev/ttyS4 115200
echo hello world > /dev/ttyS4
```

Nothing is read from the pins side...
Therefore I think the UART4 node is wrong defined at the rk3288.dtsi file level.

armoreno, 2017-11-21:
Does anybody have any idea or suggestion about how to fix the UART4 problem with the Firefly-RK3288 board?

Many thanks in advance!

armoreno, 2017-11-24:
Well, finally I got the UART4 is working.
The rk3288.dtsi pinctrl assignment is wrong. The right pins for the UART4 are:
Rx: 5 - 15
Tx: 5 - 14
CTS: 5 - 12
RTS: 5 - 13

closed 2017-11-27 by armoreno

wzyy2, 2017-11-28:
What about sending the patch to `linux-rockchip@lists.infradead.org`.
https://www.kernel.org/doc/html/v4.12/process/submitting-patches.html

armoreno, 2017-11-28:
I was not sure where to send it... So, now I will send a patch soon.