rockchip-linux/kernel #50: AP6335  wifi模块 无限重启

nova-wenbo, 2017-11-21:
[   98.584324] dhd_bus_devreset: == WLAN ON ==
[   98.592262] F1 signature read @0x18000000=0x16224335
[   98.600516] F1 signature OK, socitype:0x1 chip:0x4339 rev:0x1 pkg:0x2
[   98.607316] DHD: dongle ram size is set to 786432(orig 786432) at 0x180000
[   98.614404] dhd_os_open_image: /system/etc/firmware/config.txt (30 bytes) open success
[   98.622386] dhd_conf_read_sdio_params: dhd_slpauto = 0
[   98.627520] dhd_conf_read_sdio_params: txglom_mode = 0
[   98.632673] Final fw_path=/system/etc/firmware/fw_bcm4339a0_ag.bin
[   98.638841] Final nv_path=/system/etc/firmware/nvram_AP6335.txt
[   98.644763] Final clm_path=/system/etc/firmware/clm_bcmdhd.blob
[   98.650683] Final conf_path=/system/etc/firmware/config.txt
[   98.656244] dhd_set_bus_params: set use_rxchain 0
[   98.660953] dhd_set_bus_params: set txglomsize 36
[   98.665962] dhd_os_open_image: /system/etc/firmware/fw_bcm4339a0_ag.bin (541314 bytes) open success
[   98.740337] dhd_os_open_image: /system/etc/firmware/nvram_AP6335.txt (2366 bytes) open success
[   98.749030] NVRAM version: AP6335_NVRAM_V1.7_04102014
[   98.754731] dhdsdio_write_vars: Download, Upload and compare of NVRAM succeeded.
[   99.760558] dhdsdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[   99.766756] dhd_bus_init: clock state is wrong. state = 1
[  100.770570] dhdsdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[  100.777374] dhd_net_bus_devreset: dhd_bus_devreset: -1
[  100.782634] dhd_wl_ioctl: returning as busstate=0
[  100.787344] dhd_net_bus_devreset: dhd_bus_devreset: -35
[  100.792866] wifi_platform_set_power = 0
[  100.796710] ======== PULL WL_REG_ON(-1) LOW! ========
[  100.801803] [WLAN_RFKILL]: rockchip_wifi_power: 0
[  100.806564] [BT_RFKILL]: rfkill_get_bt_power_state: rfkill-bt driver has not Successful initialized
[  100.920464] [WLAN_RFKILL]: wifi shut off power.
[  100.924991] wl_android_wifi_on: Failed
[  100.928737] dhd_open : wl_android_wifi_on failed (-1)
[  100.933803] dhd_stop: Enter ee329800
[  100.937372] wl_android_wifi_off in 1
[  100.941019] wl_android_wifi_off in 2: g_wifi_on=0, on_failure=1
[  100.946939] dhd_wl_ioctl: returning as busstate=0
[  100.951661] dhd_net_bus_devreset: dhd_bus_devreset: -35
[  100.956947] sdioh_disable_func_intr: error for read SDIO_CCCR_IENx : 0xffffff92
[  100.964279] wifi_platform_set_power = 0
[  100.968107] ======== PULL WL_REG_ON(-1) LOW! ========
[  100.973167] [WLAN_RFKILL]: rockchip_wifi_power: 0
[  100.977981] [BT_RFKILL]: rfkill_get_bt_power_state: rfkill-bt driver has not Successful initialized
[  101.090773] [WLAN_RFKILL]: wifi shut off power.
[  101.095377] wl_android_wifi_off out
[  101.099071] dhd_stop: Exit
[  101.102016] dhd_open: Exit ret=-1
SIOCSIFFLAGS: Operation not permitted

 iwconfig可以获取到wlan0 但是要up的时候 出现上面的情况

wzyy2, 2017-11-21:
It's a known issue. 
Our wifi driver can't enable "random mac" while Networkmanager use it by default, which make it keep restarting.

You can disable it in here.
https://github.com/rockchip-linux/rk-rootfs-build/blob/master/overlay/etc/NetworkManager/NetworkManager.conf

nova-wenbo, 2017-11-21:
您好，那为什么wifi不能up呢  up后就报这个错误：SIOCSIFFLAGS: Operation not permitted  还有就是关掉后不能再次打开

nova-wenbo, 2017-11-22:
[ 1130.798587] dhdsdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[ 1130.804797] dhd_bus_init: clock state is wrong. state = 1
[ 1131.808653] dhdsdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[ 1131.815832] dhd_net_bus_devreset: dhd_bus_devreset: -1
[ 1131.821167] dhd_wl_ioctl: returning as busstate=0
[ 1131.825896] dhd_net_bus_devreset: dhd_bus_devreset: -35
[ 1131.836238] wifi_platform_set_power = 0
[ 1131.840129] ======== PULL WL_REG_ON(-1) LOW! ========
[ 1131.845175] [WLAN_RFKILL]: rockchip_wifi_power: 0
[ 1131.851745] [BT_RFKILL]: rfkill_get_bt_power_state: rfkill-bt driver has not Successful initialized
[ 1131.860800] [WLAN_RFKILL]: wifi shut off power.
[ 1131.865322] wl_android_wifi_on: Failed
[ 1131.869074] dhd_open : wl_android_wifi_on failed (-1)
[ 1131.874116] dhd_stop: Enter ee9ec800
[ 1131.877682] wl_android_wifi_off in 1
[ 1131.882051] wl_android_wifi_off in 2: g_wifi_on=0, on_failure=1
[ 1131.887961] dhd_wl_ioctl: returning as busstate=0
[ 1131.892688] dhd_net_bus_devreset: dhd_bus_devreset: -35
[ 1131.898537] sdioh_disable_func_intr: error for read SDIO_CCCR_IENx : 0xffffff92
[ 1131.905834] wifi_platform_set_power = 0
[ 1131.909690] ======== PULL WL_REG_ON(-1) LOW! ========
[ 1131.914732] [WLAN_RFKILL]: rockchip_wifi_power: 0
[ 1131.919981] [BT_RFKILL]: rfkill_get_bt_power_state: rfkill-bt driver has not Successful initialized
[ 1131.929032] [WLAN_RFKILL]: wifi shut off power.
[ 1131.933556] wl_android_wifi_off out
[ 1131.937114] dhd_stop: Exit
[ 1131.939836] dhd_open: Exit ret=-1
SIOCSIFFLAGS: Operation not permitted

您好，ifconfig wlan0 up 并没有打开wifi 并出现以上log问题，应该定位到哪呢

wzyy2, 2017-11-22:
"dhd_net_bus_devreset: dhd_bus_devrese"
So it seems that it's not the same problem i mentioned.
I'm sorry i have no idea, you could ask it in redmine.

closed 2018-04-06 by hizukiayaka

llatecomers, 2020-06-28:
> [ 1130.798587] dhdsdio_htclk: HT Avail timeout (1000000): clkctl 0x50
> [ 1130.804797] dhd_bus_init: clock state is wrong. state = 1
> [ 1131.808653] dhdsdio_htclk: HT Avail timeout (1000000): clkctl [...]

我这边打开wifi也遇到这个情况，请问该怎么解决？