rockchip-linux/kernel #52: What is difference between rga image scaling and drm image scaling

nikitos1550, 2017-11-28:
Hello!

According rk3328 datasheet 2d graphics module (rga driver) and video output driver (accesed by drm) have image scaling capabilities. What is difference? Which one should I use in which case?

wzyy2, 2017-11-29:
RGA scale is memory2memoery. 
Buffer 800*600 --> Buffer 1920*1080

VOP scale is memory2display
Buffer 800*600 --> Screen 1920*1080

nikitos1550, 2017-11-29:
So, Am I right, that if I want just to scale decoded video than there is no big difference.
For example I want to scale 1920x1080 to 1280x720. Same performance with rga and vop. Right?

wzyy2, 2017-11-29:
No......
If you want to store your decoded video, then you have to use RGA sclae.
If you want to display decoded video in screen, the you can use VOP scale, but it only could be used through libdrm.

They don't have same performance, since VOP actually didn't scale anything.
To understand what it's VOP scale, you should learn `libdrm`, `drmmodesetplane`.
To learn what it's RGA sclae, https://github.com/wzyy2/rga-v4l2-demo.

nikitos1550, 2017-11-29:
Yes, I actually want just to display video not in fullscreen mode, but in some small region.
I was thinking about two possabilities, one mpp decoding -> rga (scale) -> vop or mpp decoding -> vop (scale).