rockchip-linux/kernel #54: drivers/mfd/fusb302.c fails to compile

itdaniher, 2017-12-20:
```
drivers/mfd/fusb302.c: In function ‘vdm_send_discoverysvid’:
drivers/mfd/fusb302.c:1164:3: warning: ‘memset’ used with length equal to number of elements without multiplication by element size [-Wmemset-elt-size]
error, forbidden warning: fusb302.c:1164
scripts/Makefile.build:266: recipe for target 'drivers/mfd/fusb302.o' failed
make[2]: *** [drivers/mfd/fusb302.o] Error 1
scripts/Makefile.build:411: recipe for target 'drivers/mfd' failed
make[1]: *** [drivers/mfd] Error 2
Makefile:999: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```

```gcc version 7.2.0 (Ubuntu/Linaro 7.2.0-8ubuntu3)```
```Ubuntu 17.10```

itdaniher, 2017-12-20:
@ayufan - verified the issue presents on this kernel too, not just at https://github.com/ayufan-rock64/linux-kernel/commit/00f25666cb08d5bba0ecada39d0e820122238f22

still no closer to a resolution...

ayufan, 2017-12-20:
Can you just try to fix this issue? It seems to be an easy to fix warning.

ayufan, 2017-12-20:
```
memset(chip->vdm_svid, 0, sizeof(chip->vdm_svid));
````

itdaniher, 2017-12-20:
@ayufan 's patch solves this compilation error.