rockchip-linux/kernel #60: RK3368 support

jkstrick, 2018-01-24:
Is RK3368 supported / currently working on v4.4?

I've been trying to get release-4.4 branch to work on a Sunchip CX-R8, which contains an RK3368, and which has a UART header on uart2 @0xff690000.  I also notice that build scripts in other rockchip-linux repose do not support RK3368 out-of-the-box.  But I do see dts files for several other RK3368-based boards. 

With a minimal DTS (rk3368-min.dts) which just includes rk3368.dtsi and enables uart2 and specifies bootargs="earlycon=uart8250,mmio32,0xff690000", and using the rockchip_linux_defconfig, I haven't been able to get even the startup banner output over uart2.

```
make mrproper
make CROSS_COMPILE=aarch64-linux-gnu- ARCH=arm64 rockchip_linux_defconfig
make -j4 CROSS_COMPILE=aarch64-linux-gnu- ARCH=arm64 rk3368-min.img
```

rk3368-min.dts:
```
/dts-v1/;
#include "rk3368.dtsi"

/ {
	model = "Sunchip CX-R8";
	compatible = "sunchip,cxr8", "rockchip,rk3368";

	chosen {
		bootargs = "earlycon=uart8250,mmio32,0xff690000";
	};
};

&uart2 {
	status = "okay";
};
```

wzyy2, 2018-01-30:
RK3368 sure can work on v4.4, but it's only available for android, it don;t have official linux support.
See http://opensource.rock-chips.com/wiki_Main_Page
close it.

closed 2018-01-30 by wzyy2

rubensa, 2019-05-20:
I've found [Tronsmart Orion R68 RK3368 TV Box Firmware and Android SDK](https://www.mediafire.com/folder/4ojpuo3s4m9ct/Tronsmart_Orion_R38_SDK_0713).
Could that be of some help on this?