rockchip-linux/kernel #61: DRM: HDR "Hybrid log-gamma" (HLG) not supported

mcerveny, 2018-01-26:
Hello.
DRM/drmlib reporting in connector properties in "HDR_PANEL_METADATA" EOTF=5 (Electro optical transfer functions - TRADITIONAL_GAMMA_SDR(bit 1) | SMPTE_ST2084(bit 3)) but output supports also HLG (bit 4), expecting EOTF=13 (0xd). See EDID:
```
# od -t x1 /sys/devices/platform/display-subsystem/drm/card0/card0-HDMI-A-1/edid
0000000 00 ff ff ff ff ff ff 00 1e 6d 01 00 01 01 01 01
0000020 01 1a 01 03 80 a0 5a 78 0a ee 91 a3 54 4c 99 26
0000040 0f 50 54 a1 08 00 31 40 45 40 61 40 71 40 81 80
0000060 01 01 01 01 01 01 08 e8 00 30 f2 70 5a 80 b0 58
0000100 8a 00 40 84 63 00 00 1e 02 3a 80 18 71 38 2d 40
0000120 58 2c 45 00 40 84 63 00 00 1e 00 00 00 fd 00 3a
0000140 3e 1e 88 3c 00 0a 20 20 20 20 20 20 00 00 00 fc
0000160 00 4c 47 20 54 56 0a 20 20 20 20 20 20 20 01 9f
0000200 02 03 58 f1 54 5d 10 1f 04 13 05 14 03 02 12 20
0000220 21 22 15 01 5e 5f 62 63 64 29 3d 06 c0 15 07 50
0000240 09 57 07 78 03 0c 00 20 00 b8 3c 20 c0 8a 01 02
0000260 03 04 01 40 7f f8 18 10 28 10 16 26 e2 00 cf e3
0000300 06 0d 01 e5 0e 60 65 61 66 ee 01 46 d0 00 24 18
0000320 09 00 ad 52 44 a9 23 0c 01 1d 80 18 71 1c 16 20
0000340 58 2c 25 00 40 84 63 00 00 9e 66 21 50 b0 51 00
0000360 1b 30 40 70 36 00 40 84 63 00 00 1e 00 00 00 09

#./edid-decode  /sys/devices/platform/display-subsystem/drm/card0/card0-HDMI-A-1/edid 
Extracted contents:
header:          00 ff ff ff ff ff ff 00
serial number:   1e 6d 01 00 01 01 01 01 01 1a
version:         01 03
basic params:    80 a0 5a 78 0a
chroma info:     ee 91 a3 54 4c 99 26 0f 50 54
established:     a1 08 00
standard:        31 40 45 40 61 40 71 40 81 80 01 01 01 01 01 01
descriptor 1:    08 e8 00 30 f2 70 5a 80 b0 58 8a 00 40 84 63 00 00 1e
descriptor 2:    02 3a 80 18 71 38 2d 40 58 2c 45 00 40 84 63 00 00 1e
descriptor 3:    00 00 00 fd 00 3a 3e 1e 88 3c 00 0a 20 20 20 20 20 20
descriptor 4:    00 00 00 fc 00 4c 47 20 54 56 0a 20 20 20 20 20 20 20
extensions:      01
checksum:        9f

EDID version: 1.3
Manufacturer: GSM Model 1 Serial Number 16843009
Made in week 1 of 2016
Digital display
Maximum image size: 160 cm x 90 cm
Gamma: 2.20
RGB color display
First detailed timing is preferred timing
Display x,y Chromaticity:
  Red:   0.6396, 0.3300
  Green: 0.2998, 0.5996
  Blue:  0.1503, 0.0595
  White: 0.3125, 0.3291
Established timings supported:
  720x400@70Hz 9:5 HorFreq: 31469 Hz Clock: 28.320 MHz
  640x480@60Hz 4:3 HorFreq: 31469 Hz Clock: 25.175 MHz
  800x600@60Hz 4:3 HorFreq: 37900 Hz Clock: 40.000 MHz
  1024x768@60Hz 4:3 HorFreq: 48400 Hz Clock: 65.000 MHz
Standard timings supported:
  640x480@60Hz 4:3 HorFreq: 31469 Hz Clock: 25.175 MHz
  800x600@60Hz 4:3 HorFreq: 37900 Hz Clock: 40.000 MHz
  1024x768@60Hz 4:3 HorFreq: 48400 Hz Clock: 65.000 MHz
  1152x864@60Hz 4:3
  1280x1024@60Hz 5:4 HorFreq: 64000 Hz Clock: 108.000 MHz
Detailed mode: Clock 594.000 MHz, 1600 mm x 900 mm
               3840 4016 4104 4400 hborder 0
               2160 2168 2178 2250 vborder 0
               +hsync +vsync 
               VertFreq: 60 Hz, HorFreq: 135000 Hz
Detailed mode: Clock 148.500 MHz, 1600 mm x 900 mm
               1920 2008 2052 2200 hborder 0
               1080 1084 1089 1125 vborder 0
               +hsync +vsync 
               VertFreq: 60 Hz, HorFreq: 67500 Hz
Monitor ranges (GTF): 58-62Hz V, 30-136kHz H, max dotclock 600MHz
Monitor name: LG TV
Has 1 extension blocks
Checksum: 0x9f (valid)

CTA extension block
Extension version: 3
84 bytes of CTA data
  Video data block
    VIC  93 3840x2160@24Hz 16:9  HorFreq: 54000 Hz Clock: 297.000 MHz
    VIC  16 1920x1080@60Hz 16:9  HorFreq: 67500 Hz Clock: 148.500 MHz
    VIC  31 1920x1080@50Hz 16:9  HorFreq: 56250 Hz Clock: 148.500 MHz
    VIC   4 1280x720@60Hz 16:9  HorFreq: 45000 Hz Clock: 74.250 MHz
    VIC  19 1280x720@50Hz 16:9  HorFreq: 37500 Hz Clock: 74.250 MHz
    VIC   5 1920x1080i@60Hz 16:9  HorFreq: 33750 Hz Clock: 74.250 MHz
    VIC  20 1920x1080i@50Hz 16:9  HorFreq: 28125 Hz Clock: 74.250 MHz
    VIC   3 720x480@60Hz 16:9  HorFreq: 31469 Hz Clock: 27.000 MHz
    VIC   2 720x480@60Hz 4:3  HorFreq: 31469 Hz Clock: 27.000 MHz
    VIC  18 720x576@50Hz 16:9  HorFreq: 31250 Hz Clock: 27.000 MHz
    VIC  32 1920x1080@24Hz 16:9  HorFreq: 27000 Hz Clock: 74.250 MHz
    VIC  33 1920x1080@25Hz 16:9  HorFreq: 28125 Hz Clock: 74.250 MHz
    VIC  34 1920x1080@30Hz 16:9  HorFreq: 33750 Hz Clock: 74.250 MHz
    VIC  21 1440x576i@50Hz 4:3  HorFreq: 15625 Hz Clock: 27.000 MHz
    VIC   1 640x480@60Hz 4:3  HorFreq: 31469 Hz Clock: 25.175 MHz
    VIC  94 3840x2160@25Hz 16:9  HorFreq: 56250 Hz Clock: 297.000 MHz
    VIC  95 3840x2160@30Hz 16:9  HorFreq: 67500 Hz Clock: 297.000 MHz
    VIC  98 4096x2160@24Hz 256:135  HorFreq: 54000 Hz Clock: 297.000 MHz
    VIC  99 4096x2160@25Hz 256:135  HorFreq: 56250 Hz Clock: 297.000 MHz
    VIC 100 4096x2160@30Hz 256:135  HorFreq: 67500 Hz Clock: 297.000 MHz
  Audio data block
    DTS, max channels 6
      Supported sample rates (kHz): 48 44.1
      Maximum bit rate: 1536 kHz
    AC-3, max channels 6
      Supported sample rates (kHz): 48 44.1 32
      Maximum bit rate: 640 kHz
    Linear PCM, max channels 2
      Supported sample rates (kHz): 192 96 48 44.1 32
      Supported sample sizes (bits): 24 20 16
  Vendor-specific data block, OUI 000c03 (HDMI)
    Source physical address 2.0.0.0
    Supports_AI
    DC_36bit
    DC_30bit
    DC_Y444
    Maximum TMDS clock: 300MHz
    Extended HDMI video details:
      3D present
      3D-capable-VIC mask present
      HDMI VIC 1 3840x2160@30Hz 16:9 HorFreq: 67500 Hz Clock: 297.000 MHz
      HDMI VIC 2 3840x2160@25Hz 16:9 HorFreq: 56250 Hz Clock: 297.000 MHz
      HDMI VIC 3 3840x2160@24Hz 16:9 HorFreq: 54000 Hz Clock: 297.000 MHz
      HDMI VIC 4 4096x2160@24Hz 256:135 HorFreq: 54000 Hz Clock: 297.000 MHz
      3D: Side-by-side (half, horizontal)
      3D: Top-and-bottom
      3D VIC indices: 3 4 5 6 7 8 9 10 11 12 13 14
      VIC index 1 supports side-by-side (half, horizontal)
      VIC index 2 supports side-by-side (half, horizontal)
      VIC index 1 supports top-and-bottom
      VIC index 2 supports top-and-bottom
  Extended tag: Video capability data block
    YCbCr quantization: Selectable (via AVI YQ) (1)
    RGB quantization: Selectable (via AVI Q) (1)
    PT scan behaviour: No Data (0)
    IT scan behaviour: Support both over- and underscan (3)
    CE scan behaviour: Support both over- and underscan (3)
  Extended tag: HDR static metadata data block
    Electro optical transfer functions:
      Traditional gamma - SDR luminance range
      SMPTE ST2084
      Hybrid Log-Gamma
    Supported static metadata descriptors:
      Static metadata type 1
  Extended tag: YCbCr 4:2:0 video data block
    VIC  96 3840x2160@50Hz 16:9  HorFreq: 112500 Hz Clock: 297.000 MHz
    VIC 101 4096x2160@50Hz 256:135  HorFreq: 112500 Hz Clock: 297.000 MHz
    VIC  97 3840x2160@60Hz 16:9  HorFreq: 135000 Hz Clock: 297.000 MHz
    VIC 102 4096x2160@60Hz 256:135  HorFreq: 135000 Hz Clock: 297.000 MHz
  Extended tag: Vendor-specific video data block
Underscans PC formats by default
Basic audio support
Supports YCbCr 4:4:4
Supports YCbCr 4:2:2
1 native detailed modes
Detailed mode: Clock 74.250 MHz, 1600 mm x 900 mm
               1920 2008 2052 2200 hborder 0
                540  542  547  562 vborder 0
               +hsync +vsync interlaced 
               VertFreq: 60 Hz, HorFreq: 33750 Hz
Detailed mode: Clock 85.500 MHz, 1600 mm x 900 mm
               1360 1424 1536 1792 hborder 0
                768  771  777  795 vborder 0
               +hsync +vsync 
               VertFreq: 60 Hz, HorFreq: 47712 Hz
Checksum: 0x9 (valid)

One or more of the timings is out of range of the Monitor Ranges:
  Vertical Freq: 24 - 70 Hz
  Horizontal Freq: 15625 - 135000 Hz
  Maximum Clock: 594.000 MHz
```

Kwiboo, 2018-01-27:
@mcerveny https://github.com/Kwiboo/linux-rockchip/commit/2fffa8d339df6e433428de3ea165072200e5e531 should fix the HLG detection, `eotf_supported` can alternatively be changed to use `val = edid_ext[2] & SUPPORTED_EOTF_MASK` instead of checking each bit separately.

mcerveny, 2018-01-27:
Thanks and closing the issue due to lost confidence in Rockchip solution.

closed 2018-01-27 by mcerveny

wzyy2, 2018-01-29:
Not familiar with HDMI stuff. If you have unresolved questions, you can ask the board vendor, i think they can send the issues to the corresponding engineers.

mcerveny, 2018-01-29:
You should check your own code. I cannot find reference of SDR2HDR_FOR_HLG_HDR (vop_load_sdr2hdr_table()) and/or extend comparison like "eotf == SMPTE_ST2084" also to HLG...