rockchip-linux/kernel #65: /dev/mem can not be used to modify the GPIO registers

wuyuanyi135, 2018-02-06:
Version: Linux rock64 4.4.77-rockchip-ayufan-136

I have tried to access /dev/mem for fast GPIO operation. For example, from the documentation, the GPIO3 is bound to the base address 0xff240000. 

1. Accessing 0xff240000 was fine and returning valid register data
2. Accessing the direction register (offset = 4bytes) threw error code = 22 (Invalid argument). This error applied to all offset access. Only base address of each device was accessible.
3. Writing to 0xff240000 did not save
4. Reading more than 4 bytes from 0xff240000 resulted in repeated value of the first 4 bytes.

Is there any working example for this board to access and operate /dev/mem using mmap or so?

Many thanks

jakob-tsd, 2018-02-06:
Yes. Use devmem2: https://github.com/radii/devmem2/blob/master/devmem2.c

wuyuanyi135, 2018-02-06:
@jakob-tsd Thank you for your suggestion. But I got the same problem after execute it.

```
$sudo bash -c "echo 101 > /sys/class/gpio/export"
$sudo bash -c "echo out > /sys/class/gpio/gpio101/direction"
```

```
$ sudo ./main 0xff240000 w 0xffffffff
/dev/mem opened.
Memory mapped at address 0x7f7f373000.
Value at address 0xFF240000 (0x7f7f373000): 0x0
Written 0xFFFFFFFF; readback 0x0
```

wuyuanyi135, 2018-02-06:
If I do not perform those export the mem map is working but I suppose the register is actually not mapped. I will try to verify it later

closed 2018-04-09 by hizukiayaka