rockchip-linux/kernel #67: kernel panic 4.4.112

sndwvs, 2018-02-11:
Distributions: Slackware-current
booting: eMMC
glibc: 2.27
kernel: 4.4.112 (4.4.103 of this error was not found)
board: firefly-rk3399
arch: aarch64
drm: [0e756ede215ec8193de6cfec80e4cc6a806abbaa](https://github.com/rockchip-linux/libdrm-rockchip/commit/0e756ede215ec8193de6cfec80e4cc6a806abbaa)
xserver: [b304068103590b4d8f967469d181b625bc638ae6](https://github.com/rockchip-linux/xserver/commit/b304068103590b4d8f967469d181b625bc638ae6)
xorg driver: [glamor: EGL version 1.4 Midgard-"r13p0-00rel0"](https://github.com/rockchip-linux/libmali/tree/rockchip/lib/aarch64-linux-gnu)

xorg starts fine, but when logging in to XFCE we get an error:

`Message from syslogd@firefly-rk3399 at Sun Feb 11 03:14:41 2018 ...
firefly-rk3399 kernel: [  116.216667] BUG: spinlock lockup suspected on CPU#0, swapper/0/0
Message from syslogd@firefly-rk3399 at Sun Feb 11 03:14:41 2018 ...
firefly-rk3399 kernel: [  116.216874]  lock: 0xffffffc0ed5a83e8, .magic: dead4ead, .owner: mali-cmar-backe/801, .owner_cpu: 2
Message from syslogd@firefly-rk3399 at Sun Feb 11 03:14:41 2018 ...
firefly-rk3399 kernel: [  116.216867] BUG: spinlock lockup suspected on CPU#4, Xorg/792
Message from syslogd@firefly-rk3399 at Sun Feb 11 03:14:41 2018 ...
firefly-rk3399 kernel: [  116.220361]  lock: 0xffffffc0ed5a83e8, .magic: dead4ead, .owner: mali-cmar-backe/801, .owner_cpu: 2
`

lisa, 2019-02-19:
I encountered this with 4.4.167 - dump is up at https://gist.github.com/lisa/972b27c561e1fb58a6d268d24de88a81#file-crashlog-txt

MichaIng, 2019-06-23:
Other chip, other way of replication but the same error with this rockchip64 kernel 4.4.180: https://dietpi.com/phpbb/viewtopic.php?f=11&t=6062