rockchip-linux/kernel #72: camera doesn't work on tinker board (OV5647)

chwe17, 2018-03-08:
Today I built a new image with your buildscript using the provided rootfs from your google drive (_rootfs-debian-20171225(fix a stupid mistake, affect performance since 20170901).tar.gz_ as a side note filenames with spaces and '(' are nasty to work in console - maybe you can rename it?). The image built without issues and the tinker board boots. 
Unfortunately, the camera is not available. Dmesg showed that probing of the camera fails:
`linaro@linaro-alip:~$ dmesg | grep ov56
[    3.664487] OF: graph: no port node found in /i2c@ff660000/ov5647@36
[    3.677146] ov5647 2-0036: DT parsing error: -22
[    3.687968] ov5647: probe of 2-0036 failed with error -22
`
and as expected camera_test.sh fails too:
`linaro@linaro-alip:/usr/local/bin$ sudo ./test_camera.sh
Start MIPI CSI Camera Preview!
./test_camera.sh: 9: ./test_camera.sh: cannot create /sys/module/video_rkisp1/parameters/rkisp1_debug: Permission denied
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstRkXImageSink:rkximagesink0: GStreamer error: state change failed and some element failed to post a proper error message with the reason for the failure.
Additional debug info:
gstbasesink.c(5184): gst_base_sink_change_state (): /GstPipeline:pipeline0/GstRkXImageSink:rkximagesink0:
Failed to start
Setting pipeline to NULL ...
Freeing pipeline ...
`
a complete syslog can be found here:
[http://ix.io/UeR](http://ix.io/UeR)
and kernel config here (I didn't made any changes to default):
[http://ix.io/Uff](http://ix.io/Uff)

Any help to get in working is appreciated. 
(I add @Tonymac32 to this issue, since we're both interested in getting up the camera)

Tonymac32, 2018-03-09:
For my part I have been working with the Armbian builds, I've modified the config to use the ISP1 and mipi, camera light powers on, but there is an immediate "oops" on driver probe.  I get the same fault with the ov5647 and the imx219.

wzyy2, 2018-03-09:
[ 3.677146] ov5647 2-0036: DT parsing error: -22 [ 3.687968] ov5647: probe of 2-0036 failed with error -22

Could you update the kernel to the latest?

chwe17, 2018-03-09:
the kernel was freshly built today:
https://github.com/rockchip-linux/kernel.git

`linaro@linaro-alip:~$ uname -r
4.4.114
`
TinkerOS which uses ISP10 and some tweaks works with this camera& tinkerboard, so I don't think it's a hardware issue.

wzyy2, 2018-03-09:
You need to edit dts to make ov5647 work.

https://github.com/rockchip-linux/kernel/blob/release-4.4/arch/arm/boot/dts/rk3288-miniarm.dts#L451
https://github.com/rockchip-linux/kernel/blob/release-4.4/Documentation/devicetree/bindings/media/rockchip-mipi-dphy.txt

Tonymac32, 2018-03-09:
My results with my build (parallel effort to chwe17)

https://hastebin.com/lapivulaxa.xml

My I2C device tree entries for the camera:  

https://hastebin.com/suseqahito.pl

I changed the imx219_out label to camera_out (in all necessary places) to help aid readability.  Other than that...

wzyy2, 2018-03-09:
```
	camera0: ov5647@36 {
		compatible = "ovti,ov5647";
		reg = <0x36>;
		clocks = <&ext_cam_clk>;
		status = "okay";

		port {
			camera_out: endpoint {
				remote-endpoint = <&mipi_rx0_in>;
				data-lanes = <1 2>;
			};
		};

	};

	camera1: imx219@10 {
		compatible = "sony,imx219";
		reg = <0x10>;
		clocks = <&ext_cam_clk>;
		status = "okay";
	};

```

Tonymac32, 2018-03-09:
Well, that wasn't what was causing my kernel oops, the errors due to ov5647 are gone, but (I have imx219):

[    5.030378] imx219 2-0010: Reading register 100 from 10 failed
[    5.047697] imx219 2-0010: Model ID 0x0219, Lot ID 0x718e37, Chip ID 0x0662
[    5.053022] ERROR: Bad of_node_put() on /isp@ff910000/port

I'll swap and test ov5647 and get back to you.

Tonymac32, 2018-03-09:
Got the same with the ov5647:

[    5.221070] ERROR: Bad of_node_put() on /isp@ff910000/port

As I am actually getting a crash I will continue to review and see what is different in my config/etc.

chwe17, 2018-03-09:
@wzyy2 
changing to your suggested entry doesn't fails when the kernel is builded:

`opi@Buildmachine:~/rk_build$ build/mk-kernel.sh rk3288-miniarm
 Building kernel for rk3288-miniarm board!
 Using rockchip_linux_defconfig
#
# configuration written to .config
#
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
make[1]: 'include/generated/mach-types.h' is up to date.
  CHK     scripts/mod/devicetable-offsets.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  GZIP    kernel/config_data.gz
  CHK     kernel/config_data.h
  DTC     arch/arm/boot/dts/rk3288-miniarm.dtb
ERROR (phandle_references): Reference to non-existent node or label "mipi_rx0_in                            "

ERROR (phandle_references): Reference to non-existent node or label "imx219_out"

ERROR: Input tree has errors, aborting (use -f to force output)
scripts/Makefile.lib:299: recipe for target 'arch/arm/boot/dts/rk3288-miniarm.dt                            b' failed
make[1]: *** [arch/arm/boot/dts/rk3288-miniarm.dtb] Error 2
arch/arm/Makefile:340: recipe for target 'dtbs' failed
make: *** [dtbs] Error 2
make: *** Waiting for unfinished jobs....
 MAKE KERNEL IMAGE FAILED.
`
replacing it to:

`	camera0: ov5647@36 {
		compatible = "ovti,ov5647";
		reg = <0x36>;
		clocks = <&ext_cam_clk>;
		status = "oaky";
		port {
			imx219_out: endpoint {
				remote-endpoint = <&imx219_in>;
				data-lanes = <1 2>;
			};
		};
	};

	camera1: imx219@10 {
		compatible = "sony,imx219";
		reg = <0x10>;
		clocks = <&ext_cam_clk>;
		status = "disabled";`

builds but camera isn't recognized.

`linaro@linaro-alip:~$ dmesg | grep ov56
linaro@linaro-alip:~$
`

`linaro@linaro-alip:~$ dmesg | grep i2c
[    1.541503] i2c i2c-6: of_i2c: modalias failure on /hdmi@ff980000/ports
[    3.065533] i2c /dev entries driver
[    3.286416] of_get_named_gpiod_flags: parsed 'dvs-gpios' property of node '/i2c@ff650000/pmic@1b[0]' - status (0)
[    3.286732] of_get_named_gpiod_flags: parsed 'dvs-gpios' property of node '/i2c@ff650000/pmic@1b[1]' - status (0)
[    3.519270] rk3x-i2c ff650000.i2c: Initialized RK3xxx I2C bus at f0c26000
[    3.533604] rk3x-i2c ff150000.i2c: Initialized RK3xxx I2C bus at f0c28000
[    3.547960] rk3x-i2c ff160000.i2c: Initialized RK3xxx I2C bus at f0c2a000
[    3.562338] rk3x-i2c ff660000.i2c: Initialized RK3xxx I2C bus at f0c2c000
`
do you have any suggestions?

wzyy2, 2018-03-09:
Try issuing "media-ctl -p"

chwe17, 2018-03-09:
```
linaro@linaro-alip:~$ media-ctl -p
-bash: media-ctl: command not found
```
do I need additional packages?

wzyy2, 2018-03-09:
v4l2-utils

chwe17, 2018-03-09:
with the above config:

```
linaro@linaro-alip:~/v4l-utils/utils/media-ctl$ ./media-ctl -p
Media controller API version 0.1.0

Media device information
------------------------
driver          rkisp1
model           rkisp1
serial
bus info
hw revision     0x0
driver version  0.0.0

Device topology
- entity 1: rkisp1-isp-subdev (4 pads, 0 link)
            type V4L2 subdev subtype Unknown flags 0
        pad0: Sink
        pad1: Sink
        pad2: Source
        pad3: Source

- entity 2: rkisp1_selfpath (1 pad, 0 link)
            type Node subtype V4L flags 0
            device node name /dev/video0
        pad0: Sink

- entity 3: rkisp1_mainpath (1 pad, 0 link)
            type Node subtype V4L flags 0
            device node name /dev/video1
        pad0: Sink

- entity 4: rkisp1-statistics (1 pad, 0 link)
            type Node subtype V4L flags 0
            device node name /dev/video2
        pad0: Sink

- entity 5: rkisp1-input-params (1 pad, 0 link)
            type Node subtype V4L flags 0
            device node name /dev/video3
        pad0: Source

- entity 6: rockchip-sy-mipi-dphy (2 pads, 0 link)
            type V4L2 subdev subtype Unknown flags 0
        pad0: Sink
        pad1: Source
```

change to (remove the whole imx219 entry):
```
&i2c2 {
	status = "okay";

	camera0: ov5647@36 {
		compatible = "ovti,ov5647";
		reg = <0x36>;
		clocks = <&ext_cam_clk>;
		status = "okay";
		
		port {
			imx219_out: endpoint {
				remote-endpoint = <&imx219_in>;
				data-lanes = <1 2>;
			};
		};
	};

	m24c08@50 {
		compatible = "at,24c08";
		reg = <0x50>;
	};
};
```

gives: 
```
linaro@linaro-alip:~/v4l-utils/utils/media-ctl$ ./media-ctl -p
Media controller API version 0.1.0

Media device information
------------------------
driver          rkisp1
model           rkisp1
serial
bus info
hw revision     0x0
driver version  0.0.0

Device topology
- entity 1: rkisp1-isp-subdev (4 pads, 5 links)
            type V4L2 subdev subtype Unknown flags 0
            device node name /dev/v4l-subdev0
        pad0: Sink
                [fmt:SBGGR10_1X10/800x600 field:none
                 crop.bounds:(0,0)/800x600
                 crop:(0,0)/800x600]
                <- "rockchip-sy-mipi-dphy":1 [ENABLED]
        pad1: Sink
                <- "rkisp1-input-params":0 [ENABLED]
        pad2: Source
                [fmt:SBGGR10_1X10/800x600 field:none
                 crop.bounds:(0,0)/800x600
                 crop:(0,0)/800x600]
                -> "rkisp1_selfpath":0 [ENABLED]
                -> "rkisp1_mainpath":0 [ENABLED]
        pad3: Source
                -> "rkisp1-statistics":0 [ENABLED]

- entity 2: rkisp1_selfpath (1 pad, 1 link)
            type Node subtype V4L flags 0
            device node name /dev/video0
        pad0: Sink
                <- "rkisp1-isp-subdev":2 [ENABLED]

- entity 3: rkisp1_mainpath (1 pad, 1 link)
            type Node subtype V4L flags 0
            device node name /dev/video1
        pad0: Sink
                <- "rkisp1-isp-subdev":2 [ENABLED]

- entity 4: rkisp1-statistics (1 pad, 1 link)
            type Node subtype V4L flags 0
            device node name /dev/video2
        pad0: Sink
                <- "rkisp1-isp-subdev":3 [ENABLED]

- entity 5: rkisp1-input-params (1 pad, 1 link)
            type Node subtype V4L flags 0
            device node name /dev/video3
        pad0: Source
                -> "rkisp1-isp-subdev":1 [ENABLED]

- entity 6: rockchip-sy-mipi-dphy (2 pads, 2 links)
            type V4L2 subdev subtype Unknown flags 0
            device node name /dev/v4l-subdev1
        pad0: Sink
                [fmt:SBGGR8_1X8/1296x972 field:none]
                <- "ov5647 2-0036":0 [ENABLED]
        pad1: Source
                [fmt:SBGGR8_1X8/1296x972 field:none]
                -> "rkisp1-isp-subdev":0 [ENABLED]

- entity 7: ov5647 2-0036 (1 pad, 1 link)
            type V4L2 subdev subtype Sensor flags 0
            device node name /dev/v4l-subdev2
        pad0: Source
                [fmt:SBGGR8_1X8/1296x972 field:none]
                -> "rockchip-sy-mipi-dphy":0 [ENABLED]
```