rockchip-linux/kernel #75: Booting problem with RK3066

sziszi, 2018-03-22:
I try to boot an RK3066 device with this kernel and i cannot succeed. I tried two scenarios:

**1: I followed the guide from here http://opensource.rock-chips.com/wiki_Rockchip_Kernel**
 
  After i successfully build a boot image and flash it to the boot partition i got an error like this:
  
DDR Version 1.04 20140217
In
SRX
DDR3
300MHz
Bus Width=32 Col=10 Bank=8 Row=15 CS=1 Die Bus-Width=8 Size=1024MB
OUT
BUILD=====2
No.1 FLASH ID:ec d7 94 7a 54 43
OK! 252353
unsigned!
SecureBootEn = 0 0
Boot ver: 2014-06-27#2.15
start_linux=====259477
E:Invaid tag(0x4D524150)!
Load failed!
E:Invaid tag(0x4D524150)!
Load failed!
UsbBoot 265826
UsbHook ...920093
powerOn 920119
 1075219 UsbConnected
 1189061 UsbConnected

**2: My second approach is the following**

cat linux/arch/arm/boot/zImage linux/arch/arm/boot/dts/rk3066a-mk808.dtb > flash/zImage-dtb

mkbootimg --kernel flash/zImage-dtb --ramdisk /dev/null --base 60800000 -o flash/boot.img

mv flash/boot.img flash/boot.img-old
rkcrc flash/boot.img-old flash/boot.img

Then flash the boot.img to the device and get the following log..

start_linux=====281959
 2430047 Starting kernel...@0x60808000

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.103 (root@ubuntu-docker) (gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.9) ) #1 SMP Thu Mar 22 17:12:31 EET 2018
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: MarsBoard RK3066
[    0.000000] cma: Reserved 16 MiB at 0x9f000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] PERCPU: Embedded 14 pages/cpu @ef7c2000 s24728 r8192 d24424 u57344
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 260608
[    0.000000] Kernel command line:  console=tty0 console=ttyS2,115200n8 earlyprintk root=/dev/mtdblock2 rw rootfstype=ext4 init=/sbin/init initrd=0x62000000,0x00040000 mtdparts=rk29xxnand:0x00002000@0x00000000(parameter),0x00002000@0x00002000(boot),0x00002000@0x00004000(loader2),0x00002000@0x00006000(atf),0x00038000@0x00008000(boot:bootable),-@0x0040000(rootfs) bootver=2014-06-27#2.15 firmware_ver=6.0.0
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 1002656K/1048576K available (12288K kernel code, 942K rwdata, 3252K rodata, 1024K init, 633K bss, 29536K reserved, 16384K cma-reserved, 245760K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0d00000   (13280 kB)
[    0.000000]       .init : 0xc1200000 - 0xc1300000   (1024 kB)
[... 339 lines cut ...]
[    3.749398] ======== dhd_wlan_init_plat_data ========
[    3.749402] dhd_wlan_init_gpio: GPIO(WL_REG_ON) = -1
[    3.749406] dhd_wlan_init_gpio: GPIO(WL_HOST_WAKE) = -1
[    3.749409] [WLAN_RFKILL]: rockchip_wifi_get_oob_irq: Enter
[    3.749414] [WLAN_RFKILL]: rockchip_wifi_get_oob_irq: rfkill-wlan driver has not Successful initialized
[    3.749416] dhd_wlan_init_gpio: host_oob_irq: -1
[    3.749420] dhd_wlan_init_gpio: host_oob_irq_flags=0x414
[    3.749422] dhd_wifi_platform_load: Enter
[    3.749450] Power-up adapter 'DHD generic adapter'
[    3.749497] wifi_platform_set_power = 1
[    3.749502] ======== PULL WL_REG_ON(-1) HIGH! ========
[    3.749505] [WLAN_RFKILL]: rockchip_wifi_power: 1
[    3.749509] [WLAN_RFKILL]: rockchip_wifi_power: rfkill-wlan driver has not Successful initialized
[    3.858567] tps65910-rtc tps65910-rtc: setting system clock to 2000-01-01 00:00:01 UTC (946684801)
[    3.868506] I : [File] : drivers/gpu/arm/mali400/mali/linux/mali_kernel_linux.c; [Line] : 414; [Func] : mali_module_init(); svn_rev_string_from_arm of this mali_ko is '', rk_ko_ver is '5', built at '17:00:42', on 'Mar 22 2018'.
[    3.868761] Mali: Mali device driver loaded
[    3.869002] ALSA device list:
[    3.869005]   No soundcards found.
[    3.982197] RAMDISK: Couldn't find valid RAM disk image starting at 0.
[    3.989689] VFS: Cannot open root device "mtdblock2" or unknown-block(0,0): error -6
[    3.998321] Please append a correct "root=" boot option; here are the available partitions:
[    4.007615] 0100            4096 ram0  (driver?)
[    4.012773] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)

Please help i have no more ideas

Laszlo-Fiat, 2018-03-25:
Hello,
The first approach cannot work because rk3066 is not supported by u-boot. 
The second approach is OK, it started to boot, it just couldn't find your rootfs: 
[ 3.989689] VFS: Cannot open root device "mtdblock2" or unknown-block(0,0): error -6

mtdblock* would the the on-board NAND chip, but that would have to be configured from DTS/DTB, and I am not sure it works, I have never tried it.

I suggest you follow this approach: put your rootfs on the 1st partition of the SD-Card. Correct the CMDLINE: root=/dev/mmcblk0p1 rootwait

It works for me that way.

sziszi, 2018-03-25:
Thanks Laszlo

Unfortulately my sd card reader is not mounting also. I do not understand
why...

 ..........................................................
 3.193497] Mali: Mali device driver loaded
[    3.193750] ALSA device list:
[    3.193753]   No soundcards found.
[    3.318291] mmc_host mmc0: Bus speed (slot 0) = 3093750Hz (slot req
100000Hz, actual 96679HZ div = 16)
[    3.339305] wifi_platform_bus_enumerate device present 1
[    3.339308] ======== Card detection to detect SDIO card! ========
[    3.339312] sdio: host isn't  initialization successfully.
[    3.508283] mmc_host mmc0: Bus speed (slot 0) = 3093750Hz (slot req
400000Hz, actual 386718HZ div = 4)
[    3.552003] ttyS2 - failed to request DMA
[    3.556707] RAMDISK: Couldn't find valid RAM disk image starting at 0.
[    3.564055] Waiting for root device /dev/mmcblk0p1...
[    3.658288] mmc_host mmc0: Bus speed (slot 0) = 3093750Hz (slot req
300000Hz, actual 257812HZ div = 6)
[    3.759600] usb 2-1: New USB device found, idVendor=05e3, idProduct=0608
[    3.767040] usb 2-1: New USB device strings: Mfr=0, Product=1,
SerialNumber=0
[    3.774972] usb 2-1: Product: USB2.0 Hub
[    3.780286] hub 2-1:1.0: USB hub found
[    3.784757] hub 2-1:1.0: 4 ports detected
[    3.818293] mmc_host mmc0: Bus speed (slot 0) = 3093750Hz (slot req
200000Hz, actual 193359HZ div = 8)
[    3.978290] mmc_host mmc0: Bus speed (slot 0) = 3093750Hz (slot req
100000Hz, actual 96679HZ div = 16)
[    5.328287] failed to power up DHD generic adapter, 1 retry left
[    5.351016] wifi_platform_set_power = 0
[    5.355263] ======== PULL WL_REG_ON(-1) LOW! ========
[    5.360896] [WLAN_RFKILL]: rockchip_wifi_power: 0
[    5.366109] [WLAN_RFKILL]: rockchip_wifi_power: rfkill-wlan driver has
not Successful initialized
[    5.375963] wifi_platform_bus_enumerate device present 0
[    5.381862] ======== Card detection to remove SDIO card! ========
[    5.388626] sdio: host isn't  initialization successfully.
[    5.394772] wifi_platform_set_power = 1
[    5.399045] ======== PULL WL_REG_ON(-1) HIGH! ========
[    5.404738] [WLAN_RFKILL]: rockchip_wifi_power: 1
[    5.409964] [WLAN_RFKILL]: rockchip_wifi_power: rfkill-wlan driver has
not Successful initialized
[    5.728278] wifi_platform_bus_enumerate device present 1
[    5.734165] ======== Card detection to detect SDIO card! ========
[    5.740933] sdio: host isn't  initialization successfully.
[    5.898277] vendor storage:20160801 ret = -1
[    7.738282] failed to power up DHD generic adapter, 0 retry left
[    7.761002] wifi_platform_set_power = 0
[    7.765256] ======== PULL WL_REG_ON(-1) LOW! ========
[    7.770876] [WLAN_RFKILL]: rockchip_wifi_power: 0
[    7.776089] [WLAN_RFKILL]: rockchip_wifi_power: rfkill-wlan driver has
not Successful initialized
[    7.785940] wifi_platform_bus_enumerate device present 0
[    7.791844] ======== Card detection to remove SDIO card! ========
[    7.798611] sdio: host isn't  initialization successfully.
[    7.804688] failed to power up DHD generic adapter, max retry reached**
[    7.812117] unregister wifi platform drivers
[    7.816852] wifi_platform_bus_enumerate device present 0
[    7.822767] ======== Card detection to remove SDIO card! ========
[    7.829538] sdio: host isn't  initialization successfully.
[    7.835615] ======== dhd_wlan_deinit_plat_data ========
[    7.841426] dhd_module_init: Failed to load the driver, try cnt 0
[    7.848224] dhd_module_init: Failed to load driver max retry reached**
[    7.855477] dhd_module_init: Exit err=-19

wzyy2, 2018-03-26:
rk3066 is not supported in 4.4 kernel

closed 2018-03-26 by wzyy2

sziszi, 2018-03-27:
Jacob Chen sorry but...you are outside of your own box?

http://opensource.rock-chips.com/wiki_Rockchip_Kernel

Rockchip kernel 4.4 supports:

RK3036, RK3066, RK312X, RK3188, RK322X,RK3288
<http://opensource.rock-chips.com/wiki_RK3288>, RK3328
<http://opensource.rock-chips.com/wiki_RK3328>, RK3368, RK3399
<http://opensource.rock-chips.com/wiki_RK3399>

I see the RK3066 here!

Also when i look inside rockchip_linux_defconfig i see
CONFIG_ROCKCHIP_RK3066_HDMI=y

What about the rk3066 device tree in 4.4 kernel? You have 3 devices there!

Laszlo-Fiat, 2018-03-28:
Hello sziszi,
rk3066 and rk3188 support came into rockchip-linux kernel via mainline/kernel.org kernel. There is a basic support for these old SOCs in mainline kernel, cpu, usb, sd-card works, hdmi, vpu, nand does not work. This support was mainlined by the linux-rockchip guys, who may have used the leaked rockchip 3.0.8+ and 3.0.36+ kernel sources as reference to write their drivers for mainline linux.
Rockchip-linux mainly focuses adding support for new SOCs, currently rk3328, rk3399, rk3308. 
That is why I was quite surprised to see HDMI code appearing for RK3066 in rockchip-linux. I am thankful for it.
sziszi: What kind of hardware do you have? What do you want to use it for?
I ask because one DTS file describe one type of board, with GPIOs, and Power Management IC (voltages), that can be different on each type of board. If you use a DTS/DTB file that is not for your board, you may damage your hardware.

Regarding your SD-Card problem, I managed to solve it a few months ago. The rk3066a-marsboard.dts came directly from mainline linux, and it was not customized for rockchip-linux. You should base your DTS file on rk3066a-rayeager.dts, as it has much more components enabled. In case of SD-Card, please look at both DTS files, look for the &mmc0 definition, and copy the missing lines from rayeager to marsboard. It should find the rootfs on your SD-Card after that. It did find it for me.

Regarding HDMI (if you want that): HDMI console worked for me, but I couldn't manage to start X in 1920x1080, I got an out of memory error (surely some bug), it worked for 1280x720. The rockchip rk3066 hdmi driver only likes TV timing modes (CEA modes), and not monitor timing modes (DMT modes). So it worked with a native HDMI display (in 720p), but it rejected all the modes of my DVI monitor. I managed to make it work on my monitor by adding 
video=HDMI-A-1:1920x1080@60 to the CMDLINE, that way the driver got a resolution, and it worked out a mode that was close enough to a CEA mode timing, something like 1920x1080@52.
That was around mid-February, with rockchip-linux v4.4.112. Maybe it would work better since then, but I haven't tried.

sziszi, 2018-03-28:
Hi Laszlo

Maybe we can chat about this on a private channel to not pollute this
thread? My email address is palczaszilard@gmail.com. Please drop an email
if you are kind to help me with this.

Thank you!

sziszi, 2018-03-30:
I managed to bring to life the device! Thanks for your help Laszlo

llvasconcellos, 2019-01-20:
@sziszi did you get kernel 4.x to run on rk3066? HDMI is working?

lntrx, 2019-03-01:
Hey guys, is there any source to make it work on NAND-chip instead of SD-Card. I made it work on SD-Card, but I really need to get it work on NAND. I tried to port older version of rknand driver, but no luck

sziszi, 2019-03-01:
Yes i succeded with it. It was too long ago and i do not have the sources
anymore but i remember that after many weeks of trial and error i just
enabled it in the dts file and it worked. It worked with the default nand
driver from the kernel.

lntrx, 2019-03-03:
&nandc {
	u-boot,dm-spl;
	status = "okay";
	#address-cells = <1>;
	#size-cells = <0>;
};

Did you add something like this?

sziszi, 2019-03-03:
Yes a nandc section but without u-boot.dm-spl for sure

lntrx, 2019-03-07:
Do you remember if you add any driver for it. I couldn't make it work

sziszi, 2019-03-07:
I did not add any drivers as far as i remember. But i tried with many
kernel versions. It was a painful experience :) I also tried with
mainstream kernel. Some things worked with Rockchip kernel other things
with mainstream.Maybe i used mainstream in the latest working version. Also
beware of the GPIO pins. It is really hardware dependant. It is not enough
to enable a component you also need to power it. Without creating a proper
dts file for your specific device it is almost impossible. Creating a
proper dts without the schematics is like walking in the dark. I used the
ryeager dts already existing in the linux kernel. That was the most
featured one and i managed to bring to life some components. I was not
succesful with the GPIO of wireless. I enabled just the driver and i gave
the Vcc voltage directly from PCB Vin by soldering a wire. I managed to
bring up also the wireless that way.

lntrx, 2019-03-07:
I found somewhere this entry and tried didn't work for me 

	nandc: nandc@10500000 {
		compatible = "rockchip,rk-nandc";
		reg = <0x10500000 0x2000>;
		interrupts = <GIC_SPI 27 IRQ_TYPE_LEVEL_HIGH>;
		clock-names = "hclk";
		clocks = <&cru HCLK_NANDC0>;
		status = "disabled";
	};

sziszi, 2019-03-07:
Well with status="disabled" why you expect it to work? And even if you
change it to enabled? Why do you expect that all those settings will fit
your device? As i said before it is a little bit of gambling. What works
for a device will not work for other devices. You can also have a high
chance damaging your device while messing with those settings. Just a
warning.

lntrx, 2019-03-07:
Thanks for warning mate. This is in main .dts something like rk3xx.dts in marsboard.dts I make it status=okay. I am really not sure, I tried lots of things I am even not sure where I got this part. There's no official sources. If you remember or find your dts conf somehow even which kernel version you tried that would be really nice. Device seems okay for now. I am still able to boot

lntrx, 2019-03-07:
http://u-boot.10912.n7.nabble.com/PATCH-v4-00-19-add-support-for-rk3066-platform-td303420.html 

This is where I got the idea

sziszi, 2019-03-07:
You can forget it totally. I even contacted Pavel Jarosz a year ago about
it. He confirmed that u-boot is not working for RK3066. I cannot find my
dts anymore. I can help you with my latest kernel image if you want. But
tell me what kind of device you have exactly?

lntrx, 2019-03-07:
That would be really nice mate. I have Marsboard RK3066

http://www.marsboard.com/marsboard_rk3066_feature.html

sziszi, 2019-03-07:
Your email address please

lntrx, 2019-03-07:
demire801 [at] gmail

darkalexpp, 2020-10-04:
Hi, I am trying to get the 4.4 kernel to work on a rk3066 clone, some things are different but it is similar to the marshboard,
I'm trying to get nand to work, I don't know what the dts settings are for it.

lntrx, 2020-10-05:
https://github.com/lntrx/rk3066_kernel4-4 checky my repo. Back then I made
changes and create a script to completely build image

darkalexpp, 2020-10-16:
> Hello sziszi,
> rk3066 and rk3188 support came into rockchip-linux kernel via mainline/kernel.org kernel. There is a basic support for these old SOCs in mainline kernel, cpu, usb, sd-card works, hdmi, [...]

@Laszlo-Fiat hi, how could you get the hdmi working?
What configuration to use in the kernel to compile?
I've been trying but I only get a black screen

Yaksinikos, 2023-10-28:
> @Laszlo-Fiat hi, how could you get the hdmi working? What configuration to use in the kernel to compile? I've been trying but I only get a black screen

Does someone have an Idea of what kernel version supports RK3066 with HMDI, best possible adding support for MTK5931 & 6622 - would like to update my HTC-T011-V4 based SoC.