rockchip-linux/kernel #77: no camera clock with new RKISP1 driver

teseo-sw, 2018-03-28:
Hi,
I'm checking new rkisp1 driver with a ov13850 camera but something go wrong. I can't detect sensor cause clock is not enabled while I detect sensor using old cif_isp10.

I followed Rockchip-isp1 WIKI to configure my device tree. Those are its details:

```
#include "rk3288-rkisp1.dtsi"

&isp {
	status = "okay";

	port {
		#address-cells = <1>;
		#size-cells = <0>;

		isp_mipi_in: endpoint {
			reg = <0>;
			remote-endpoint = <&dphy_rx0_out>;
		};
	};
};

&isp_mmu {
	status = "okay";
};

&mipi_phy_rx0 {
	status = "okay";

	ports {
		#address-cells = <1>;
		#size-cells = <0>;

		port@0 {
			reg = <0>;
			#address-cells = <1>;
			#size-cells = <0>;

			mipi_rx0_in: endpoint {
				reg = <0>;
				remote-endpoint = <&camera0_out>;
				data-lanes = <1 2>;
			};
		};

		port@1 {
			reg = <1>;
			#address-cells = <1>;
			#size-cells = <0>;

			dphy_rx0_out: endpoint {
				reg = <0>;
				remote-endpoint = <&isp_mipi_in>;
			};
		};
	};
};
```

```
	ext_cam_clk: external-camera-clock {
		compatible = "fixed-clock";
		clock-frequency = <12000000>;
		clock-output-names = "CLK_CAMERA_12MHZ";
		#clock-cells = <0>;
	};

&i2c3 {
	camera0: ov13850@10 {
		compatible = "ovti,ov13850";
		reg = <0x10>;
		clocks = <&ext_cam_clk>;
		status = "okay";

		port {
			camera0_out: endpoint {
				remote-endpoint = <&mipi_rx0_in>;
				data-lanes = <1 2>;
			};
		};
	};
};
```

I add a /drivers/media/i2c/ov13850.c camera driver based on ov5647.c one. Clock is correctly read and, even if enabled on probe 

`ret = clk_prepare_enable(sensor->xclk);`

it does't seem to reach the camera and sensor isn't detected (pwd/rst gpio are ok). What could be the issue? thank you for the attention
best regards
Andrea

teseo-sw, 2018-04-03:
In detail my problem is that, during camera probe function, is not possible to detect CHIP_ID via i2c cause clock isn't enabled. Even if I add a clk_prepare_enable, clock start only after probe function ended. Has someone any suggestion?
regards

jsmrb88, 2018-11-12:
Hi @teseo-sw I am facing similar issue, I am working with custom board based on Phycore RK3288 SOM and trying to interface ov4689 MIPI CMOS module with MIPI DPHY RX0. But am not able to see sensor over I2C lines for address 0x36. Instead I see all address available on bus when I do i2cdetect -y -r 2 . Do you have any hint on this situation?

simoncheen, 2018-11-29:
I have the same problem these days, but resovled.
clocks = <&ext_cam_clk>;
change that as 
clocks = <&cru SCLK_VIP_OUT>;

it should works.

dmikushin, 2018-12-23:
> Hi @teseo-sw I am facing similar issue, I am working with custom board based on Phycore RK3288 SOM and trying to interface ov4689 MIPI CMOS module with MIPI DPHY RX0. But am not able to see sensor [...]

If there was an error during camera initialization (which could be seen in dmesg), then the device is unloaded and therefore the i2c lanes are free.