rockchip-linux/kernel #81: rockchip_vpu breaks the kernel build

bholland, 2018-04-01:
In file included from include/linux/kernel.h:13:0,
                 from include/linux/list.h:8,
                 from include/linux/kobject.h:20,
                 from include/linux/device.h:17,
                 from include/linux/platform_device.h:14,
                 from drivers/media/platform/rockchip-vpu/rockchip_vpu_common.h:27,
                 from drivers/media/platform/rockchip-vpu/rockchip_vpu_hw.c:17:
drivers/media/platform/rockchip-vpu/rockchip_vpu_hw.c: In function ‘rockchip_vpu_vp8e_assemble_bitstream’:
drivers/media/platform/rockchip-vpu/rockchip_vpu_common.h:474:13: warning: format ‘%d’ expects argument of type ‘int’, but argument 6 has type ‘size_t {aka long unsigned int}’ [-Wformat=]
error, forbidden warning: rockchip_vpu_common.h:474
scripts/Makefile.build:277: recipe for target 'drivers/media/platform/rockchip-vpu/rockchip_vpu_hw.o' failed
make[4]: *** [drivers/media/platform/rockchip-vpu/rockchip_vpu_hw.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/media/platform/rockchip-vpu' failed
make[3]: *** [drivers/media/platform/rockchip-vpu] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/media/platform' failed
make[2]: *** [drivers/media/platform] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/media' failed
make[1]: *** [drivers/media] Error 2

This is simple to repeat. Select all required options and make the rockchip_vpu module. It should fail with this warning.

Pillar1989, 2018-04-02:
@bholland  Just a warning

bholland, 2018-04-02:
By default, warnings break the kernel build with the rockchip build
scripts. The fix is fairly simple. Instead of having %d in your debug_vpu
function, use %zu for unsigned size_t. I probably can get a patch out for
it, if you would like.

Thanks,
~Ben

closed 2018-04-09 by hizukiayaka