rockchip-linux/kernel #83: Missing PHY driver for USB 3 in mainline

Xenopathic, 2018-04-03:
I'm not sure if this is the right place to report a potential issue with the upstreaming effort, but here goes.

The mainline kernel has support for USB 2, but is missing USB 3 since drivers/phy/rockchip/phy-rockchip-inno-usb3.c exists only here and not in mainline. Since the other PHY drivers exist in mainline, this seems like an inconsistency to me, perhaps something has been missed?

At the very least, the [status matrix](http://opensource.rock-chips.com/wiki_Status_Matrix) states that USB for the RK3328 is SUPPORT, while it should be INCOMPLETE due to this missing USB 3 support.

ShapeShifter499, 2018-07-15:
This issue is still a problem with kernel 4.18-rc4     It's missing in the official mainline kernel.  https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/phy/rockchip?h=v4.18-rc4

ayufan, 2018-07-18:
My mainline kernel does support USB3: https://github.com/ayufan-rock64/linux-mainline-kernel/ :)

ShapeShifter499, 2018-07-18:
@ayufan I'd need to convince Arch Linux ARM to make use of your kernel instead of mainline.  Not sure who I should contact about that.  Currently Arch Linux ARM claims support for Rock64 but doesn't mention anything on their wiki about USB 3.0 support.

ayufan, 2018-07-18:
It is about missing PHY nodes for Rock64: https://github.com/ayufan-rock64/linux-mainline-kernel/commit/84802c45a7cff42215869de40a2632db5f4a3be6 and something like this: https://github.com/ayufan-rock64/linux-mainline-kernel/commit/4109a410cdcfbd0c716cb765c582261f2197a73b.

Pretty fair you can add that on top of existing dtbs.

ShapeShifter499, 2018-07-18:
@ayufan so this file isn't required?    drivers/phy/rockchip/phy-rockchip-inno-usb3.c

ayufan, 2018-07-18:
No. Just dts change.

ShapeShifter499, 2018-07-21:
@ayufan Thanks, people are working now to get the required DTS for the Rock64 into mainline. https://archlinuxarm.org/forum/viewtopic.php?p=58866#p58866  

Other rockchip boards maybe still affected by the original issue above of drivers/phy/rockchip/phy-rockchip-inno-usb3.c missing in mainline.

vitalif, 2019-02-25:
@ayufan it seems USB3 does not work in your 4.20 mainline kernel

ayufan, 2019-02-25:
Thanks. I need help