rockchip-linux/kernel #95: pcie to sata board not work on firefly rk3399

idreamerhx, 2018-05-21:
Hi, 

I got a pcie (firefly m.2 ngff M-Key)  to sata board. Kernel source is from this repository and archlinux arm rootfs.

I config the dts to turn on pcie 3.3v power, and I tested ok. 

lspci shows:
00:00.0 PCI bridge: Fuzhou Rockchip Electronics Co., Ltd Device 0100
01:00.0 IDE interface: ASMedia Technology Inc. ASM1061 SATA IDE Controller (rev 01)

lspci -vv shows:

01:00.0 IDE interface: ASMedia Technology Inc. ASM1061 SATA IDE Controller (rev 01) (prog-if 85 [Master SecO PriO])
	Subsystem: ASMedia Technology Inc. Device 1060
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 227
	Region 0: I/O ports at <unassigned> [disabled]
	Region 1: I/O ports at <unassigned> [disabled]
	Region 2: I/O ports at <unassigned> [disabled]
	Region 3: I/O ports at <unassigned> [disabled]
	Region 4: I/O ports at <unassigned> [disabled]
	Region 5: Memory at fa010000 (32-bit, non-prefetchable) [disabled] [size=512]
	[virtual] Expansion ROM at fa000000 [disabled] [size=64K]

and parts of dmesg:

[    0.229086] of_get_named_gpiod_flags: parsed 'gpio' property of node '/vcc3v3-pcie-regulator[0]' - status (0)
[    0.229143] vcc3v3_pcie: 3300 mV 
[    0.229333] reg-fixed-voltage vcc3v3-pcie-regulator: vcc3v3_pcie supplying 3300000uV

[    0.335569] phy phy-pcie-phy.5: Looking up phy-supply from device tree
[    0.335580] phy phy-pcie-phy.5: Looking up phy-supply property in node /pcie-phy failed
[    0.337230] rockchip-pcie f8000000.pcie: GPIO lookup for consumer ep
[    0.337240] rockchip-pcie f8000000.pcie: using device tree for GPIO lookup
[    0.337269] of_get_named_gpiod_flags: parsed 'ep-gpios' property of node '/pcie@f8000000[0]' - status (0)
[    0.337507] rockchip-pcie f8000000.pcie: Looking up vpcie3v3-supply from device tree
[    0.337518] rockchip-pcie f8000000.pcie: Looking up vpcie3v3-supply property in node /pcie@f8000000 failed
[    0.337534] rockchip-pcie f8000000.pcie: no vpcie3v3 regulator found

[    0.428810] PCI host bridge /pcie@f8000000 ranges:
[    0.429266]   MEM 0xfa000000..0xfbdfffff -> 0xfa000000
[    0.429748]    IO 0xfbe00000..0xfbefffff -> 0xfbe00000
[    0.430605] rockchip-pcie f8000000.pcie: PCI host bridge to bus 0000:00
[    0.431194] pci_bus 0000:00: root bus resource [bus 00-1f]
[    0.431698] pci_bus 0000:00: root bus resource [mem 0xfa000000-0xfbdfffff]
[    0.432308] pci_bus 0000:00: root bus resource [io  0x0000-0xfffff] (bus address [0xfbe00000-0xfbefffff])
[    0.433193] pci 0000:00:00.0: [1d87:0100] type 01 class 0x060400
[    0.433308] pci 0000:00:00.0: supports D1
[    0.433320] pci 0000:00:00.0: PME# supported from D0 D1 D3hot

[    0.434549] pci_bus 0000:01: busn_res: can not insert [bus 01-ff] under [bus 00-1f] (conflicts with (null) [bus 00-1f])
[    0.434632] pci 0000:01:00.0: [1b21:0611] type 00 class 0x010185
[    0.434689] pci 0000:01:00.0: reg 0x10: initial BAR value 0x00000000 invalid
[    0.435318] pci 0000:01:00.0: reg 0x10: [io  size 0x0008]
[    0.435340] pci 0000:01:00.0: reg 0x14: initial BAR value 0x00000000 invalid
[    0.435992] pci 0000:01:00.0: reg 0x14: [io  size 0x0004]
[    0.436015] pci 0000:01:00.0: reg 0x18: initial BAR value 0x00000000 invalid
[    0.436677] pci 0000:01:00.0: reg 0x18: [io  size 0x0008]
[    0.436699] pci 0000:01:00.0: reg 0x1c: initial BAR value 0x00000000 invalid

I don't know if I need to modify kernel source to add some more space for more than one pcie?

Thanks all.

idreamerhx, 2018-05-21:
I tried this:
https://www.linuxquestions.org/questions/linux-kernel-70/kernel-fails-to-assign-memory-to-pcie-device-4175487043-print/

but not work. dmes gives the same message.

alexeijd, 2018-06-07:
apologies for the possibly dumb question, but what changes did you have to make to the dts to enable power on the pci-e?

idreamerhx, 2018-06-12:
@alexeijd  Thanks for your reply. I nearly forget this problem after some days of hard work but not solved this problem. Here is my dts.
`
diff --git a/arch/arm64/boot/dts/rockchip/rk3399-firefly-linux.dts b/arch/arm64/boot/dts/rockchip/rk3399-firefly-linux.dts
index beffd8dbb219..ff21202852b7 100644
--- a/arch/arm64/boot/dts/rockchip/rk3399-firefly-linux.dts
+++ b/arch/arm64/boot/dts/rockchip/rk3399-firefly-linux.dts
@@ -245,6 +245,8 @@
 		pinctrl-names = "default";
 		pinctrl-0 = <&pcie_drv>;
 		regulator-name = "vcc3v3_pcie";
+		regulator-min-microvolt = <3300000>;
+		regulator-max-microvolt = <3300000>;
 	};
 
 	vcc3v3_sys: vcc3v3-sys {
@@ -663,6 +665,40 @@
 		io-channels = <&saradc 4>;
 		hp-det-adc-value = <500>;
 	};
+
+	camera0: camera-module@00 {
+	};
 };
 
 &i2c3 {
@@ -741,6 +777,16 @@
 	gpio1830-supply = <&vcc_3v0>;		/* gpio1833_gpio4cd_ms */
 };
 
+&cif_isp0 {
+	rockchip,camera-modules-attached =
+		<&camera0>;
+	status = "okay";
+};
+
+&isp0_mmu {
+	status = "okay";
+};
+
 &pcie_phy {
 	status = "okay";
 };
@@ -750,6 +796,7 @@
 	num-lanes = <4>;
 	pinctrl-names = "default";
 	pinctrl-0 = <&pcie_clkreqn_cpm>;
+	phy-supply = <&vcc3v3_pcie>;
 	status = "okay";
 };
 
@@ -778,9 +825,9 @@
 	pcie {
 		pcie_drv: pcie-drv {
 			rockchip,pins =
-				<1 17 RK_FUNC_GPIO &pcfg_pull_none>;
-			};
-			pcie_3g_drv: pcie-3g-drv {
+				<1 17 RK_FUNC_GPIO &pcfg_pull_up>;
+		};
+		pcie_3g_drv: pcie-3g-drv {
 			rockchip,pins =
 				<0 2 RK_FUNC_GPIO &pcfg_pull_up>;
 		};

`

alexeijd, 2018-06-14:
Thanks for the pointers, In my case I'm getting a different output:
[    0.929421] rockchip-pcie: probe of f8000000.pcie failed with error -110
with an empty lspci output makes me think that my pcie is disabled or broken
my card is getting power from the bus (it lights an led on the card fine) and I know the card (dual intel gige m.2) is working fine, and should be ok (3.3v), I just can't see anything

wombat, 2019-01-08:
@alexeijd I have close to the similar problem, except my card is a Delock PCI Express Card > Mini PCIe with a Telit LM960 LTE module and the LED on my PCIe board is not lighting up.
How can I check if PCIe is enabled and if not change that?

My dmesg: https://pastebin.com/FHMGRgVG

radfish, 2020-09-03:
FWIW, had the same -110 error with M.2 SATA card (ASM106x chipset) on NanoPC-T4 (RK3399) with 4.4.x from FriendlyARM and from Rockchip, but problem went away on mainline 5.8.5. Details here: https://discuss.96boards.org/t/pcie-link-failure-with-pcie-device-card/9643/7?u=redfish

philippludwig, 2022-05-14:
@radfish How did you install 5.8.5 on your board?