rockchip-linux/kernel #98: Build failure after recent update to Mali Midgard Driver

Tonymac32, 2018-05-31:
https://github.com/rockchip-linux/kernel/commit/91842c9d8b4f21a431643d97269d142d9524cd9a#diff-60ad033154c5f79f2acf5a2d6576198e

Configured "Mali for Android", worked prior to update, now
build failure on a warning:

```
drivers/gpu/arm/midgard/ipa/mali_kbase_ipa.c:582:22: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
error, forbidden warning: mali_kbase_ipa.c:582
  CC      drivers/regulator/core.o
scripts/Makefile.build:277: recipe for target 'drivers/gpu/arm/midgard/ipa/mali_kbase_ipa.o' failed
make[4]: *** [drivers/gpu/arm/midgard/ipa/mali_kbase_ipa.o] Error 1
make[4]: *** Waiting for unfinished jobs....
```

Tonymac32, 2018-05-31:
Also, if using "Midgard for Linux" we get a version mismatch of kernel 10.4, user 10.6 with the rockchip user driver.

Tonymac32, 2018-06-01:
OK, it was determined selecting CONFIG_MALI_PWRSOFT_765=y makes it compile, but there were still some other issues.  Is there a config recommendation as that has obviously been affected?

Tonymac32, 2018-06-07:
OK:  Midgard for Android
Config mali as module or else everything explodes
CONFIG_MALI_PWRSOFT_765=y 

boots.  But, no serial console on HDMI, no HDMI output at all until X starts.  

also:
```
[    4.453675] mali ffa30000.gpu: Failed to get leakage
[    4.453722] mali ffa30000.gpu: Failed to get pvtm
[    4.454086] W : [File] : drivers/gpu/arm/midgard/platform/rk/mali_kbase_config_rk.c; [Line] : 113; [Func] : kbase_platform_rk_init(); power-off-delay-ms not available.
[    4.454958] mali ffa30000.gpu: GPU identified as 0x0750 r0p0 status 1
[    4.455114] mali ffa30000.gpu: Protected mode not available
[    4.455386] ERROR: Bad of_node_put() on /gpu@ffa30000
[    4.455395] CPU: 3 PID: 159 Comm: systemd-udevd Not tainted 4.4.135-rockchip #46
[    4.455399] Hardware name: Rockchip (Device Tree)
[    4.455423] [<c0110104>] (unwind_backtrace) from [<c010bde0>] (show_stack+0x20/0x24)
[    4.455435] [<c010bde0>] (show_stack) from [<c05ab490>] (dump_stack+0x84/0xa4)
[    4.455447] [<c05ab490>] (dump_stack) from [<c0930658>] (of_node_release+0x34/0x9c)
[    4.455458] [<c0930658>] (of_node_release) from [<c05ad514>] (kobject_put+0xbc/0xdc)
[    4.455466] [<c05ad514>] (kobject_put) from [<c092fd1c>] (of_node_put+0x24/0x28)
[    4.455475] [<c092fd1c>] (of_node_put) from [<c092c718>] (of_find_compatible_node+0x48/0x98)
[    4.455553] [<c092c718>] (of_find_compatible_node) from [<bf0209e4>] (get_model_dt_node+0x4c/0x90 [midgard_kbase])
[    4.455689] [<bf0209e4>] (get_model_dt_node [midgard_kbase]) from [<bf020d5c>] (kbase_ipa_model_add_param_string+0x34/0xdc [midgard_kbase])
[    4.455822] [<bf020d5c>] (kbase_ipa_model_add_param_string [midgard_kbase]) from [<bf0208e8>] (kbase_simple_power_model_init+0xbc/0xdc [midgard_kbase])
[    4.455955] [<bf0208e8>] (kbase_simple_power_model_init [midgard_kbase]) from [<bf020eb4>] (kbase_ipa_init_model+0x6c/0xb4 [midgard_kbase])
[    4.456087] [<bf020eb4>] (kbase_ipa_init_model [midgard_kbase]) from [<bf020fb4>] (kbase_ipa_init+0xb8/0x17c [midgard_kbase])
[    4.456218] [<bf020fb4>] (kbase_ipa_init [midgard_kbase]) from [<bf02bf8c>] (kbase_devfreq_init+0x3fc/0x4e0 [midgard_kbase])
[    4.456348] [<bf02bf8c>] (kbase_devfreq_init [midgard_kbase]) from [<bf0189d8>] (kbase_platform_device_probe+0x5e0/0xae0 [midgard_kbase])
[    4.456422] [<bf0189d8>] (kbase_platform_device_probe [midgard_kbase]) from [<c06c335c>] (platform_drv_probe+0x60/0xac)
[    4.456433] [<c06c335c>] (platform_drv_probe) from [<c06c16a0>] (driver_probe_device+0x1a8/0x298)
[    4.456442] [<c06c16a0>] (driver_probe_device) from [<c06c17ec>] (__driver_attach+0x5c/0x9c)
[    4.456451] [<c06c17ec>] (__driver_attach) from [<c06bfb84>] (bus_for_each_dev+0x90/0xa0)
[    4.456459] [<c06bfb84>] (bus_for_each_dev) from [<c06c1128>] (driver_attach+0x28/0x30)
[    4.456466] [<c06c1128>] (driver_attach) from [<c06c0d64>] (bus_add_driver+0x184/0x1ec)
[    4.456475] [<c06c0d64>] (bus_add_driver) from [<c06c2354>] (driver_register+0xb0/0xf0)
[    4.456482] [<c06c2354>] (driver_register) from [<c06c329c>] (__platform_driver_register+0x40/0x54)
[    4.456549] [<c06c329c>] (__platform_driver_register) from [<bf054018>] (kbase_platform_driver_init+0x18/0x1000 [midgard_kbase])
[    4.456617] [<bf054018>] (kbase_platform_driver_init [midgard_kbase]) from [<c0101ba8>] (do_one_initcall+0x1a0/0x1d4)
[    4.456629] [<c0101ba8>] (do_one_initcall) from [<c01ecabc>] (do_init_module+0x6c/0x3a8)
[    4.456641] [<c01ecabc>] (do_init_module) from [<c01a5668>] (load_module+0x1d48/0x1d94)
[    4.456651] [<c01a5668>] (load_module) from [<c01a5880>] (SyS_finit_module+0x70/0x78)
[    4.456660] [<c01a5880>] (SyS_finit_module) from [<c0107600>] (ret_fast_syscall+0x0/0x48)
[    4.456691] mali ffa30000.gpu: Using configured power model mali-simple-power-model, and fallback mali-simple-power-model
[    4.456812] I : [File] : drivers/gpu/arm/midgard/backend/gpu/mali_kbase_devfreq.c; [Line] : 399; [Func] : kbase_devfreq_init(); success initing power_model_simple.
[    4.457789] mali ffa30000.gpu: Probed as mali0
[    4.470893] devfreq ffa30000.gpu: Couldn't update frequency transition information.
```

I also get a heap of 
```

[   85.943087] [drm:dw_hdmi_rockchip_set_property] *ERROR* failed to set rockchip hdmi connector property
[   85.943112] [drm:dw_hdmi_rockchip_set_property] *ERROR* failed to set rockchip hdmi connector property
[   85.943124] [drm:dw_hdmi_rockchip_set_property] *ERROR* failed to set rockchip hdmi connector property
[   85.943145] [drm:dw_hdmi_rockchip_set_property] *ERROR* failed to set rockchip hdmi connector property
[   85.943156] [drm:dw_hdmi_rockchip_set_property] *ERROR* failed to set rockchip hdmi connector property
[   85.943168] [drm:dw_hdmi_rockchip_set_property] *ERROR* failed to set rockchip hdmi connector property
[   85.943574] rockchip-vop ff930000.vop: [drm:vop_crtc_enable] Update mode to 1920x1080p0, type: 11

```

I'm reading through commits to try to get a feel of what's going on.

Tonymac32, 2018-06-08:
I will be updating as I resolve/uncover more issues or until intervention.

Ntemis, 2018-06-08:
Will bring out the big guns

Tonymac32, 2018-06-08:
@Ntemis your config updates brought back the framebuffer console, cheers!

Ntemis, 2018-06-08:
rk3288 do not support hdr metadata so your dmesg log about DW HDMI error messages is normal

Ntemis, 2018-06-09:
btw on armbian rockchip-default http://dpaste.com/37Y884R

Ntemis, 2018-06-09:
somehow they were locked to root

Tonymac32, 2018-06-09:
interesting.  I'll clone it to another machine and see what I get.

chwe17, 2018-06-11:
CONFIG_OF_OVERLAY=y leeds to a conflict with your ISP1 driver, see dmesg:
```
root@tinkerboard:~# dmesg | grep isp
[    1.877338] iommu: Adding device ff910000.isp to group 0
[    2.387517] rockchip-drm display-subsystem: devfreq is not set
[    2.389212] rockchip-drm display-subsystem: bound ff940000.vop (ops 0xc0c761d4)
[    2.390343] rockchip-drm display-subsystem: bound ff930000.vop (ops 0xc0c761d4)
[    2.392101] rockchip-drm display-subsystem: bound ff980000.hdmi (ops 0xc0c6ed78)
[    2.392331] rockchip-drm display-subsystem: failed to parse display resources
[    2.392389] rockchip-drm display-subsystem: No connectors reported connected with modes
[    2.434774] rockchip-drm display-subsystem: fb0:  frame buffer device
[    2.788622] ERROR: Bad of_node_put() on /isp@ff910000/port
[    2.788955] [<c0818fdc>] (v4l2_async_notifier_parse_fwnode_endpoints) from [<c08507f4>] (rkisp1_plat_probe+0x2bc/0x440)
[    2.788977] [<c08507f4>] (rkisp1_plat_probe) from [<c06a5fbc>] (platform_drv_probe+0x60/0xac)
[    2.789152] [<c06a5efc>] (__platform_driver_register) from [<c103bf1c>] (rkisp1_plat_drv_init+0x18/0x20)
[    2.789174] [<c103bf1c>] (rkisp1_plat_drv_init) from [<c0101ba8>] (do_one_initcall+0x1a0/0x1d4)
[    2.789678] [<c0818fdc>] (v4l2_async_notifier_parse_fwnode_endpoints) from [<c08507f4>] (rkisp1_plat_probe+0x2bc/0x440)
[    2.789698] [<c08507f4>] (rkisp1_plat_probe) from [<c06a5fbc>] (platform_drv_probe+0x60/0xac)
[    2.789867] [<c06a5efc>] (__platform_driver_register) from [<c103bf1c>] (rkisp1_plat_drv_init+0x18/0x20)
[    2.789888] [<c103bf1c>] (rkisp1_plat_drv_init) from [<c0101ba8>] (do_one_initcall+0x1a0/0x1d4)
[    2.789993] ERROR: Bad of_node_put() on /isp@ff910000/port
[    2.790304] [<c0818fdc>] (v4l2_async_notifier_parse_fwnode_endpoints) from [<c08507f4>] (rkisp1_plat_probe+0x2bc/0x440)
[    2.790325] [<c08507f4>] (rkisp1_plat_probe) from [<c06a5fbc>] (platform_drv_probe+0x60/0xac)
[    2.790493] [<c06a5efc>] (__platform_driver_register) from [<c103bf1c>] (rkisp1_plat_drv_init+0x18/0x20)
[    2.790514] [<c103bf1c>] (rkisp1_plat_drv_init) from [<c0101ba8>] (do_one_initcall+0x1a0/0x1d4)
[    2.790609] ERROR: Bad of_node_put() on /isp@ff910000/port
[    2.790940] [<c0818fdc>] (v4l2_async_notifier_parse_fwnode_endpoints) from [<c08507f4>] (rkisp1_plat_probe+0x2bc/0x440)
[    2.790960] [<c08507f4>] (rkisp1_plat_probe) from [<c06a5fbc>] (platform_drv_probe+0x60/0xac)
[    2.791130] [<c06a5efc>] (__platform_driver_register) from [<c103bf1c>] (rkisp1_plat_drv_init+0x18/0x20)
[    2.791151] [<c103bf1c>] (rkisp1_plat_drv_init) from [<c0101ba8>] (do_one_initcall+0x1a0/0x1d4)
[    2.792857] [<c06a5efc>] (__platform_driver_register) from [<c103bf3c>] (rockchip_isp_mipidphy_driver_init+0x18/0x20)
[    2.792880] [<c103bf3c>] (rockchip_isp_mipidphy_driver_init) from [<c0101ba8>] (do_one_initcall+0x1a0/0x1d4)
[    2.793427] [<c06a5efc>] (__platform_driver_register) from [<c103bf3c>] (rockchip_isp_mipidphy_driver_init+0x18/0x20)
[    2.793450] [<c103bf3c>] (rockchip_isp_mipidphy_driver_init) from [<c0101ba8>] (do_one_initcall+0x1a0/0x1d4)
[    2.794055] [<c06a5efc>] (__platform_driver_register) from [<c103bf3c>] (rockchip_isp_mipidphy_driver_init+0x18/0x20)
[    2.794078] [<c103bf3c>] (rockchip_isp_mipidphy_driver_init) from [<c0101ba8>] (do_one_initcall+0x1a0/0x1d4)
[    2.794668] [<c06a5efc>] (__platform_driver_register) from [<c103bf3c>] (rockchip_isp_mipidphy_driver_init+0x18/0x20)
[    2.794691] [<c103bf3c>] (rockchip_isp_mipidphy_driver_init) from [<c0101ba8>] (do_one_initcall+0x1a0/0x1d4)
[    2.795235] [<c06a5efc>] (__platform_driver_register) from [<c103bf3c>] (rockchip_isp_mipidphy_driver_init+0x18/0x20)
[    2.795258] [<c103bf3c>] (rockchip_isp_mipidphy_driver_init) from [<c0101ba8>] (do_one_initcall+0x1a0/0x1d4)
[    2.795867] [<c06a5efc>] (__platform_driver_register) from [<c103bf3c>] (rockchip_isp_mipidphy_driver_init+0x18/0x20)
[    2.795889] [<c103bf3c>] (rockchip_isp_mipidphy_driver_init) from [<c0101ba8>] (do_one_initcall+0x1a0/0x1d4)
[    2.796499] [<c06a5efc>] (__platform_driver_register) from [<c103bf3c>] (rockchip_isp_mipidphy_driver_init+0x18/0x20)
[    2.796522] [<c103bf3c>] (rockchip_isp_mipidphy_driver_init) from [<c0101ba8>] (do_one_initcall+0x1a0/0x1d4)
[    2.797078] [<c06a5efc>] (__platform_driver_register) from [<c103bf3c>] (rockchip_isp_mipidphy_driver_init+0x18/0x20)
[    2.797100] [<c103bf3c>] (rockchip_isp_mipidphy_driver_init) from [<c0101ba8>] (do_one_initcall+0x1a0/0x1d4)
[    2.797657] [<c06a5efc>] (__platform_driver_register) from [<c103bf3c>] (rockchip_isp_mipidphy_driver_init+0x18/0x20)
[    2.797680] [<c103bf3c>] (rockchip_isp_mipidphy_driver_init) from [<c0101ba8>] (do_one_initcall+0x1a0/0x1d4)
```
and:
```
root@tinkerboard:~# dmesg | grep ff910000
[    1.877338] iommu: Adding device ff910000.isp to group 0
[    2.788622] ERROR: Bad of_node_put() on /isp@ff910000/port
[    2.789993] ERROR: Bad of_node_put() on /isp@ff910000/port
[    2.790609] ERROR: Bad of_node_put() on /isp@ff910000/port
```

Tonymac32, 2018-06-28:
27 days open, I had to take a break due to some other issues outside the software world.  I noticed @Ntemis my framebuffer console quit working again, so that will need looked at again. (it's been a couple weeks since I built a kernel)

Tonymac32, 2018-06-28:
Ah, I forgot the dtsi with the ugly "chosen" stuff was changed, I'll have to retool my patchset and try again.

skyway22, 2019-06-16:
So, is there a solution?

Tonymac32, 2019-06-18:
A stable point is used in Rockchip 4.4 kernel, it builds fine now.