rockchip-linux/kernel #112: driver of ov13850 for ubuntu (firefly rk3399)

yolo19048, 2018-07-31:
hi everybody 
i'm looking for the driver of the ov13850 for ubuntu in firefly rk3399
i'm totaly new in github :p

i found this topic : https://github.com/rockchip-linux/kernel/issues/33 
but he closed a long time ago and i don't understand if they succeed to finish this driver 

i see that they put patch in this link :  https://patchwork.linuxtv.org/project/linux-media/list/?submitter=7231
but i don't know if i can use and how to use then. 

i hope somebody can help,
thank you in advance

jkiggins-cl, 2018-08-02:
@yolo19048 I'm fairly new to the rock-chip platform (my company is looking at using the firefly-rk3399 for vision applications) but I'm having issues with the ov13850 too, so I'll share my understanding so far.

The [isp1 driver](https://github.com/rockchip-linux/kernel/tree/release-4.4/drivers/media/platform/rockchip/isp1) has been integrated into the 4.4-stable branch as mentioned in #33. This driver can be enabled by setting VIDEO_ROCKCHIP_ISP1 from menuconfig (though it is enabled in rockchip_linux_defconfig).

If you look at the [wiki page](http://opensource.rock-chips.com/wiki_Rockchip-isp1) (written by @wzyy2 I believe) under the device tree bindings section there needs to be isp0 and mipi_dphy_rx0 entries in the .dtsi file and corresponding entries in the .dts file. For the firefly, entries exist in rk3399.dtsi but not rk3399-firefly-linux.dts. I would think the device tree needs modification for the camera to work.

The ov13850 specific stuff, which really just sets up the sensor and any sub-devices over i2c-1: [ov13850_v4l2-i2c-subdev.c](https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/media/i2c/soc_camera/rockchip/ov13850_v4l2-i2c-subdev.c), [ov13850.c](https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/media/i2c/ov13850.c)

Using the 4.4-stable kernel with a default configuration (ie. ``make rockchip_linux_defconfig && make``) without touching any device tree files, I get the following error messages during boot.

```
$ dmesg | grep cam
[    1.359132] ov13850.ov_camera_module_write_config(182) ERR: no active sensor configuration
[    1.359858] ov13850.ov_camera_module_write_config(233) ERR: failed with error -14
[    1.361451] ov13850.pltfrm_camera_module_read_reg(996) ERR: i2c read from offset 0x0000300a failed with error -6
[    1.362526] ov13850.pltfrm_camera_module_read_reg(996) ERR: i2c read from offset 0x0000300b failed with error -6
[    1.363434] ov13850.ov13850_check_camera_id(1820) ERR: register read failed, camera module powered off?
[    1.364265] ov13850.ov13850_check_camera_id(1838) ERR: failed with error (-6)        
[    1.364933] ov13850.ov_camera_module_attach(256) ERR: failed with error -6
```

If you also get these at least you'll know the driver is being loaded. I'll post updates here If I make any useful progress.

Caesar-github, 2018-08-06:
I think the ov13850 can work with release4.4 kernel on rk3399 firefly board.

e.g:  https://github.com/rockchip-linux/kernel
commit 8c7bd35 ARM64: dts: rockchip: add firefly ov13850 camera

jkiggins-cl, 2018-08-07:
@Caesar-github, unfortunately this isn't true in my case. For me the camera works in android 7.1 but not Debian. This is true regardless of which MIPI CSI port is used, and I've tested with two camera modules.

```
$ git pull
Already up-to-date.
$ git status
On branch release-4.4
Your branch is up-to-date with 'origin/release-4.4'.
nothing to commit, working directory clean
$ make rockchip_linux_defconfig
$ cd ../..

# Custom scripts to update the kernel on sd card using mk-image and flash-tool 
$ ./update-kernel.sh /dev/sdh
$ ./update-modules.sh /dev/sdh5
```
mk-image and flash-tool came from http://rockchip.wikidot.com/linux-user-guide
[update-kernel.sh](https://github.com/rockchip-linux/kernel/files/2266763/update-kernel.txt)
[update-modules.sh](https://github.com/rockchip-linux/kernel/files/2266764/update-modules.txt)

After booting to the updated SD card I'm still getting the same errors as in my previous comment. Have you been able to interface with the ov13850 on the firefly board running linux?

yolo19048, 2018-08-25:
@Caesar-github  i'm under the kernel 4.4.77 and nothing work... like @jkiggins-cl i tried the 2 different port...

jkiggins-cl, 2018-08-25:
I've got the camera working on the old ISP10 driver with some device tree modifications. See this branch https://github.com/jkiggins-cl/kernel/tree/release-4.4-ov13850

You might need to enable some stuff in menuconfig. I can post my .config file if you have issues.

jkiggins, 2018-09-20:
[kernel-config.txt](https://github.com/rockchip-linux/kernel/files/2402511/kernel-config.txt)

jkiggins-cl, 2018-09-20:
[kernel-config.txt](https://github.com/rockchip-linux/kernel/files/2402515/kernel-config.txt)

braincore, 2018-10-17:
Has anyone been successful in the last month? Are there any rk3399 SBCs that have this figured out?

arvcode, 2018-10-26:
Hi,
The ov13850 camera is working with resolution settings 2112x1568 using the old ISP driver in media/platform/rk-isp10/ in Rk3399 firefly board.
But there are 2 issues.

1. We cannot configure ov13850 to choose 4224x3136 settings from ov13850_v4l2-i2c-subdev.c.
If we force it to 4224*3136 then ISP driver returns CIF_MIPI_SYNC_FIFO_OVFLW error.

2. If we use platform/rockchip/isp1/ ISP driver and ov13850.c then there is no camera detected as I2C communication fails.

Can you update the complete code for ov13850 that supports the full resolution of 13MP[4224*3136]?
Also can you push the ov13850 code that supports ISP1 driver ?

hosaka, 2018-11-19:
Also interested in this. I'm using a NanoPC T4 which has both CSI1 and CSI2 headers and would like to use dual camera. CSI1 is detected no problem, another has the same dmesg errors as @jkiggins-cl posted earlier.

braincore, 2018-11-19:
@hosaka: Which distribution are you using? FriendlyCore with the 4.4 kernel or armbian?

hosaka, 2018-11-19:
@braincore using friendlycore 4.4.143, but have no issues switching if necessary. Have you had any luck getting it to work? I'm considering working with their fork here https://github.com/friendlyarm/kernel-rockchip

braincore, 2018-11-19:
@hosaka: No, I've been avoiding trying so far :) I'm watching from afar how many issues the rk3399 SBCs have (camera, big-little architecture, heat dissipation) before investing further.

hosaka, 2018-11-19:
@braincore I've had it for a few weeks now, and to be honest, this was my oversight, I just assumed that both cameras will work in linux. Other than that, it's been okay so far for my application.

Caesar-github, 2018-11-19:
Hmmm.... both cameras had work in linux. (At least the buildroot....)
We had used the new ISP framework (isp11) and enabled the 3A function.

@braincore  Can you point out the issue as the above said "the big-little architecture heat dissipation.." ?

-Caesar

hosaka, 2018-11-19:
@Caesar-github Looking at https://github.com/friendlyarm/kernel-rockchip/commit/8c7bd35db47726af4be5d2e5364aeace053ab1f0 the same commit you pointed out earlier is present. Yet the dmesg errors suggest that loading has failed. CSI1 ports work absolutely fine however. Do you have any ideas about what can be causing this?

Thank you

braincore, 2018-11-19:
@Caesar-github: Sorry if I phrased it incorrectly. I'm looking for reviews that cover those topics, not so much that there are alarming issues. From my notes, here are a couple resources.

Big-little: I'm interested in performance for heterogeneous workloads. I'm curious because of the ideas presented [here](https://elinux.org/images/f/f0/Doing_big.LITTLE_right-_little_and_Big_Obstacles.pdf), which is relevant to all big-little archs. I don't know what the current state is in 2018.

Heat dissipation: While it looks like the ODROID-N1 was never released, they have some cooling results [here](https://com.odroid.com/sigong/blog/blog_list.php?tag=ODROID-N1). It appears that passive cooling is insufficient for sustained workloads, which isn't unexpected.

hosaka, 2018-11-19:
@braincore You might want to take a lot at the next odroid h2 [here](https://forum.odroid.com/viewtopic.php?f=29&t=32536) in case you missed it! They also provide some temperature stress tests. The heatsink there is rather large though.

yolo19048, 2019-01-24:
apparently there  a solution here  http://bbs.t-firefly.com/forum.php?mod=viewthread&tid=2490&extra=page%3D1 but i didn't manage to flash the good image :/ and i didn't try another image

gounthar, 2019-05-03:
Has anyone managed to get a CSI camera working with the OrangePi RK3399? 
I bought this [camera](https://www.friendlyarm.com/index.php?route=product/product&path=78&product_id=228).