rockchip-linux/kernel #114: Probably wrong conditional statements in halbtc8822b2ant.c

markh-de, 2018-08-08:
Branch: `release-4.4`.

Some statements in `halbtc8822b2ant.c` (`rockchip_wlan` subtree) are strangely indented, suggesting a possible **programming mistake**.

---

In the following files I think **the `if` statement should be commented out** along with the already commented-out assignment. The empty line following the commented-out statement suggests that the **following three statements are supposed to be outdented** by 1 level (i.e. unrelated to the `if`).

https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/net/wireless/rockchip_wlan/rtl8188eu/hal/btc/halbtc8822b2ant.c#L1730

https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/net/wireless/rockchip_wlan/rtl8723cs/hal/btc/halbtc8822b2ant.c#L1730

---

In the following file, @rkhuangtao already tried to fix the compile warning, but to me it seems the "fix" only silences the warning, but still does the wrong thing. Admitted, I did not have time to dive into the code, but I do not think that only the first line (in a block of three) should be executed conditionally. Like above, I suggest to **comment out the `if` as well and outdent the first statement** within the `/* Ext switch buffer mux */`-titled block.

https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/net/wireless/rockchip_wlan/rtl8822be/hal/btc/halbtc8822b2ant.c#L1733

---

I stumbled upon these issues when I got **compile errors with GCC 7** (along with `-Wmisleading-indentation`).

Please comment and/or fix, or tell me to prepare a PR.

Thank you!

rkhuangtao, 2018-08-09:
These drivers are came from WiFi vendor. We don't know whether they do this intentionally or just typo. So it is
```
	if (board_info->ant_div_cfg)
		btcoexist->btc_write_1byte(btcoexist, 0x974, 0xff);
	btcoexist->btc_write_1byte_bitmask(btcoexist, 0x1991, 0x3, 0x0);
	btcoexist->btc_write_1byte_bitmask(btcoexist, 0xcbe, 0x8, 0x0);
```
or
```
	if (board_info->ant_div_cfg) {
		btcoexist->btc_write_1byte(btcoexist, 0x974, 0xff);
		btcoexist->btc_write_1byte_bitmask(btcoexist, 0x1991, 0x3, 0x0);
		btcoexist->btc_write_1byte_bitmask(btcoexist, 0xcbe, 0x8, 0x0);
	}
```
I can't tell you right now. I will let our engineer to check with vendor.
But I think the fist one just work because it pass the tests.

BTW, we don't merge patches on GitHub, this just release branch. But we can merge your patch on our internal repository, then push to GitHub.

markh-de, 2018-08-09:
Thanks for looking into this.

From a functional standpoint, it doesn't seem to make much sense to only do the first of three write commands conditionally.

**Hint:** Just grep for `/* Ext switch buffer mux */` and you'll see some more examples where they execute all three or more commands as a group.

It seems to me that **only** the commented-out statement was supposed to be guarded by the `if` (that's what you'd probably see in the file history). That is, the `if` should simply be commented out, too, and the three write commands should be executed unconditionally.

Also, this kind of blocks should generally be properly outdented. Looks like copy/paste.

Did you check for passing tests if you implement it the way I suggested? For me, the original source does not even pass a compile run (GCC 7 with -Wmisleading-indentation).

Please keep us informed about Realtek's response.

Thank you.

rkhuangtao, 2018-08-13:
Vendor will release new driver to fix this problem. And you are right:

> It seems to me that only the commented-out statement was supposed to be guarded by the if (that's what you'd probably see in the file history). That is, the if should simply be commented out, too, and the three write commands should be executed unconditionally.

Like this:
```
	// if (board_info->ant_div_cfg)
		/*ctrl_type = BT_8822B_2ANT_EXT_ANT_SWITCH_CTRL_BY_ANTDIV;*/
	/* Ext switch buffer mux */
		btcoexist->btc_write_1byte(btcoexist, 0x974, 0xff);
		btcoexist->btc_write_1byte_bitmask(btcoexist, 0x1991, 0x3, 0x0);
		btcoexist->btc_write_1byte_bitmask(btcoexist, 0xcbe, 0x8, 0x0);
```