rockchip-linux/kernel #115: RK3328, USB 3.0 hangs when accessing two HDDs simultaneously

h31, 2018-08-10:
Hello. I own a Rock64 board, running Linux kernel 4.4.132-1075-rockchip-ayufan-ga83beded8524 and Ubuntu Bionic [0.7.9](https://github.com/ayufan-rock64/linux-build/releases/tag/0.7.9).

I'm trying to connect several 2.5" HDDs to the Rock64 board. To do this, I use an externally-powered USB 3.0 hub. The problem is that I'm receiving errors when trying to access two drives simultaneously.

I've used fio and dd utilities to access hard drives. The example command is "dd if=/dev/sda of=/dev/null bs=128k". When I'm reading from one drives, everything works well. When I'm reading from two drives in parallel, dd and fio utilities hang and ROCK64 disconnects hard drives.

Dmesg log can be found [here](https://forum.pine64.org/showthread.php?tid=5557).

Even when I'm keeping only 2 hard drives connected, the error still occurs. But everything works well when I'm plugging an USB hub cable to a USB 2.0 port of the ROCK64. Also, everything works perfectly when I'm connecting the USB hub to my Intel laptop (using the same set of cables and a PSU). The Rock64 computer is powered by an official 3A PSU. The USB hub is powered by 2A PSU. Tried several PSUs - no difference.

There are several other similar reports (for example, ayufan-rock64/linux-build#112 and [forum.armbian.com](https://forum.armbian.com/topic/7740-rock64-usb-30-host-controller-crashes-when-accessing-2-disks-simultaneously/)).

h31, 2018-08-10:
The forum post I've linked was written in January, but the problem still exists. At least the last case (where `xHCI host not responding to stop endpoint command` is printed) still occurs.

markh-de, 2018-08-10:
Please make very sure that you this is not a power issue. I myself are running a 4x 3TB WD Red (3.5in) NAS on a ROCK64 (latest Armbian build), and I had several power issues during bring-up. Now they are all gone, but it took me a while to accept that it's really a 5V **drive** supply issue.

Yes, your USB 3.0 hub is powered, but USB-to-SATA adapters often draw a **lot** of current on the 5V line. I too thought it should be sufficient to provide the 12V for the spindle motors externally to the drives and let the 4x ORICO (JMS578 inside) devices be powered from the USB hub (which is powered via dedicated 5V externally). But these USB-to-SATA adapters also need to power the 5V logic inside the drives, and in your case even the spindle motors (2.5in drives). Four of them was simply too much.

The effect was the same as yours: As soon as I accessed more than one or two drives, the other drives clicked and got a USB reset, simply because the 5V supply line (from the USB hub to the adapters and to the drives) broke down and the logic of the drives could not be powered properly.

My solution was simple: I used SATA extension cables and only connected the SATA **data** lines to the drives, while the SATA power connectors are provided by my 5V and 12V system power supplies. Since this change everything works well.

Got a RAID5 array with effectively 396 MB/s throughput. Not bad at all.

Good luck!

kskewes, 2019-01-06:
There is definitely something broken with USB3.0 on RK3288 (at least Rock64 implementation).
USB2.0 is mostly very stable, whether externally power 2.5" SSD's or not.
I've given up on USB3.0 even though it is much faster when it works and I'm only using 1 drive.
I have 6x V2 Rock64's and 3x USB->SATA so can run any debugging commands that would help.