rockchip-linux/kernel #116: PCIe-port not working on RK3399

k-a-z-u, 2018-08-10:
We have a RockPro64 board here and tried to get the pcie port working.
Depending on the card that was inserted, the port was either simply disabled, or the kernel panicked during boot.

Dmesg when no card is in the slot (4.4.138-1094):
- Also with this card: Intel Ethernet LAN Card D33025
- Also with this card: Logilink Dual LAN-Card with two Realtek chips on it [http://www.logilink.com/media/datasheets/PC0075.pdf](Datasheet)
```
[    3.361232] rockchip-pcie f8000000.pcie: GPIO lookup for consumer ep
[    3.361239] rockchip-pcie f8000000.pcie: using device tree for GPIO lookup
[    3.361260] of_get_named_gpiod_flags: parsed 'ep-gpios' property of node '/pcie@f8000000[0]' - status (0)
[    3.361450] rockchip-pcie f8000000.pcie: Looking up vpcie3v3-supply from device tree
[    3.361533] rockchip-pcie f8000000.pcie: Looking up vpcie1v8-supply from device tree
[    3.361539] rockchip-pcie f8000000.pcie: Looking up vpcie1v8-supply property in node /pcie@f8000000 failed
[    3.361556] rockchip-pcie f8000000.pcie: no vpcie1v8 regulator found
[    3.366975] rockchip-pcie f8000000.pcie: Looking up vpcie0v9-supply from device tree
[    3.366992] rockchip-pcie f8000000.pcie: Looking up vpcie0v9-supply property in node /pcie@f8000000 failed
[    3.367016] rockchip-pcie f8000000.pcie: no vpcie0v9 regulator found
[    3.371361] rockchip-pcie f8000000.pcie: invalid power supply
[    3.620975] EXT4-fs (mmcblk0p7): mounted filesystem with writeback data mode. Opts: (null)
[    3.875231] rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout!
[    3.890425] rockchip-pcie: probe of f8000000.pcie failed with error -110
```

With cards like Dell PowerEdge Perc 5i SAS RAID Controller, the kernel seldomly boots, ignoring the card and giving the dmesg-output from above, but most of the times, the kernel crashes with a couple of different [stack-traces](https://imgur.com/a/6zd65tz).

This is a stack-trace with this card using a mainline kernel. In contrast to this kernel, mainline continues booting, thus we were able to copy it out:
```
[    7.044117] Hardware name: Pine64 RockPro64 (DT)
[    7.044656] pstate: 60000085 (nZCv daIf -PAN -UAO)
[    7.045223] pc : rockchip_pcie_rd_conf+0x18c/0x1f8 [pcie_rockchip_host]
[    7.045989] lr : rockchip_pcie_rd_conf+0x17c/0x1f8 [pcie_rockchip_host]
[    7.046748] sp : ffff00000e6f3730
[    7.047137] x29: ffff00000e6f3730 x28: 0000000000000001 
[    7.047754] x27: 0000000000000000 x26: 0000000000000000 
[    7.048372] x25: 0000000000000000 x24: ffff00000e6f3854 
[    7.048990] x23: ffff8000f1557800 x22: ffff00000e6f37b4 
[    7.049607] x21: ffff8000f0f5c398 x20: 0000000000000004 
[    7.050225] x19: ffff000010100000 x18: ffffffffffffffff 
[    7.050842] x17: 0000000000000000 x16: 0000000000000000 
[    7.051460] x15: 0000000000000000 x14: 000000000000024e 
[    7.052077] x13: 0000000000000001 x12: 0000000000000000 
[    7.052694] x11: 0000000000000001 x10: 0000000000000960 
[    7.064723] x9 : 0000000000000000 x8 : 0000000000000000 
[    7.076596] x7 : 0000000000000000 x6 : 0000000000000000 
[    7.088340] x5 : 0000000000100000 x4 : 0000000000c00008 
[    7.099979] x3 : ffff000013000000 x2 : 000000000080000a 
[    7.111498] x1 : ffff000013c00008 x0 : ffff000010000000 
[    7.122902] Process systemd-udevd (pid: 2317, stack limit = 0x(____ptrval____))
[    7.134519] Call trace:
[    7.145583]  rockchip_pcie_rd_conf+0x18c/0x1f8 [pcie_rockchip_host]
[    7.157044]  pci_bus_read_config_dword+0x84/0xe0
[    7.168278]  pci_bus_read_dev_vendor_id+0x2c/0x1a0
[    7.179422]  pci_scan_single_device+0x78/0xf8
[    7.190431]  pci_scan_slot+0x34/0xf0
[    7.201243]  pci_scan_child_bus_extend+0x50/0x290
[    7.212087]  pci_scan_bridge_extend+0x2ec/0x4e0
[    7.222814]  pci_scan_child_bus_extend+0x1e4/0x290
[    7.233469]  pci_scan_root_bus_bridge+0x58/0xd8
[    7.244022]  rockchip_pcie_probe+0x60c/0x750 [pcie_rockchip_host]
[    7.254833]  platform_drv_probe+0x50/0xa0
[    7.265419]  driver_probe_device+0x208/0x2e8
[    7.275995]  __driver_attach+0xd4/0xd8
[    7.286409]  bus_for_each_dev+0x74/0xc8
[    7.296708]  driver_attach+0x20/0x28
[    7.306848]  bus_add_driver+0x1ac/0x218
[    7.316892]  driver_register+0x60/0x110
[    7.326831]  __platform_driver_register+0x40/0x48
[    7.336604]  rockchip_pcie_driver_init+0x20/0x1000 [pcie_rockchip_host]
[    7.336621]  do_one_initcall+0x5c/0x178
[    7.355610]  do_init_module+0x58/0x1b0
[    7.364755]  load_module+0x1e14/0x2210
[    7.373778]  sys_finit_module+0xcc/0xe8
[    7.382700]  __sys_trace_return+0x0/0x4
[    7.391247] Code: 7100129f 54fff921 f94002a0 8b130013 (b9400273) 
[    7.399680] ---[ end trace 706cbd252753b386 ]---
```

brucesm, 2018-08-11:
Bootlogs below.

Kernel 4.4.132-1075 (ayufan 0.7.9):
[Intel I350-T4 -- works perfectly](https://gist.githubusercontent.com/brucesm/729fc372fb73fe988e6c227867516afc/raw/45a91b0cfe4bf1265dd3dceed0cca7a2624b5a8f/working%2520(Intel%2520I350-T4))
[Mellanox ConnectX-2 MPNA19-XTR -- fails, trace included](https://gist.githubusercontent.com/brucesm/729fc372fb73fe988e6c227867516afc/raw/45a91b0cfe4bf1265dd3dceed0cca7a2624b5a8f/fails%2520(Mellanox%2520MPNA19-XTR))
[Mellanox ConnectX-2 MHQH29C -- rockchip-pcie: probe of f8000000.pcie failed with error -110](https://gist.githubusercontent.com/brucesm/729fc372fb73fe988e6c227867516afc/raw/45a91b0cfe4bf1265dd3dceed0cca7a2624b5a8f/fails%2520(Mellanox%2520MHQH29C)) 

Kernel 4.18.0-rc8-1060 (ayufan):
[Intel I350-T4 -- working perfectly](https://gist.githubusercontent.com/brucesm/f97c94cfd190e5a24f2d2ec7ed7b4dc1/raw/838550c54f0a417372a70294a9626b7c2d88b0e0/working%2520--%2520Intel%2520I350-T4)
[Mellanox ConnectX-2 MPNA19-XTR -- fails with call trace](https://gist.githubusercontent.com/brucesm/f97c94cfd190e5a24f2d2ec7ed7b4dc1/raw/838550c54f0a417372a70294a9626b7c2d88b0e0/fails%2520--%2520mellanox%2520mnpa19-xtr)
[Mellanox ConnectX-2 MHQH29C -- rockchip-pcie: probe of f8000000.pcie failed with error -110](https://gist.githubusercontent.com/brucesm/729fc372fb73fe988e6c227867516afc/raw/45a91b0cfe4bf1265dd3dceed0cca7a2624b5a8f/fails%2520(Mellanox%2520MHQH29C))

hopkinskong, 2018-08-18:
Having the same issue, cross ref:
https://github.com/ayufan-rock64/linux-build/issues/254

luckcolors, 2018-09-04:
Can we please have some updates on this issue?

foundObjects, 2018-09-06:
Anyone? I'm about ready to sell my RockPro64 and just use x86_64 for my project.

I'm 100% willing to supply any debug information you might need, and I've got about a dozen different PCIe network cards here I can test with.

rich0, 2018-11-30:
I'm running into similar issues with an LSI HBA card.  It works without issue in a standard x86 motherboard.  gen1 training times out when the board is inserted and nothing shows up in lspci.  If I plug in a USB3 host card it seems to work fine, so the PCIe slot itself is fine (gen2.1 rockpro64 board).

I built 4.4.154-1124-rockchip-ayufan with PCI_DEBUG enabled and captured this dmesg output with the LSI card installed:
https://pastebin.com/SAZPFpXD

Now, this card is an 8x card in a 4x slot, so as an experiment I ran it through a 1x PCIe mining adapter.  It works just fine in an x86 motherboard in this config.  When I use this with the rockpro64 I get a new error:
https://pastebin.com/QmyqyNNX

To try something different I rebuilt the kernel, but extending the gen1 PCIe training timeout from 500ms to 5s (drivers/pci/host/pcie-rockchip.c line 619).  The board boots normally without the LSI card, just giving the usual gen1 timeout message.  If I boot it with the LSI card installed directly (no 1x adapter) now I get the error again:
https://pastebin.com/vixEZKr4

So, perhaps it is timing out too quickly or taking too long to train without the 1x adapter, which prevents it from getting to the error.  For some reason it trains faster with the 1x adapter.

Apologies if this is an unrelated issue - if so I'm happy to create a new one.  I'm happy to test anything at this point.

nuumio, 2018-12-19:
Any news about this? I'm having same the same problem with LSI 9201 card. If it's of any help here's few logs from my Rockpro64.

With 4.4 kernel (ayufan's), serial console log (3 crashes):
- https://pastebin.com/0AhYnWkm

With 4.20-rc6 (ayufan's + patch to disable mmc command queueing), serial console log (3 crashes) and dmesg from last attempt:
- https://pastebin.com/RDT4XmPv
- https://pastebin.com/EJ9wCKjd

wombat, 2019-01-08:
I am having the same issue with a Delock PCI Express Card > Mini PCIe adapter connected to a Telit LM960 LTE module: https://pastebin.com/FHMGRgVG

foundObjects, 2019-02-16:
Has there been any motion on this at all? I'm sitting on what's effectively a useless board for my application (10GbE routing) at the moment since I can't bring up any of the PCIe NICs I've tested (I've tried around 10 different NICs at this point.) The only NICs I've managed to use successfully are Intel i350 and similar boards.

samex, 2019-03-12:
Didn´t debug why is my rockpro64 not booting with an **LSI 3Ware 9650** , but I bet I have the same issues like above some people reporting.

Would be nice to know where we can start to solve that problem?

nuumio, 2019-04-03:
Updating my status: Looks like I got my LSI 9201 working at least with one SSD drive. For some reason PCIe driver seems to need some delay between training and bus scanning. I built a test kernel with this workaround on top of ayufan's latest 4.4: https://github.com/nuumio/linux-kernel/releases/tag/nuumio-4.4-pcie-scan-sleep-02

The most relevant change is: https://github.com/nuumio/linux-kernel/commit/5a65b17686002dc84d461bffa324a2cb68e67aee (in branch: https://github.com/nuumio/linux-kernel/commits/nuumio-4.4-pcie-scan-sleep).

Last time I tried this with a bit older kernel I got the controller up but it kept resetting the connection to SSD every few seconds. Now with more patches it seems somewhat stable. I have no idea about the root cause but hopefully this gives ideas where the actual problem is. Curiously the delay needed for this is about the same that was needed earlier for deferring SDIO initialization to get WiFi/BT module and PCIe working at the same time for Rockpro64 (that was finally done so that SDIO driver waits until PCIe is finished).

My current setup:

- Rockpro64 4GB (power from 12V 6A PSU)
- LSI 9201
- WD Blue 250GB SSD (power from ATX PSU)
- eMMC as boot + rootfs media
- Pine64 WiFi/BT module connected, no WiFi nor BT configured
- Network via ethernet
- OS: ayufan's bionic minimal build + my own kernel

4.4-development branch seems quite active currently. I hope you get this one resolved too :)

rich0, 2019-04-06:
Just to comment for the record and the benefit of the many others with this issue, nuumio's patch (which seems to be in line to be released on ayufan) fixes my issue.  You just need to set a command line parameter to enable the delay (I haven't worked out the minimum required delay yet).

I was also having power issues which were solved by a 1x mining adapter.  Using a 5A power supply is likely to address that problem though nobody has tested the whole thing under heavy load yet.  

I'll be doing actual testing of the drives/etc but for now I can get the HBA to show up in lspci.  ayufan also enabled LSI HBAs in his kernels.

RchGrav, 2019-06-01:
> I was also having power issues which were solved by a 1x mining adapter. 
> 

The riser is shorting PCIe pin A1 to B17 to provide "presence" as a 1x card.. 
See https://imgur.com/a/AJB71Ih 

Shorting pin A1 to the second presence pin B31 would make a PCIe card detect as a 4x.*
Explanation: PCIe Cards short these presence pins to indicate the number of PCIe lanes / BUS width the connection will be using. (Note: I don't yet have a RockPro64 to test this on yet..  but here is an example of shorting the pin on the riser https://imgur.com/a/4rl7T5I taken from my plex server...

PCIe Pinout https://imgur.com/a/tqKJLd2

rich0, 2019-06-02:
The card works fine on a powered 16x riser cable as well, like this one:
https://www.amazon.com/gp/product/B01NAE4O7I/

The only downside to this powered riser is that it seems to drive power back into the rockpro64 such that it remains powered on even after disconnected from the power supply.  I don't generally run it this way as I am not certain that drawing current in this way isn't harmful.

So, aside from the likely power issue, the current ayufan kernels address my issues.

foundObjects, 2019-06-04:
I have similar back-powering issues just using a USB UART; I've found I
have to disconnect *everything* from the board when powering it down.

I rarely shut mine off so it hasn't been much of a problem but if I were
shutting off regularly I'd put everything powered on one power strip with a
switch and just use that.

StuartIanNaylor, 2019-06-07:
I am the same with a rockpi4b

```
[    1.489159] of_get_named_gpiod_flags: parsed 'gpio' property of node '/vcc3v3-pcie-regulator[0]' - status (0)
[    1.489201] reg-fixed-voltage vcc3v3-pcie-regulator: Looking up vin-supply from device tree
[    1.489236] vcc3v3_pcie: supplied by vcc3v3_sys
[    1.489697] vcc3v3_pcie: at 3300 mV 
[    1.489857] reg-fixed-voltage vcc3v3-pcie-regulator: vcc3v3_pcie supplying 0uV
[    1.623989] phy phy-pcie-phy.9: Looking up phy-supply from device tree
[    1.623999] phy phy-pcie-phy.9: Looking up phy-supply property in node /pcie-phy failed
[    1.625451] rockchip-pcie f8000000.pcie: GPIO lookup for consumer ep
[    1.625461] rockchip-pcie f8000000.pcie: using device tree for GPIO lookup
[    1.625490] of_get_named_gpiod_flags: parsed 'ep-gpios' property of node '/pcie@f8000000[0]' - status (0)
[    1.625725] rockchip-pcie f8000000.pcie: Looking up vpcie3v3-supply from device tree
[    1.625736] rockchip-pcie f8000000.pcie: Looking up vpcie3v3-supply property in node /pcie@f8000000 failed
[    1.625748] rockchip-pcie f8000000.pcie: no vpcie3v3 regulator found
[    1.626340] rockchip-pcie f8000000.pcie: Looking up vpcie1v8-supply from device tree
[    1.626350] rockchip-pcie f8000000.pcie: Looking up vpcie1v8-supply property in node /pcie@f8000000 failed
[    1.626360] rockchip-pcie f8000000.pcie: no vpcie1v8 regulator found
[    1.626951] rockchip-pcie f8000000.pcie: Looking up vpcie0v9-supply from device tree
[    1.626960] rockchip-pcie f8000000.pcie: Looking up vpcie0v9-supply property in node /pcie@f8000000 failed
[    1.626970] rockchip-pcie f8000000.pcie: no vpcie0v9 regulator found
[    2.172391] rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout!
[    2.173158] rockchip-pcie: probe of f8000000.pcie failed with error -110
```

lspci returns nothing then other times I will boot and

```
rock@linux:/boot$ lspci
00:00.0 PCI bridge: Fuzhou Rockchip Electronics Co., Ltd Device 0100
01:00.0 SATA controller: ASMedia Technology Inc. ASM1062 Serial ATA Controller (rev 02)
```

Seems completely spurious sometimes I hits runs of it working sometimes runs not.
Its like when you get 2 services clash that by chance change order at times.

prusnak, 2020-04-22:
I see the same issue with the following setup:

* PINE64 RockPro64
* [ROCKPro64 PCI-e X4 to M.2/NGFF NVMe SSD Interface Card](https://store.pine64.org/?product=rockpro64-pci-e-x4-to-m-2ngff-nvme-ssd-interface-card)
* ADATA XPG SX6000 Pro SSD 256GB

```
[    2.172391] rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout!
[    2.173158] rockchip-pcie: probe of f8000000.pcie failed with error -110
```

PhoenixMage, 2020-05-14:
I see similar issues to this on a Rock Pi 4 being unable to detect, pcie (error -110) and hence the NVMe m.2 drive running kernel 5.6.7.

The issue is intermittent as the NVMe drive is detected in linux about 5% of the time.

If I use the u-boot provided by radxa (rather then mainline with rockchip patches) this then becomes 100% so I am not sure if there is some form of PCIe initialisation that is happening in the u-boot that resolves the issue. I would still like to see my NVMe working with mainline kernel and mainline u-boot.

jkoppen-headsfirst, 2020-07-07:
Here too a boot freeze when a PCIe adapter is present (PCIe x1 to Mini PCIe adapter with Coral Edge TPU).
This is my output without:

> dmesg | grep pci
[    1.473348] of_get_named_gpiod_flags: parsed 'gpio' property of node '/vcc3v3-pcie-regulator[0]' - status (0)
[    1.473399] reg-fixed-voltage vcc3v3-pcie-regulator: Looking up vin-supply from device tree
[    1.473442] vcc3v3_pcie: supplied by dc_12v
[    1.473509] vcc3v3_pcie: 3300 mV
[    1.473666] reg-fixed-voltage vcc3v3-pcie-regulator: vcc3v3_pcie supplying 3300000uV
[    1.892811] phy phy-pcie-phy.5: Looking up phy-supply from device tree
[    1.892821] phy phy-pcie-phy.5: Looking up phy-supply property in node /pcie-phy failed
[    1.894568] rockchip-pcie f8000000.pcie: GPIO lookup for consumer ep
[    1.894578] rockchip-pcie f8000000.pcie: using device tree for GPIO lookup
[    1.894607] of_get_named_gpiod_flags: parsed 'ep-gpios' property of node '/pcie@f8000000[0]' - status (0)
[    1.894856] rockchip-pcie f8000000.pcie: Looking up vpcie3v3-supply from device tree
[    1.894949] rockchip-pcie f8000000.pcie: Looking up vpcie1v8-supply from device tree
[    1.894960] rockchip-pcie f8000000.pcie: Looking up vpcie1v8-supply property in node /pcie@f8000000 failed
[    1.894974] rockchip-pcie f8000000.pcie: no vpcie1v8 regulator found
[    1.895002] rockchip-pcie f8000000.pcie: Looking up vpcie0v9-supply from device tree
[    1.895013] rockchip-pcie f8000000.pcie: Looking up vpcie0v9-supply property in node /pcie@f8000000 failed
[    1.895025] rockchip-pcie f8000000.pcie: no vpcie0v9 regulator found
[    1.895049] rockchip-pcie f8000000.pcie: bus-scan-delay-ms in device tree is 1000 ms
[    1.895084] rockchip-pcie f8000000.pcie: missing "memory-region" property
[    1.895121] PCI host bridge /pcie@f8000000 ranges:
[    1.942370] rockchip-pcie f8000000.pcie: invalid power supply
[    2.442415] rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout!
[    2.442463] rockchip-pcie f8000000.pcie: deferred probe failed
[    2.442725] rockchip-pcie: probe of f8000000.pcie failed with error -110
[    2.737720] ehci-pci: EHCI PCI platform driver
[    4.379833] vcc3v3_pcie: disabling

clarkis117, 2020-08-10:
Here's my output from a rockpro64 with a Compex WLE1216VX attached to the PCIE slot
root@FarmBox:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.52 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r14101-5d8fded26a)) #0 SMP PREEMPT Sun Aug 9 12:01:52 2020
...
root@FarmBox:~# dmesg | grep  pci
[    0.286897] vcc3v3_pcie: supplied by vcc12v_dcin
[    0.314101] rockchip-pcie f8000000.pcie: no vpcie1v8 regulator found
[    0.314139] rockchip-pcie f8000000.pcie: no vpcie0v9 regulator found
[    0.868307] rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout!
[    0.868506] rockchip-pcie: probe of f8000000.pcie failed with error -110
[    0.990147] ehci-pci: EHCI PCI platform driver

StuartIanNaylor, 2020-08-11:
@clarkis117 
You prob want to check the difference between m.2 & mini pcie.
RockPro64 is m.2 is it not?

clarkis117, 2020-08-11:
@StuartlanNaylor it is a mini pice card, which I have in a mini pcie to pcie adapter. The rockpro64 has a 4x pcie card slot on its board. It may be a power design issue as I was able to use an Intel wifi adapter in the same setup, and the compex card in an x86 pc with the same adapter. The compex card has a TDP greater than 10 watts.

nullr0ute, 2020-10-22:
So one thing I found with testing is if you enable CONFIG_DEBUG_SHIRQ it shows up some issues on the driver

vebmaster, 2021-11-16:
@jkoppen-headsfirst 
> Here too a boot freeze when a PCIe adapter is present (PCIe x1 to Mini PCIe adapter with Coral Edge TPU).

Hello. Have you solved the problem with the "Coral Edge TPU"?

nullr0ute, 2021-11-29:
> @jkoppen-headsfirst
> 
> > Here too a boot freeze when a PCIe adapter is present (PCIe x1 to Mini PCIe adapter with Coral Edge TPU).
> 
> Hello. Have you solved the problem with the "Coral Edge TPU"?

I've had reports the Coral Edge TPU does work on Fedora. Note the Edge TPU PCIe driver which was in staging upstream has now been dropped from the upstream kernel so the testing was by someone that built their own kernel to bring those drivers back.

vebmaster, 2021-11-30:
@nullr0ute 
I don't have a coral edge yet, I'm still picking up a board.
Have you tried installing drivers according to the instructions https://coral.ai/docs/m2/get-started?

daiaji, 2022-06-07:
https://gitlab.manjaro.org/manjaro-arm/packages/core/linux/-/issues/34
There are some compatibility issues, a piece that can use an SSD on a PC will cause a kernel freeze/ on the RK3399.