rockchip-linux/kernel #117: Stop booting in RK3188

dcostan, 2018-08-11:
Hello all
I have a chinese board with RK3188 and I tried to build and use this kernel with Radxa Rock device tree. Unfortunately the boot process stops after a few seconds and it doesn't go on. Here is the serial
```
DDR Version 1.04 20140217
In
DDR3
300MHz
Bus Width=32 Col=10 Bank=8 Row=16 CS=1 Die Bus-Width=8 Size=2048MB
Memory OK
OUT

barebox 2016.02.0 #18 Sat Mar 5 09:20:35 CET 2016

Board: Radxa Rock
clk_register clk xin24m is already registered, skipping!

of_clk_init: failed to init clock for /oscillator: 1

arc-emac 10204000.ethernet: ARC EMAC detected with id: 0x7fd02
mdio_bus: miibus0: probed
dw_mmc 10214000.dwmmc: registered as 10214000.dwmmc
mshc1: detected SD card version 1.0
mshc1: registered mshc1
netconsole: registered as netconsole-1
i2c-gpio i2c-gpio0: using pins 58 (SDA) and 59 (SCL)
malloc space: 0x9ff7d800 -> 0xdfefafff (size 1023.5 MiB)
envfs: wrong magic
running /env/bin/init...

Hit any key to stop autoboot:  3 2 1 0
blspec: blspec_scan_directory: /mnt/mshc1.0 loader/entries
ext4 ext40: EXT2 rev 1, inode_size 256
blspec: blspec_scan_directory: /mnt/mshc1.1 loader/entries
blspec: blspec_scan_directory: mshc1 loader/entries
blspec: booting Debian
 from mshc1

Loading ARM Linux zImage '/mnt/mshc1.0/rk3188-radxarock.img.neww
'
Loading devicetree from '/mnt/mshc1.0/rk3188-radxarock.dtb.neww
'
commandline:  console=ttyS2,115200 console=tty0 console=ttyS2,115200 root=/dev/mmcblk0p2 rootdelay=3 rootfstype=ext4 rw clocksource=jiffies earlyprintk debug
  
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.143 (matlin@matlin-PC) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05) ) #5 SMP Wed Aug 8 15:48:09 CEST 2018
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Radxa Rock
[    0.000000] cma: Reserved 16 MiB at 0xdf000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 524288
[    0.000000] free_area_init_node: node 0, pgdat c10d5bc0, node_mem_map eeffa000
[    0.000000]   Normal zone: 1536 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO batch:31
[    0.000000]   HighMem zone: 327680 pages, LIFO batch:31
[    0.000000] PERCPU: Embedded 14 pages/cpu @eef9d000 s24728 r8192 d24424 u57344
[    0.000000] pcpu-alloc: s24728 r8192 d24424 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 522752
[    0.000000] Kernel command line:  console=ttyS2,115200 console=tty0 console=ttyS2,115200 root=/dev/mmcblk0p2 rootdelay=3 rootfstype=ext4 rw clocksource=jiffies earlyprintk debug
  
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 2046320K/2097152K available (10240K kernel code, 892K rwdata, 2912K rodata, 1024K init, 582K bss, 34448K reserved, 16384K cma-reserved, 1294336K highmem)
[    0.000000] Virtual kernel memory layout:
[... 95 lines cut ...]
[    0.456753] io scheduler noop registered
[    0.456813] io scheduler deadline registered
[    0.456941] io scheduler cfq registered (default)
[    0.465623] dma-pl330 20018000.dma-controller: Loaded driver for PL330 DMAC-241330
[    0.465721] dma-pl330 20018000.dma-controller: 	DBUFF-32x8bytes Num_Chans-6 Num_Peri-12 Num_Events-12
[    0.470158] dma-pl330 20078000.dma-controller: Loaded driver for PL330 DMAC-241330
[    0.470253] dma-pl330 20078000.dma-controller: 	DBUFF-64x8bytes Num_Chans-7 Num_Peri-20 Num_Events-14
[    0.472143] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[    0.475742] 10124000.serial: ttyS0 at MMIO 0x10124000 (irq = 22, base_baud = 1500000) is a 16550A
[    0.477354] 10126000.serial: ttyS1 at MMIO 0x10126000 (irq = 23, base_baud = 1500000) is a 16550A
[    0.479018] console [ttyS2] disabled
[    0.479205] 20064000.serial: ttyS2 at MMIO 0x20064000 (irq = 30, base_baud = 1500000) is a 16550A
[    1.373506] console [ttyS2] enabled
[    1.378975] 20068000.serial: ttyS3 at MMIO 0x20068000 (irq = 31, base_baud = 1500000) is a 16550A
[    1.391560] [drm] Initialized drm 1.1.0 20060810
[    1.399586] usbcore: registered new interface driver udl
[    1.409777] brd: module loaded
[    1.428425] loop: module loaded
[    1.433928] zram: Added device: zram0
[    1.438175] lkdtm: No crash points registered, enable through debugfs
[    1.446961] tun: Universal TUN/TAP device driver, 1.6
[    1.452791] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>
```
In addition mainline kernel from Torvalds boots, however it does not contain board specific drivers
Thank you all

peppelinux, 2020-09-21:
Got it to work here:
https://github.com/peppelinux/Radxa-Rock-RK3188/blob/master/README.md

now I'm compiling rockchip-linux 4.4 instead of linux-stable 5.8.7, further updates on that repository.
Don't know why you're encourring in that, I've added init= parameter in boot arguments, I can see systemd starting properly

peppelinux, 2020-09-21:
Got this

````
Loading ARM Linux zImage '/mnt/mshc1.0/zImage'
zImage: concatenated oftree detected
commandline: <NULL>
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.194 (root@peppelinux-desktop) (gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu1) ) #1 SMP PREEMPT Mon Sep 21 14:29:25 CEST 2020
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Radxa Rock
[    0.000000] cma: Reserved 16 MiB at 0xdf000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] PERCPU: Embedded 13 pages/cpu @eeda7000 s23948 r8192 d21108 u53248
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 522560
[    0.000000] Kernel command line: console=ttyS2,115200  root=/dev/mmcblk0p2 rootwait init=/init
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 2041400K/2097152K available (10240K kernel code, 1193K rwdata, 4160K rodata, 1024K init, 2163K bss, 39368K reserved, 16384K cma-reserved, 1294336K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0b00000   (11232 kB)
[    0.000000]       .init : 0xc1000000 - 0xc1100000   (1024 kB)
[    0.000000]       .data : 0xc1100000 - 0xc122a458   (1194 kB)
[    0.000000]        .bss : 0xc122c000 - 0xc1448da4   (2164 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000021] sched_clock: 64 bits at 150MHz, resolution 6ns, wraps every 2199023255551ns
[    0.000056] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0x2298375bd0, max_idle_ns: 440795208267 ns
[    0.001227] Calibrating delay loop... 1194.16 BogoMIPS (lpj=1990656)
[    0.040248] pid_max: default: 32768 minimum: 301
[    0.040370] Security Framework initialized
[    0.040391] SELinux:  Initializing.
[    0.040531] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.040556] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.041729] Initializing cgroup subsys memory
[    0.041791] Initializing cgroup subsys freezer
[    0.041889] CPU: Testing write buffer coherency: ok
[    0.042531] CPU0: update cpu_capacity 1024
[    0.042556] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.042684] Setting up static identity map for 0x60100000 - 0x60100058
[    0.090418] Brought up 1 CPUs
[    0.090445] SMP: Total of 1 processors activated (1194.16 BogoMIPS).
[    0.090457] CPU: All CPU(s) started in SVC mode.
[    0.092163] devtmpfs: initialized
[    0.104126] VFP support v0.3: implementor 41 architecture 3 part 30 variant 9 rev 4
[    0.105136] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6370867519511994 ns
[    0.105193] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.108924] pinctrl core: initialized pinctrl subsystem
[    0.110968] NET: Registered protocol family 16
[    0.113069] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.133638] cpuidle: using governor ladder
[    0.160299] cpuidle: using governor menu
[    0.160392] Registered FIQ tty driver
[    0.177995] hw-breakpoint: found 5 (+1 reserved) breakpoint and 1 watchpoint registers.
[    0.178018] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.262724] sdmmc-supply: regulator get failed, ret=-517
[    0.265499] SCSI subsystem initialized
[    0.266017] usbcore: registered new interface driver usbfs
[... 197 lines cut ...]
[    2.533985] device-mapper: ioctl: 4.34.0-ioctl (2015-10-28) initialised: dm-devel@redhat.com
[    2.547101] device-mapper: verity-avb: AVB error handler initialized with vbmeta device: 
[    2.555448] Bluetooth: HCI UART driver ver 2.3
[    2.560048] Bluetooth: HCI UART protocol H4 registered
[    2.565318] usb 2-1: new high-speed USB device number 2 using dwc2
[    2.571630] Bluetooth: HCI UART protocol LL registered
[    2.577082] rtk_btusb: RTKBT_RELEASE_NAME: 20200318_BT_ANDROID_9.0
[    2.583511] rtk_btusb: Realtek Bluetooth USB driver module init, version 5.2.1
[    2.590791] rtk_btusb: Register usb char device interface for BT driver
[    2.598016] usbcore: registered new interface driver rtk_btusb
[    2.604184] cpu cpu0: OPP-v2 not supported
[    2.611311] cpu cpu0: OPP-v2 not supported
[    2.616099] Synopsys Designware Multimedia Card Interface Driver
[    2.623317] dwmmc_rockchip 10214000.dwmmc: Using PIO mode.
[    2.628908] dwmmc_rockchip 10214000.dwmmc: Version ID is 240a
[    2.634888] dwmmc_rockchip 10214000.dwmmc: DW MMC controller at irq 27,32 bit host data width,256 deep fifo
[    2.644961] dwmmc_rockchip 10214000.dwmmc: No vqmmc regulator found
[    2.783233] dwmmc_rockchip 10214000.dwmmc: 1 slots initialized
[    2.793070] hidraw: raw HID events driver (C) Jiri Kosina
[    2.802481] usb 2-1: New USB device found, idVendor=05e3, idProduct=0608
````

IT seems to me that:
mmcblk0 haven't been recognized (we use this in the kernel command line)
that serial connections stops

peppelinux, 2020-09-21:
using vanilla kernel 5.8.7 I can get a working prompt instead
https://github.com/peppelinux/Radxa-Rock-RK3188/blob/master/batrebox_rk3188_serial.log

my kernel configuration is here:
https://github.com/peppelinux/Radxa-Rock-RK3188/blob/master/linux-5.8.7.config

you can also read the README with my build instruction.