rockchip-linux/kernel #118: RK3399 PCIe not work

lee9505123, 2018-08-17:
When you boot a PCIe card, a kernel panic occurs.
Or I can not recognize when I plug in the card after booting.
Why is this happening?
RK3399 is the latest repo.

hopkinskong, 2018-08-18:
#116 already opened, don't duplicate.

lee9505123, 2018-08-20:
Please refer to attached the text file.
Both logs are panic when booting with PCIe card inserted.
The card is a USB to PCIe card.
Model: 1. 305NEC [EX(](http://www.ez-net.co.kr/new_2012/product/view.php?cid=5&sid=116&seq=51)
              2. 212U3 [PCIe(](http://www.compuzone.co.kr/product/product_detail.htm?ProductNo=178432&banner_check=naver&NaPm=ct%3Djl1llnbs%7Cci%3D42d8311af32aec79e7f87a975ad7971de39d939f%7Ctr%3Dslc%7Csn%3D116863%7Chk%3D7251662b412770762e124ab999cd2980c28cb96e)
[error_case2.txt](https://github.com/rockchip-linux/kernel/files/2300896/error_case2.txt)
[error_case1.txt](https://github.com/rockchip-linux/kernel/files/2300897/error_case1.txt)