rockchip-linux/kernel #123: dw_wdt: probe of ff1a0000.watchdog failed with error -2

berturion, 2018-09-27:
Hello,
With the help of **summers**, a member of ArchLinuxARM forum, we try to enable the watchdog device. But it seems that the device is not well described in the device-tree and the official documentation lacks the necessary information to fix it. The thread is here: https://archlinuxarm.org/forum/viewtopic.php?f=67&t=13087 and contains all the information we could gather across official documentations.

Here is an interesting investigation from my friend summers:

--------------------------------
From https://github.com/torvalds/linux/blob/master/arch/arm64/boot/dts/rockchip/rk3328.dtsi the rk3328 device tree is malformed:

       wdt: watchdog@ff1a0000 {
          compatible = "snps,dw-wdt";
          reg = <0x0 0xff1a0000 0x0 0x100>;
          interrupts = <GIC_SPI 40 IRQ_TYPE_LEVEL_HIGH>;
    };

Compare this to another snps,dw-wdt device, the rk3288 : https://github.com/torvalds/linux/blob/master/arch/arm/boot/dts/rk3288.dtsi it has:

    wdt: watchdog@ff800000 {
          compatible = "rockchip,rk3288-wdt", "snps,dw-wdt";
          reg = <0x0 0xff800000 0x0 0x100>;
          clocks = <&cru PCLK_WDT>;
          interrupts = <GIC_SPI 79 IRQ_TYPE_LEVEL_HIGH>;
          status = "disabled";
    };

It may be worth checking other snps,dw-wdt devices, to see how they are defined. (...) We need to know how the chip is wired, that may be in data sheets, but I can't find it. Only really rockchip can solve.

Now look at the rk3288 device, obvious change is the "clocks = <&cru PCLK_WDT>;" Now that is defined in https://github.com/torvalds/linux/blob/master/include/dt-bindings/clock/rk3288-cru.h and the clock is defined as

    #define PCLK_WDT   368

Now what you can find on the TRM for the rk3328 is that it has a PCLK clock: http://opensource.rock-chips.com/images/9/97/Rockchip_RK3328TRM_V1.1-Part1-20170321.pdf, so check the clocks defined in the kernel: https://github.com/torvalds/linux/blob/master/include/dt-bindings/clock/rk3328-cru.h and the only PCLK are:

    /* pclk gates */
    #define PCLK_GPIO0      200
    #define PCLK_GPIO1      201
    #define PCLK_GPIO2      202
    #define PCLK_GPIO3      203
    #define PCLK_GRF      204
    #define PCLK_I2C0      205
    #define PCLK_I2C1      206
    #define PCLK_I2C2      207
    #define PCLK_I2C3      208
    #define PCLK_SPI      209
    #define PCLK_UART0      210
    #define PCLK_UART1      211
    #define PCLK_UART2      212
    #define PCLK_TSADC      213
    #define PCLK_PWM      214
    #define PCLK_TIMER      215
    #define PCLK_BUS_PRE      216
    #define PCLK_PERI_PRE      217
    #define PCLK_HDMI_CTRL      218
    #define PCLK_HDMI_PHY      219
    #define PCLK_GMAC      220
    #define PCLK_H265      221
    #define PCLK_MAC2PHY      222
    #define PCLK_MAC2IO      223
    #define PCLK_USB3PHY_OTG   224
    #define PCLK_USB3PHY_PIPE   225
    #define PCLK_USB3_GRF      226
    #define PCLK_USB2_GRF      227
    #define PCLK_HDMIPHY      228
    #define PCLK_DDR      229
    #define PCLK_PERI      230
    #define PCLK_HDMI      231
    #define PCLK_HDCP      232
    #define PCLK_DCF      233
    #define PCLK_SARADC   234

And none of those look obviously like a WDT clock - so it doesn't have an obvious name.

So we just learn that its not clear, that we don't have a solution.

--------------------------------

Could you please help us, you at rockchip are the only ones that can complete the device tree for your board in order to make the watchdog device work.

Thanks.

itdaniher, 2019-01-16:
http://opensource.rock-chips.com/images/9/97/Rockchip_RK3328TRM_V1.1-Part1-20170321.pdf is looking pretty helpful. Same issue in mainline at https://github.com/ayufan-rock64/linux-mainline-kernel. Discussion at http://lists.infradead.org/pipermail/linux-rockchip/2018-October/021902.html. Relevant patchset at http://lists.infradead.org/pipermail/linux-rockchip/2015-January/002314.html.

Gonna circle around to this...

itdaniher, 2019-01-24:
@berturion I got it working without help from Rockchip. Not sure what my next step is, but....

```bash
username@rock64:/sys/devices/platform/ff1a0000.watchdog/watchdog/watchdog0$ find . -type f | xargs -i sh -c 'echo {}: $(cat {})'
./uevent: MAJOR=249 MINOR=0 DEVNAME=watchdog0
./nowayout: 0
./timeleft: 28
./dev: 249:0
./identity: Synopsys DesignWare Watchdog
./status: 0x8000
./state: active
./timeout: 28
./bootstatus: 0```

berturion, 2019-01-25:
@itdaniher, what is your kernel and distro please ? I am on Aarch64 (Arch linux), and my kernel is 4.19.12-1-ARCH. (I did not upgrade to 4.20.x because [it breaks all USB ports](https://archlinuxarm.org/forum/viewtopic.php?t=13321)).

itdaniher, 2019-01-25:
Armbian's spin of Ubuntu 18.04.1 with a patched 4.20.0 - watchdog needs small patches to kernel and dtb.

berturion, 2019-01-25:
Ok. I love Arch Linux too much to change to Armbian (even though it is also a very good distro). I must admit that the mainline kernel for rock64 is a little experimental, but I can cope with it for now.
I am happy to see your interest in this issue.

gymnodemi, 2019-02-20:
@itdaniher Would you be willing to share your patches for getting the watchdog on rk3328 at least somewhat functional? I've been looking at the docs and the patches you referenced but I'm not getting there very fast. Thanks

itdaniher, 2019-02-20:
@gymnodemi for sure. Here you go: https://gist.github.com/itdaniher/3dcd02a50292f42aa764c3f3808f977d

applied to 4.20 perfectly IIRC. HMU with issues, gonna rehash it for 5.0. Haven't built recently.

itdaniher, 2019-02-20:
WRT 4.20 - note that you'll have to custom patch a DTB, or grab from 5.0 RC or 4.19, as mentioned above. Happy to share mine if it'd be helpful.

gymnodemi, 2019-02-21:
@itdaniher Thank you this has been super helpful. 

I am a little confused about needing a custom patch DTB  I'm trying to use the rk3328-roc-cc.dtb from this kernel line: https://github.com/FireflyTeam/kernel

The watchdog timer is in the dtb file with the correct address and it shows up in my /sys/firmware/devicetree/base/watchdog@ff1a0000

Is there something that needs to change in the DTB file?

Thanks for all the help

gymnodemi, 2019-02-21:
Just to follow up in case anyone else ends up at this thread (like I did). The patches that itdaniher provided do provide a working watchdog (created /dev/watchdog) for use with the watchdog service and it will reset the board. 

One note, I did not need to patch or change the DTB file - your millage may vary depending on what source tree you are using. 

@itdaniher is it okay to propose your patches to the FireflyTeam kernel repo as a potential fix for the watchdog timer? I will totally give you the credit :) I'm just lazy and prefer to clone their repo instead of branching/merging with the commits. 

Thanks,

itdaniher, 2019-02-21:
@gymnodemi Please go ahead and share with them, wouldn't mind the credit either. :) Hopefully I'll be able to find the time to work towards upstreaming it after 5.0 is released. Note that `PCLK_WDT` might need to be changed to `236` because I think `235` was claimed for something else in upstream after 4.20 - (TS_ADC? IIRC?) - the index doesn't matter at all, though! Just an int for referring to the clock.

itdaniher, 2019-02-21:
Also, I'm happy to help people apply the patch if one of their maintainers wants to hit me up in this thread, or <handle>@gmail.com.

berturion, 2019-02-21:
> WRT 4.20 - note that you'll have to custom patch a DTB, or grab from 5.0 RC or 4.19, as mentioned above. Happy to share mine if it'd be helpful.

Does that mean that USB ports work again in 5.0 RC ?

gymnodemi, 2019-02-21:
> ``Note that `PCLK_WDT` might need to be changed to `236` because I think `235` was claimed for something else in upstream after 4.20 - (TS_ADC? IIRC?) - the index doesn't matter at all, though! Just an int for referring to the clock.

At least in the kernel branch (4.4.114) I'm using it was taken by:

`#define PCLK_ACODEC             235`

I'm glad to hear it doesn't matter I totally just reused 235 and it is working despite the double use. I will go ahead and increment that now that I now it will still work at 236.

itdaniher, 2019-02-21:
@berturion - yup! Also it's trivial to fix by hand in 4.20. I have a patch if you want it. The bad edit to the rock64 dts was reverted after 4.20 release.

itdaniher, 2019-02-21:
@gymnodemi I didn't try reusing a number but I'm glad to hear it worked! Guessing the worst case is sightly higher power usage, as an unused part of the chip would be activated / clocked. NBD but not best practices.

berturion, 2019-02-22:
@itdaniher thank you very much but I will wait for the 5.0 release as it will be done soon.

JpegXguy, 2019-06-15:
Good news: This is being handled upstream. http://lists.infradead.org/pipermail/linux-rockchip/2019-June/025357.html

It's on it's way to getting merged, hopefully for 5.3