rockchip-linux/kernel #135: rockchip_vop_reg.c artificially limits rk3399_vop_lit max_output to 2560x1600

fosf0r, 2018-11-29:
I have been unable to switch my HDMI output to 4K resolution until I edited and recompiled rockchip_vop_reg.c

For the vop_data struct of rk3399_vop_lit, I edited the properties and changed line 636 from:
	.max_output = {2560, 1600},
to:
	.max_output = {3840, 2160},

And my 4K TV booted right up in 3840x2160 mode in framebuffer.
Now all my GBM/DRM programs can use 4K too:  the program KODI could never see my TV's 4K resolution before, but now it can detect and switch to it.

Everything works fine so far; what is the purpose of the limit being 2560x1600 ?
Also, when is rk3399_vop_**big** used?
I notice that "big" has a better max_output.
Is it the fact that my system was using vop_lit instead of vop_big the actual problem?
Thanks in advance.

avafinger, 2018-12-28:
@fosf0r 
I was following this issue and I thought it was a bug.
BTW, Did you succeed to build and run Kodi 18rc3?

avafinger, 2018-12-31:
Never mind. Kodi 18rc4 is working now.