rockchip-linux/kernel #136: RKISP1  for ov5648 not works.

simoncheen, 2018-12-03:
hello,

I wrote a driver for ov5648 (base of ov5695), the i2c communication is working , and we and read and write regs via i2c , we try to run the fllowing scripts:
```bash
v4l2-ctl -d /dev/video0 --set-ctrl 'exposure=1216,analogue_gain=10'  --set-selection=target=crop,top=0,left=0,width=1296,height=972 --set-fmt-video=width=1296,height=972,pixelformat=BG10 --stream-mmap=3 --stream-to=/root/raw.out --stream-count=1 --stream-poll
```

but we got the fllowing logs:
```
[   49.617432] rkisp1: rkisp1_set_fmt: stream: 0 req(1296, 972) out(1296, 972)
[   49.617544] rkisp1: stream 0 crop(0,0)/1296x972
[   49.618786] rkisp1: vid-cap-mplane count 3, size 2519424
[   49.730434] ov5648 3-0036: set isp clk = 312500000Hz
[   49.730489] rkisp1: s_power: 1
[   49.730962] rkisp1: stream 0 crop disabled
[   49.731059] rkisp1: SP streaming = 0, MP streaming = 1
[   49.731099] rkisp1: CIF_ID 0x10853017
[   49.731225] rkisp1: 
                 MIPI_CTRL 0x00061f00
                 MIPI_IMG_DATA_SEL 0x0000002b
                 MIPI_STATUS 0x00002000
                 MIPI_IMSC 0x03f33333
[   49.731321] rkisp1: SP streaming = 0, MP streaming = 1
[   49.733348] rkisp1: SP streaming = 0, MP streaming = 1 MI_CTRL 0x00ba2008
                 ISP_CTRL 0x00000111 MIPI_CTRL 0x00061f01
[   52.768271] rkisp1: waiting on event return error 0
[   52.774877] rkisp1: MIPI mis error: 0x00020000
[   52.780272] rkisp1: SP streaming = 0, MP streaming = 0
[   52.780359] rkisp1: streaming(MP:0, SP:0), MI_CTRL:ba2000, ISP_CTRL:100, MIPI_CTRL:61f00
[   52.780600] rkisp1: s_power: 0
```

the problem is at ```rkisp1: MIPI mis error: 0x00020000```

how can I check the problem , and resovle the problem ?

thanks

simoncheen, 2018-12-19:
it works now

closed 2018-12-19 by simoncheen

ajaixavier, 2019-05-14:
@simonchen007  I am getting the a similar issue with imx214.   
[  703.383408] rkisp1: MIPI mis error: 0x00800000
[  703.448651] rkisp1: MIPI mis error: 0x01800000

How did you solve it?

simonchen007, 2019-05-14:
hi, 
for your problem , it's just few log, I can not comfirm what's the problem.
for ov5648, it's just because the reg: 0x4800, just add it at __ov5648_start_stream , it should works

> 
> 
> @simonchen007 I am getting the a similar issue with imx214.
> [ 703.383408] rkisp1: MIPI mis error: 0x00800000
> [ 703.448651] rkisp1: MIPI mis error: 0x01800000
> 
> How did you solve it?