rockchip-linux/kernel #137: linux/rockchip/iomap.h file missing

chozhana, 2018-12-15:
iomap.h missing at include/linux/rockchip/iomap.h which is needed for enabling rockchip hdmi kernel module

fosf0r, 2018-12-17:
Are you positive that your board uses it?

For instance, I have a Pine64 brand Rockpro64 (RK3399) and it uses Synopsys DesignWare HDMI (I see in my lsmod: dw_hdmi_cec, and dw_hdmi_i2s_audio)

Just throwing that out there. 
I'm surprised by the very few actual Rockchip modules that my board uses.

chozhana, 2018-12-18:
i was trying to revive my old MK918 box which runs on RK3188. The main battleneck with this hardware was the non availability of latest kernel (struck with old 3.0.36 kernel). So was trying to compile 4.4 kernel for RK3188. The older version was using many of rockchip specific modules, one among them is RK_HDMI.

fosf0r, 2018-12-18:
WELP:
https://github.com/rockchip-linux/kernel/commit/e85d7dbaa051106d0f4a354076c1938e8f75745a
¯\_(ツ)_/¯

The module is all wrong then -- I guess a rockchip-specific iomap is no longer necessary?

keveryang, 2019-01-08:
I'm sorry that RK3188 is not support in kernel 4.4, please try to use mainline kernel if you want to try with new kernel.

worldomkar, 2020-01-18:
Hi @keveryang ,
Thanks for your response, at least that explains why the support for RK3188 is missing even from https://github.com/rockchip-linux/kernel (especially for rockchip LCDC support). I noticed kernels are focused on 64-bit rockchips.. and I've been trying to resolve every C error by un-intuitively copying over files.. It would be greatly appreciated if you could share any pointer to information which could help me port below to https://github.com/rockchip-linux/kernel: "rk-plat" and "mach-rk3188", DDR3 support, especially GPIO and IT66121, LCDC0 support:

[image]
[image]
[image]
[image]

Its just hard to let MK908v3 go even after years and even after knowing it is based on 32-bit quad-core RK3188 as it is a beautiful piece of hardware with so much of possibility.

worldomkar, 2020-01-18:
Hi @fosf0r ,

The default rockchip_defconfig/rockchip_linux_defconfig works as they don't use RK frame buffer/LCDC0/GPIO or IT66121 (HDMI 1.4 transmitter) for my board (mk908v3).

As soon as one ticks any of above, kernel no longer is in good shape to compile.

worldomkar, 2020-05-14:
Hmm... AI got upto linux kernel 3.0.101.
For MK908v3
Working:
1. Power led (blue)
2. Boots from sdcard/USB
3. Debian Jessie can be debootstrapped (infact running mate-desktop-environment on Jessie)
4. Display and text-mode (fbcon) working
5. USB serial gadget (cdc_acm) works as default on /dev/ttyGS0 on OTG <=> PC connection

Not working:
1. HDMI Audio
2. WiFi (AP6210)

https://github.com/worldomkar/mk908-3-0-101

worldomkar, 2020-05-14:
> iomap.h missing at include/linux/rockchip/iomap.h which is needed for enabling rockchip hdmi kernel module

@chozhana 
You may find this interesting. Bumping your thread as you have recent post and RK3188 device.